daringer/asus-fan #6: Support for N551JK

KastB, 2015-01-04:
Hi,

this module runs just great, nevertheless I've some questions:
The only change I did was to add my model to the whitelist then I compiled, modprobed and run that script to read the fan speed.
I got values between 18 (cold state) and 35 (after some minutes of high cpu usage), thus I tried the "set" functionality.
When I set 18 (the read out speed at cold state) the fan just turned off. I had to set a value of 80 for the same noise than on 18 in auto mode. The read out speeds were those I set, and the fan ran at full speed for a set value of 255.
Is it a model specific behavior or just normal, that the read out fan speeds depend on the mode of the fan?
Additionally the registers for fan speed don't seem to be updated during manual mode, which can be observed by "watch -n 1 acer.pl regs".
Ist that normal, too?
Finally I'd really like to use your module to control the fans automatically, as the fan runs all the time without any reason. Unfortunatelly pwmconfig won't find the fan, even if the module is loaded, as it searches /sys/class/hwinfo[]/ for devices

daringer, 2015-01-04:
If the only thing you've done is adding your laptop model, please post the exact string and I will include it...

your questions:
- for me I see values between 0 and 255 during auto mode, but in auto mode the value never rises above 130 or so
- the difference between your reported value in auto mode and the set value in manual mode could originate in a not exactly matching acpi interface (looks like reading from the registers does not perfectly match my model)
- in manual mode it is true the registers don't get updated, so the module contains a workaround to expose the previously set manual value through "cur_state"
- currently only /sys/..../thermal/cooling_deviceX/ is supported, personally I am not happy with it, too... I could easily provide more data like fan-rpm , but the thermal-kernel-function-family does not allow more.... I am looking into other methods to expose the fan to the userland - any hints and contributions are welcome

admittedly my time will be quite limited in january, so don't except any bigger changes into this direction before february ;)

KastB, 2015-01-04:
Thank you for your prompt response!
The string, I've added is "N551JK" (actually '!strcmp(name, "N551JK")') on line 330 => there is only one single fan.
* how could I determine if the layout fits to yours? The registers for the fan speed output match to those described in the linked post of your anser "Support for Asus TP300L" (93,94 in little endian format). When calculating the RPMs from this value it seems feasible, too.
* which registers do you read or write?  I could only see the ACPI calls you did, but nothing that I could check or set manually.
* I guess a sysfs interface would solve several problems at once:
- pwmconfig and fancontrol could use it for automatic control (no need an internal algorithm)
- RPMs could be exported with that, too
but I've never programmed something with it, but just controlled the fans on an older pc - the RPMs where shown there, too. Here https://www.kernel.org/doc/Documentation/hwmon/sysfs-interface the fans and pwms are mentioned, too.

daringer, 2015-01-04:
thanks for the contribution, 
looked into the hwmon interface stuff - motivates me ;) pretty much what I was planing... let's see were I can scrape together the time to do it... closing this issue ... support for the N551JK is included

closed 2015-01-04 by daringer