daringer/asus-fan #8: Hwmon interface

KastB, 2015-02-10:
Hi,

there is a fork,  that uses the hwmon interface. It features compatibility to standard monitoring tools (xsensors, conky) and fan controllers (fancontrol with pwmconfig).
But it's a pre alpha release, as I didn't any module adaption before, and it's only tested on my notebook model (n551jk). 
It would be great if it could be merged to this project - or live as a parallel version and being reviewed by more experienced programmers.

Cheers
Bernd

daringer, 2015-03-01:
merged, things on the todo-list:
- rpm calculation (manual pwm set, method in general)
- auto mode activation "better" than setting the pwm to 256 (auto-mode always resets both fans to auto-mode, thus the auto-mode-on-switch is "system-wide" not "fan-wide" )

daringer, 2015-03-01:
here is a DSDT derived bpm-calculation based on the provided fan-speed numbers:
http://forum.notebookreview.com/threads/fan-control-on-asus-prime-ux31-ux31a-ux32a-ux32vd.705656/#post-9073954

the only strange thing here are the originating values, which seem to be taken directly from ACPI, but more importantly: they seem to lack a unit or similar - sooooo: try to scale our range of 0 - 255 onto these values could be pretty tedious,...

KastB, 2015-03-01:
I don't think the problem is the strange raw values for the RPMs that could be read out of the ACPI-registers. We know the function to calculate the RPMs from them (4312500 / (SPEED * 2)) but it's no use, as we can simply call the TACH function, and - even more important - they won't get updated on manual mode.

Probably The "cleanest" solution is to report zero RPMs when switching to manual mode, as we simply can't calculate the real value. I wasn't happy about that, as I wanted at least having a guess of the current fan speed  on manual mode. Thus I heated up my notebook and reduced the "fan1_speed_max" till I heard some change in speed. By writing down the RPMs and values I've written to "fan1_speed_max" I got some mapping table and fitted a square function to it. I get decent results using that function, but I don't know, if we want to set up such a mapping table for every single model?
On the other hand, I don't know a better solution. I couldn't find any information on the maximum speed of my fan to set up an universial function. Additionally the correlation between PWM values and RPMs isn't linear but quadratic.

Even worse, that we need that mapping table the other way round for reporting the correct pwm value at automatic mode. When not doing that fancontrol scripts could do crazy things, when they are started, as the first value they read is simply wrong.

Thus the best solution - till someone gets a great idea - is probably to report 0 RPMs on manual mode an 255 as the PWM value on auto mode for models without a mapping table and generating mapping tables for all supported models, that have to be converted to quadratic functions and written to  "__fan_get_cur_state" that calculates the PWM value with given RPMs and to "__fan_rpm" for calculating RPMs from a PWM signal

daringer, 2015-03-02:
ok, there is a python script collecting all information and showing them, I am using the info from [asus fanspeed thread](http://forum.notebookreview.com/threads/fan-control-on-asus-prime-ux31-ux31a-ux32a-ux32vd.705656/#post9146324) to calculate the rpm values from the values I get from ACPI-registers (using acer_ec.pl)...

My impression is that a calculated RPM-value (while the fan is manually set) is easily done, I'll do this, the tool to debug/evaluate this is already there. But more importantly a "0 RPM" does not only look crappy, moreover already a rough approximation delivers useful information for a userspace-fan-controller. 

rpm = 4312500 / (acpi-raw * 2) 

looks quite managable I think, the only thing remaining now is the conversion from the [0-255] values to the <acpi-raw> codomain [100 ~ 600] .... In particular, it looks like the value inside the acpi-register is already lineary correlated to the fan-speed, look here: [graph](http://fooplot.com/#W3sidHlwZSI6MCwiZXEiOiI0MzEyNTAwLygyKngpIiwiY29sb3IiOiIjRkMwMDAwIn0seyJ0eXBlIjowLCJlcSI6Iih4XjIpKjEwMDAvLTE2MDU0K3gqMzI2NDgvMTAwMC0zNjUiLCJjb2xvciI6IiM1RTc2RUIifSx7InR5cGUiOjEwMDAsIndpbmRvdyI6WyItMjA0My42NDQ4ODkxOTg0MDciLCIyMTE2LjExNzgwMzEwOTI4MDQiLCItMTAwLjEyMzY2NzY5MjMwNDI3IiwiMzkzMi44NjA5NDc2OTIzMDQzIl19XQ--) 

So, what should be done from here:

- revive the possibility to get the [0-255] values from the /sys/-interface, having those the relationship between 'acpi-raw' may be automatically deducted. (Also it feels awfully inconsistent, if one puts a number between 0-255 in and on the other side receiving RPM values, both should be supported (haven't checked the hwmon-docs for that, yet)

- reduce the relation between rpm <-> acpi-raw <-> [0-255] to a set of constants. (Personally, I don't think this must be very good, an userland-controller-tool should handle all those things, especially more sophisticated calculations and such---this module provides a (low-lvl) interface exposing as much data as available.

- to keep the constants, the module needs parameters

- finally, the auto-mode should have a better interface ...

gotta go, have to start ;)

KastB, 2015-03-02:
As stated above: there is no source for the RPMs, that is updated on auto mode. Not even those raw ACPI-registers.

Comments on your TODO list:
- this possibility never died: you can just read "pwmX" to get the current pwm-speed. It might not be correct on auto-mode, but that's a similar problem than wrong RPMs on manual mode

- the relation between rpm and acpi-raw is fixed. There is this formula, thats exact - no additional parameters or guessing. For calculating PWM from RPM device specific Parameters have to be found. By now it's done using 3 parameters : a*RPM^2+b*RPM+c = PWM. Probably that could be further reduced (for example we could try to find a formula, that derives these parameters with just the maximal fan-speed given) but we would need further mapping tables (the one of my laptop  can be found in the source-code; the description of how I got it can be found on my previous comment)

- we could try setting these parameters depending on the model name we read out

- what do you mean with that? That the controlling is weak or that it's not user friendly to use?

daringer, 2015-03-15:
only a quick one for the last point: I think setting pwm1 to 256 to activate auto-mode seems very hackish to me... a ```echo 1 > auto_enabled``` would be more desirable I guess, especially as for the two-fan-systems auto-mode always applies to both fans and may not be set to one of them exclusively...

concerning the previous points, guess I can invest some time the following week to push this a little more

KastB, 2015-03-18:
yes, that's right. But the standard behaviour is implemented now, and auto mode can be enabled by "echo 0 > pwm1_enable". We could disable the possibility of reenabling automode by writing 256. Nevertheless I think there should be two of the  "_enable" files - one for each fan - even if auto mode can only be enabled for both fans fancontrol for examples requires them.

daringer, 2015-06-21:
did some cleanups, removed auto-by-setting-256-functionality and now allow resetting to auto only through `echo 0 > pwmX_enable` .... added thermald config, see issue #18 ....

closed 2015-06-24 by KastB