daringer/asus-fan #10: Auto mode never seems to adjust fan speed past 50 on UX31A

frederickjh, 2015-02-23:
Hi!
 I have been using this project on my Asus UX31A. I have noticed that it at times that the computer seems to lock up. The mouse cursor most times still moves but nothing else works.

syslog reveals this:

Feb 24 00:33:34 frederick kernel: [806296.183024] CPU0: Core temperature above threshold, cpu clock throttled (total events = 421879)
Feb 24 00:33:34 frederick kernel: [806296.183025] CPU2: Core temperature above threshold, cpu clock throttled (total events = 421878)
Feb 24 00:33:34 frederick kernel: [806296.183027] CPU1: Package temperature above threshold, cpu clock throttled (total events = 504381)
Feb 24 00:33:34 frederick kernel: [806296.183029] CPU3: Package temperature above threshold, cpu clock throttled (total events = 504381)
Feb 24 00:33:34 frederick kernel: [806296.183030] CPU2: Package temperature above threshold, cpu clock throttled (total events = 504380)
Feb 24 00:33:34 frederick kernel: [806296.183038] CPU0: Package temperature above threshold, cpu clock throttled (total events = 504381)
Feb 24 00:33:34 frederick kernel: [806296.184042] CPU2: Core temperature/speed normal
Feb 24 00:33:34 frederick kernel: [806296.184042] CPU0: Core temperature/speed normal
Feb 24 00:33:34 frederick kernel: [806296.184043] CPU1: Package temperature/speed normal
Feb 24 00:33:34 frederick kernel: [806296.184044] CPU3: Package temperature/speed normal
Feb 24 00:33:34 frederick kernel: [806296.184045] CPU0: Package temperature/speed normal
Feb 24 00:33:34 frederick kernel: [806296.184051] CPU2: Package temperature/speed normal

When I can get the computer to respond during these incidents fan_speed.sh get never gives me more than a 50.

I can set the fan speed to max using:
sudo <path>/fan_speed.sh set 255

I do hear a noticeable increase in the speed and fan noise when I do this but if  I put it back on auto it never gets set above 50.

It seems to me that the fan speed should be ramping up much more quickly than in is on auto to reduce the temperature before it gets too high that the cpu clock gets throttled. It seems to me that the current fan speed adjustments are a bit too conservative. Is it possible to provide a more aggressive setting for quicker heat removal, even with increased fan noise and possibly the fan speed going up and down more than it currently does?

Thanks in advance for your time and help!

Frederick

daringer, 2015-02-24:
for my UX32D it's pretty much the same, (except for the freezes)...
I am currently into a "thing", there is a pull request pending implementing the hwmon interface, which will enable the control of the fan(s) from the userspace, maybe this will solve this issue.... I will comment here if it's done...

KastB, 2015-02-25:
Hi,

I think there are 3 problems described within this issue.

1) freezes - I did not recognize any on my laptop, and the log seems just to report some throttling - somthing quite "normal" not good, but not unusual and nothing related to monitoring the fan speed.
2) When reading out the pwm-signal the reported speed is way too low. I tried to adaped that on the "hwmon-version" but it's not perfect there either. The problem is, that we read out the speed with any unknown unit, that can be converted to RPMs with a known formula. Nevertheless we don't now the maximal speed of the fan and thus can't convert the RPMs to a valid pwm value. By now I have no idea how to implement a good and universial formula to report the correct value. The RPMs reported by the "hwmon-version" should be correct.
3) The auto mode built in the firmware of the laptop won't command the maximum speed - I recognized that, too, and there's nothing that can be done against it but using youre own fan controller (the easiest way would be waiting for the "hwmon-version" to be tested (or test it on your own responsibility) and use "pwmconfig" and "fancontrol")

frederickjh, 2015-02-25:
Well I am glad for you, KastB, that your laptop does not  freeze up when your cpu gets throttled for over temperature conditions. By me the story is much different. In the best of situations program windows go gray and I have to wait till they get color back to work with them.

At the next level the mouse cursor still will move and the seconds on the clock in the task bar will advance. Normally I cannot switch windows. Some times at this level I can still do a Ctrl+Alt+F1 to get to a terminal and see the CPU throttling messages.

At the most extreme level I can do nothing except wait. If things go on too long I have to do a Alt+Sysrq+B to reboot the machine.

I agree that it is good that the machine throttles the CPU on over temperature to prevent damage, however what I want is that the fan speed ramps up as the temperature goes up and also looks at the 1 minute load average and adjust the amount that the fan speed gets ramped up.

I find it totally ludicrous that the auto fan control currently maxes out at 50 out of 255 and then my CPU gets throttled because of CPU over temperature conditions. I would most likely be better of setting the fan manual to anything above 50 that I can handle with the noise until I can put together a script that will automatically adjust it. 

I need to first make a flow chart that would map out the strategy for automatically adjusting the fan speed based on the following:

<ul>
<li>Acceptable operation temperature range of the CPU</li>
<li>Current temperature of the CPU</li>
<li>Current 1 minute load average</li>
<li>Difference between the current temperature and the high and critical temperature ratings.</li>
</ul>

Anyone have any ideas on this strategy? I am happy to hear them.

Frederick

KastB, 2015-02-25:
The idea of this module seems to be to provide a user interface to monitor and setting the fans speed. That could be done using the version in the pull-request, pwmconfig and fancontrol. You can configure some parameters, like a minimal speed, the minimal temperature and the temperature where the fans run at maximal speed but this controller did worse for me than the original one. And that version of asus-fan is hardly tested.

Since freezing seems to appear even without this module it's probably related to the  scheduler  or somthing I've never heard of. Nevertheless speeding up the fans will probably just extend the time till freezes appear but not prevent them. And (most important for a bug report) it's not the fault of the module, that your machine freezes, but you want to solve your problem by using it.

Like mentioned above, the reported speed "50" is not 50 out of 255. If you would command "50" your fans would slow down. I guess "50" equals about 200 on the pwm-scale. Thus not full speed, but significantly more than just 20% of the maximal speed

daringer, 2015-03-02:
Take a look into issue #8, looks like there could be a solution for this in sight. pwmconfig+fancontrol works pretty good currently for the time I tested it --- I use [thermald](https://github.com/01org/thermal_daemon) to handle the performance vs. heat trade-off. despite the (untested ;) fan-control it uses the intel_rapl, intel_pstate, c-states, cpufreq and intel_clamp (make sure all are modprobed) drivers to regulate the whole system... My impression is that this works by far better than the fan exclusively with no (felt) performance drawbacks...

frederickjh, 2015-04-04:
@daringer I upgraded to the lastest version and installed thermald. This works very well even without adjusting the thermald config file at all.

What does not work is the setting of the fan speed with <pre>echo 123 > ${fpath}/pwmX</pre>  Here is a sample terminal session.

<pre> $ cat /sys/class/hwmon/hwmon2/pwm1
171
$ echo 180 > /sys/class/hwmon/hwmon2/pwm1
bash: /sys/class/hwmon/hwmon2/pwm1: Permission denied
$ sudo echo 180 > /sys/class/hwmon/hwmon2/pwm1
bash: /sys/class/hwmon/hwmon2/pwm1: Permission denied
$</pre> 

I am going to close this as the auto mode, it seems from the comments, comes from somewhere else.

closed 2015-04-04 by frederickjh

Tharre, 2015-04-04:
> sudo echo 180 > /sys/class/hwmon/hwmon2/pwm1

This can't work because `sudo` only executes `echo` as root but the redirection of the ouput is done by bash which does not have root privileges. Either change into a root shell by executing `sudo -i` or do it like this:

    echo 180 | sudo tee /sys/class/hwmon/hwmon2/pwm1