daringer/asus-fan #13: Should be using WMI

Tharre, 2015-03-06:
If this module is ever to be merged into the mainline linux kernel we to make use of the exported WMI interface. And at least to me, that seems like a goal worth pursuing as maintaining a module outside of the kernel tree is very annoying.

As far as I can see, most of the work has already been done anyway already [1].

Objections?

[1] https://lkml.org/lkml/2013/10/8/800

KastB, 2015-03-10:
You're right, but as long as we don't find a solution that doesn't need " virt_to_phys" and uses the wmi interface it seems useless. Otherwise we would already see some cooling device with the newer kernels - the patch is older than one year...

daringer, 2015-03-15:
I've seen this patch while first looking into the fan-stuff - admittedly I haven't digged deep enough to really understand why the asus-wmi module does not provide a fan-interface. KastB do you understand why, could you further explain? 

merging into mainline would be a mid/longterm goal, but yes already asked myself if inside asus-wmi or a separate module, once I've enough insight and overview (and maybe a better working module) I would also try to ask inside the acpi-kernel mailinglist for further guidance/hints...

Tharre, 2015-03-15:
I assume the reason why the asus-wmi module does not provide a fan interface is simply because nobody cared to implement it yet. After all it's not necessary, fans do work without a driver, and most asus laptops seem to have rather sane fan behavior as well.

daringer, 2015-03-15:
Personally, I'd like to be able to 
- see RPM (for both fans) (e.g., be able to see and react on a failure)
- use different profiles (silent, cool, performance) (especially the latter is something important inherent to ultrabook-designs---get max-performance far longer, if the fan is used more aggressively---which won't happen in auto-mode)
- control my hardware, if possible (matter of principle ;D)

Tharre, 2015-03-15:
I agree that integrating this into asus-wmi is trivial, if that's even the correct choice, and that having a standalone module is a lot better for development. However, even for that moving to the WMI interface seems to be the right choice as we don't have to fix issues like the virt_to_phys stuff just yet anyway.

KastB, 2015-03-17:
I tried that patch  - additionally to the virt_to_phys thing I suffered the problem, that writing to the device won't work for the first time. If you write a second time, short after the first time, the speed is set, nevertheless, if you wait too long you have to write three or even four times till something happens - that isn't really good.
Did someone else experience the same behaviour?
If we would get that working it would be great if we could compare our dsdt.dsl as the MF13, that is used provides additional intersting functions - for example "GFNN" probably returns the number of installed fans => we could automatically determine which devices to set up
Within the "AGFN" function we could probably access additional temperature sensors for the graphics card. Some WMI methods, whose IDs are listed in the kernel are just listed but unused - there are some interesting ones there, too.
But we would need a stable interface to those functions, that access the system memory, and all my tries failed by now.
I couldn't find any documentation on that, too.
The only module that uses virt_to_phys is "dell-laptop.c" - and it's quite old and with no additional information on how to implement it properly.
When experimenting with own code there is often some memory corruption, thus you can't power down but have to force quit your laptop, that doesn't seem to be optimal, too.
Thus, it would be great to find some docu on how to do it - after the method for calling the acpi function is written lots of things would be possible - not just fan control.
Perhaps some other module already implemented such function. But I didn't find any

KastB, 2015-04-06:
Hi,
I solved the memory corruption problems, thus the new patch works on the first time and the laptop kept running all the time.
Additionally the number of fans should be automatically determined.
And I found a function within acpica that does the address translation, but it's marked thus unusable by now.
Thus it would be great if there would be some feedback on that patch. Espacially if it works on different machines, with more than one fan. Or if a machine, that doesn't support fan control is properly recognized.
If everything works fine we could try again to push it to the kernel - perhaps they remove the marking and no virt_to_phys is needed anymore.
The patched sources can be found at https://github.com/KastB/asus_wmi

daringer, 2015-07-19:
Well, KastB. My impression is that from what I currently know, the asus-fan module has no future (Correct me if I'm wrong). Should we maybe abandon and stall the development for this project and fully concentrate on the inclusion of it into the asus-wmi module?

how do you think? just freeze asus-fan and maintain building for the latest kernel? place this info on top of repo-readme and directly link to asus-wmi as the successor?

Tharre, 2015-07-19:
Well, yes. For all we know `asus_fan` has no chance of ever being merged into mainline linux, and modules outside of it are a hassle to maintain, both for users and developers.

However, for now there is still value in keeping `asus_fan` around, as it supports both fans. So I think it should be put in "maintenance only" mode until we can fully replace it.

daringer, 2015-07-19:
yes! had it running with hardcoded two fans setting today ;) check this pull request https://github.com/KastB/asus_wmi/pull/3 especially the comment...

KastB, 2015-07-19:
I think this project has a future for several reasons:
1) the second fan will probably never be controlled by the kernel natively
2) by now I had no time to test the thermald configs, but that could be something valuable, too
3) it takes years till everyone has a kernel, that can even control the 1st fan (probably 4.2 can, but it's not even released), and there are just very few people, that like to compile a whole kernel just for controlling fans.

Tharre, 2015-07-19:
> yes! had it running with hardcoded two fans setting today ;) check this pull request https://github.com/KastB/asus_wmi/pull/3 especially the comment...

I'm a bit confused right now, what good does hardcoding two fans
setting? `asus_wmi` shouldn't be able to control the GPU fan regardless?

daringer, 2015-07-19:
Not full functional, but "hard-coded" (the number) 2 fans and fixed some minor stuff, I got the impression that some stuff must be duplicated, but can't see a limitation why a second fan shouldn't work. KastB has most likely a more founded opinion. Is it not maybe not accessible through the acpi-through-wmi-interface. 

Maybe something general concerning this wmi-stuff: 
- is it not possible to address, thus read/write, *any* acpi-address? 
- Windows Management Instrumentation (WMI) (omg!) haven't read the full name yet 
- is mostly embedded into ACPI? right? Could really mean the necessary parts are either not exported (from ACPI into WMI?) or hidden?

KastB, 2015-07-20:
Tharre is right - we won't be able to control the GPU fan using WMI (it's not accessable)

It's not possible to call any ACPI method using WMI.
WMI is like a special method - you can call this method with different variables thus different submethods are executed - but it is just a specific set, that is executable.

closed 2016-11-05 by daringer