daringer/asus-fan #15: depmod -a required before activating module

phihos, 2015-04-02:
Hi,

I needed to run
```
sudo depmod -a
```
before running 
```
sudo modprobe asus_fan
```

Without it, modprobe does not find the module. This may be important to add to your readme, because for noobs like me this is not self-explanatory.

Tharre, 2015-04-03:
The Makefile should run depmod for you, actually. But you're right, I just tried it out myself, even though it says DEPMOD  <kernel version> it doesn't work.

berthinsandino, 2015-04-04:
Hi, I tried to install it according to the instructions provided in the README file, however I got this:
```
Can't read private key
  DEPMOD  3.16.0-33-generic
```
Then, I tried again, but this time using depmod, and same thing, "cant read private key ...". 
I am not sure, but t seems that it was installed correctly (or not?)
```
$ modinfo asus_fan
filename:       /lib/modules/3.16.0-33-generic/extra/asus_fan.ko
license:        GPL
description:    ASUS fan driver (ACPI)
author:         Bernd Kast <kastbernd@gmx.de>
author:         Markus Meissner <coder@safemailbox.de>
author:         Felipe Contreras <felipe.contreras@gmail.com>
srcversion:     C4C084DC14F221A4116468D
depends:        
vermagic:       3.16.0-33-generic SMP mod_unload modversions 
```
Also, I tempted to run "asus_fan" and I received this message:
```
$ sudo modprobe asus_fan
modprobe: ERROR: could not insert 'asus_fan': No such device
```

Any suggestions?
Thanks!

frederickjh, 2015-04-04:
I also saw the <b>Can't read private key</b> message that the other posted mentioned. Looks like the post was removed due to private contact information in it.

I saw themessage in the terminal after I ran <pre>sudo make install</pre>

phihos, 2015-04-04:
I can help you with that one:

This *Can't read private key* message is normal and simply means, that you are no official kernel developer. SO you can simply ignore it.

Tharre, 2015-04-04:
> Looks like the post was removed due to private contact information in it.

He must have removed the post himself, the contact information of the authors is public and can be found in asus_fan.c.

In any case I'm still going to address his issue, maybe he reads it: the module was correctly installed, however your notebook seems to be unsupported. This could still mean that this driver is going to work on your system, take a look at #2 to make it work. If you need any help just open a new issue with the modelname of your notebook, always happy to see the driver working on other systems.

berthinsandino, 2015-04-04:
Hello,
I removed my post because I found the solution on #14 (my notebook is ASUS N550JV, so I added the model to that if-clause). Seems that all is working now, but I dont know how to control the fan speed. The README says 
```echo 123 > ${fpath}/pwmX```

what does ${fpath} suppose to be?

I have tried to see all files in order to find something similar, but they are a lot of them.
```
$ sudo find / -iname 'pwm*'
/sys/devices/platform/asus-nb-wmi/hwmon/hwmon2/pwm1
/sys/devices/platform/asus_fan/hwmon/hwmon3/pwm1
/sys/devices/platform/asus_fan/hwmon/hwmon3/pwm1_enable
/sys/class/pwm
/sys/kernel/debug/pwm
/lib/modules/3.16.0-33-generic/kernel/drivers/pwm
/lib/modules/3.16.0-33-generic/kernel/drivers/pwm/pwm-lpss.ko
/lib/modules/3.16.0-33-generic/kernel/drivers/pwm/pwm-twl.ko
/lib/modules/3.16.0-33-generic/kernel/drivers/pwm/pwm-twl-led.ko
/lib/modules/3.16.0-33-generic/kernel/drivers/pwm/pwm-lp3943.ko
/lib/modules/3.16.0-33-generic/kernel/drivers/input/misc/pwm-beeper.ko
/lib/modules/3.16.0-33-generic/kernel/drivers/video/backlight/pwm_bl.ko
/lib/modules/3.16.0-30-generic/kernel/drivers/pwm
/lib/modules/3.16.0-30-generic/kernel/drivers/pwm/pwm-lpss.ko
/lib/modules/3.16.0-30-generic/kernel/drivers/pwm/pwm-twl.ko
/lib/modules/3.16.0-30-generic/kernel/drivers/pwm/pwm-twl-led.ko
/lib/modules/3.16.0-30-generic/kernel/drivers/pwm/pwm-lp3943.ko
/lib/modules/3.16.0-30-generic/kernel/drivers/input/misc/pwm-beeper.ko
/lib/modules/3.16.0-30-generic/kernel/drivers/video/backlight/pwm_bl.ko
/usr/src/linux-headers-3.16.0-30/arch/arm/plat-samsung/include/plat/pwm-core.h
/usr/src/linux-headers-3.16.0-30/include/linux/pwm_backlight.h
/usr/src/linux-headers-3.16.0-30/include/linux/pwm.h
/usr/src/linux-headers-3.16.0-30/include/dt-bindings/pwm
/usr/src/linux-headers-3.16.0-30/include/dt-bindings/pwm/pwm.h
/usr/src/linux-headers-3.16.0-30/drivers/pwm
/usr/src/linux-headers-3.16.0-33-generic/include/config/backlight/pwm.h
/usr/src/linux-headers-3.16.0-33-generic/include/config/pwm
/usr/src/linux-headers-3.16.0-33-generic/include/config/leds/pwm.h
/usr/src/linux-headers-3.16.0-33-generic/include/config/pwm.h
/usr/src/linux-headers-3.16.0-33-generic/include/config/input/pwm
/usr/src/linux-headers-3.16.0-33/arch/arm/plat-samsung/include/plat/pwm-core.h
/usr/src/linux-headers-3.16.0-33/include/linux/pwm_backlight.h
/usr/src/linux-headers-3.16.0-33/include/linux/pwm.h
/usr/src/linux-headers-3.16.0-33/include/dt-bindings/pwm
/usr/src/linux-headers-3.16.0-33/include/dt-bindings/pwm/pwm.h
/usr/src/linux-headers-3.16.0-33/drivers/pwm
/usr/src/linux-headers-3.16.0-30-generic/include/config/backlight/pwm.h
/usr/src/linux-headers-3.16.0-30-generic/include/config/pwm
/usr/src/linux-headers-3.16.0-30-generic/include/config/leds/pwm.h
/usr/src/linux-headers-3.16.0-30-generic/include/config/pwm.h
/usr/src/linux-headers-3.16.0-30-generic/include/config/input/pwm
```

Tharre, 2015-04-04:
It should be `/sys/class/hwmon/hwmonX/pwmX`, however controlling the fan speed via thermald or something like that is probably the better idea.

frederickjh, 2015-04-04:
I too found that setting of the fan speed with <pre>echo 123 > ${fpath}/pwmX</pre>  did not work. Here is a sample terminal session.

<pre>$ cat /sys/class/hwmon/hwmon2/pwm1
171
$ echo 180 > /sys/class/hwmon/hwmon2/pwm1
bash: /sys/class/hwmon/hwmon2/pwm1: Permission denied
$ sudo echo 180 > /sys/class/hwmon/hwmon2/pwm1
bash: /sys/class/hwmon/hwmon2/pwm1: Permission denied
$</pre> 

I am able to read the speed but even as root I cannot set it. I installed <a href="https://01.org/linux-thermal-daemon/documentation/introduction-thermal-daemon">thermald</a> and this works very well to control the thermal condition of my ultrabook along with asus-fan. I left the default settings for thermald. It just works. I can even watch youtube videos full screen without lockups during the video playback that I had before.

Tharre, 2015-04-04:
@frederickjh take a look at my comment on #10.

And we should stop polluting this issue.

daringer, 2015-05-07:
to return to the topic :) 
depmod is also not executed correctly for me, until this is fully fixed I'll added a `depmod -a` inside the Makefile, dirty but working for me...

daringer, 2015-06-21:
Let's call this fixed for now...

closed 2015-06-21 by daringer