daringer/asus-fan #17: Installed, how do I get fans to run more and keep laptop cool ?

stuaxo, 2015-06-16:
I've got an N56VZ ... installing this, I now can see the asus_fan RPM in xsensors .... any idea how I get this to run more, the laptop constantly runs hot (CPUs are all currently 68-70c)

Realise this is probably more a configuration issue than asus_fan ... but avoiding overheating might be the best thing thats happened me with this laptop !

frederickjh, 2015-06-16:
I installed thermald. The default settings work great.

KastB, 2015-06-16:
Hey,
that's strange - I guess there's something wrong then. My CPU has 50°C right now, and the fan runs at 1800RPMs (controlled by the firmware N551JK which seems to have a similar hardware and chassis than yours). Perhaps:
1) your thermal grease is old and could/should be renewed
2) your fan should be cleaned
3) something uses your CPU
4) graphics switching doesn't work

1, 2 are quite hard to solve, as it's tricky (but not impossible) to disassemble your notebook.
3, 4 could be checked by reading your power consumption (unplug AC and run powertop ) which shouldn't exceed 20W, I  think. Or call "cat /proc/acpi/bbswitch" which should report OFF

To speed up the fans you can use fancontrol. Nevertheless it was controlling the fan worse than the firmware in my case (very frequent and wast speed changes) thus I added a hystereses and damping to that script:https://github.com/KastB/fancontrol

frederickjh, 2015-06-16:
Actually if the unit overheated it will melt the heatsink grease like  
butter and it will flow out and leave a gap between the CPU and the  
heatsink. You will need to clean up the heatsink grease and replace it.

The ultrabooks are very easy to open and do this. A few screws Trox 6, and  
disconnect the battery and you are just about there.  Use rubbing alcohol  
and q-tips to clean up the grease, to get the majority you can use a paper  
towel.

Artic Silver heat sink grease is the best and what I use. Use very little  
part of a drop put on the heat sink fully take it off. Does the heat sink  
grease cover everything good put your heat sink back on and assemble. If  
not add a little more and repeat.

Been there done that!

stuaxo, 2015-06-17:
I don't know for sure it was overheating, but randomly turning off sometimes several times a day .. seems consistent, though could be a loose connection somewhere though ?

Possibly I do need to have a look inside - and get thermal grease re-applied, currently it looks like fan rpm 2420, core temperatures are all a bit more reasonable at around 55-60... watts is indeed about 20.

EDIT:  Just tried an HD youtube video and I get 56 Watts, 4800RPM and core temp 97c  ... so it seems video is the main thing making it heat up... at least the fan is responding [/EDIT]

N55 seems to have an i5, which might just not get as hot as the i7 in the N56 ... maybe having an i7 in a laptop isn't the best idea anyway ?

Graphics switching is not working, since I'm using Nouveau .. it basically only uses the intel graphics right now (have spent too long in the past staring at the terminal trying to fix it when the nvidia driver stops working).

Yup bbswitch says 'no', powertop showed a number of the tuneables as 'bad'

CPU Usage:  It seems to often shutdown when running chrome, so probably is related.

Googling: ```n56 randomly shuts down``` it seems to happen to people under windows too, so might be a design flaw....  I guess it's off to find some thermal paste...

frederickjh, 2015-06-17:
I have a Ultrabook with a i7 and it tends to run hotter that the 50  
degrees that KastB said his is at. Mine UX31À tends to run in the upper  
50s and lower 60.

Youtube video will get the temperature up, especially if I increase the  
size to full screen. Then  temperatures tend to run around 70 degrees and  
the fan speed hovers around 4000 rpm

If you are experiencing shut downs then you probable also have what seem  
to be lock-up but are thermal throttling. You can watch for the messages  
about thermal throttling in the terminal. I am guessing they are logged  
somewhere too.

Ctrl+Alt+F1 will take you to a terminal to get back to the GUI  
Ctrl+Alt+F7. I did not need to log in to see these message about thermal  
trotting just look in the terminal.

stuaxo, 2015-06-17:
Cool, will certainly see if I can see any messages - I guess they should appear in dmesg... will have a look for these next time it happens.

It hasn't shutdown in the last day, this might be as I've just reinstalled ubuntu, and installed asus_fan, thermald and a bunch of other stuff...  haven't watched any films, yet.

I have been running cpu intensive tasks and the fan speed does seem to change, which I'm fairly sure it wasn't before.. will try a film and see if it gets to the end.

KastB, 2015-06-17:
You could use this driver to increase the fan speed, as the firmware won't (at least on mine) run them on full speed, and react slowly on temperature changes.

The rest of the text semms to be kind of off-topic for this driver:

The core problem seems to be your high CPU temperature of 97°C which is quite hot - at 105°C it will switch off (http://ark.intel.com/de/products/64899/Intel-Core-i7-3610QM-Processor-6M-Cache-up-to-3_30-GHz)
This could be solved by removing the dusk and apply a good thermal paste. Asus is said to use a poor one, that is dry after a few months. I don't know how often you did that, but keep in mind, that the most frequent mistake is to apply too much grease - it should only be a thin film and the rest should flow out like butter ;) and this notebook is quite tricky to disassemble, as you have to disconnect some wires with only little space for your fingers: https://www.youtube.com/watch?v=qTYA897wPto
Other notebooks are really easy, but I'm afraid this one isn't.

I don't have a N55XX but a N550XX with an i7. Additionally we should compare the TDP which is 45W on yours and 47W on mine (thus comparable) and probalby 17W on a completely different chassis on frederick's (thus not comparable) to keep things short - your temperatures are to high on both, idle and under load.

Additionally to that it seems like you are not using a hardware decoder, thus your CPU is highly demanded when watching youtube. You could try using HTML5 or activating the hardware acceleration in flash.

stuaxo, 2015-06-17:
Cheers - that's a lot of useful info to go on ... good to know it can be cooler if I get thermal grease sorted.

Playing with hw decoding as we speak.

Will shut this now as I've mostly taken it off topic, thanks :)

closed 2015-06-17 by stuaxo