daringer/asus-fan #18: thermal_daemon configuration

daringer, 2015-06-21:
Anyone interested in a [thermal_daemon](https://github.com/01org/thermal_daemon) configuration file making use of the hwmon interface exposed by asus-fan? try https://github.com/daringer/asus-fan/tree/master/misc/thermald .... You'll also need https://github.com/daringer/asus-fan/tree/master/misc/create_symlinks.sh .... For me it works quite well, any feedback/additions would be highly appreciated...

daringer, 2015-07-19:
Bump! Is nobody interested using asus-fan together with thermald in order to trade 
- heat vs. performance
- battery life vs. performance

Intentionally choose to have:
- high performance and a noisy/loud fan
- low performance and a quiet fan

Those are my typical needs while using the laptop - Sure I am not alone?! Thermald seems to me _the_ only choice to have all this including the utilization of the Intel MSR API (powerclamp, intel_pstate, intel_rapl)?

Tharre, 2015-07-19:
You're definitely not alone in wanting to control how your fans are operating.

Personally I haven't been using thermald very much because I dislike the idea of having a userspace program control my fans (what if it crashes? what if I mess something up in custom kernel builds?).
That, and that it's not included in the official packages of my distribution have kept me from using it so far.

Also, I just tried to use your configuration, but it seems like asus_fan is broken for me currently, I'll open up a separate issue if this proves to be a problem with the module, rather than my setup.

daringer, 2015-07-19:
yes, especially the crash could be pretty bad, although Systemd knows `Restart=always`. Despite this, I really like that thermald handles this intel-cpu MSR vodoo, which seem to be at least as useful for heat reduction as fan(s)---at last a question of personal preference I believe. 

I just updated the configuration once more, maybe give it another try, I admit it's a little painful, especially as I still haven't seen the *right* way to maintain a persistent hwmon-device labeling across reboots---see  https://github.com/daringer/asus-fan/blob/master/misc/create_symlinks.sh for my workaround for this---combined with *both* thermald-config files is the setup I currently use.

Tharre, 2015-07-19:
systemd's `Restart=always` is very nice indeed, but fails in situations were thermald cannot be restarted (i.e. because of configuration problems after an upgrade).

I'll try it, this MSR stuff seems interesting. The problem I'm facing right now is that `asus_fan` crashes, I'll open up a seperate issue.

daringer, 2015-08-08:
now the gpu temperature sensor is considered for the gfx-fan speed control. The current thermald-config.xml (https://github.com/daringer/asus-fan/tree/master/misc/thermald) is considered working and secure, while being tuned for max performance, thus maxing out the fans, if necessary. 

PID-values are not optimal---fans tend to be too jumpy in some load-scenarios.

Currently I don't care for jumpy fans, as this config enables the laptop to keep performing under maximum load permanently without, or just minimal, forced throttling of the CPUs.

frederickjh, 2015-08-12:
@daringer I have been using thermald for some time now. The default configuration was working fine I reported this in another [issue](https://github.com/daringer/asus-fan/issues/10) on April 4.

What are the advantages / disadvantages to the thermald XML files you have added?

Is there anything that one needs to configure for thermald to use these?  If nothing special needs to be done other than the normal install, then I am already using them as each time the kernel changes I need to do a fresh install of asus-fan on my ultrabook.

In any case I am always monitoring with psensor to see if the fan speed is reported. If so asus-fan is up and running. :)

frederickjh, 2015-08-18:
@daringer  Ok, I have done a diff of the thermald files and see the differences to the stock configuration files.  Do I understand it correctly that each time I reboot must run the create_symlinks.sh for this configuration to work in thermald as the configuration files use the symlinks.

Too bad that thermald does not allow for wildcards in path names to devices in the configuration files.

frederickjh, 2015-08-18:
OK, looks like I am up and running with these configs.  I have added code to **/etc/init.d/thermald** to run the  **create_symlinks.sh** that I placed in **/usr/local/sbin/**  The bold section below is what I added. It must be added before the line **set -e**

Not sure if this is something you can incorporate into the project, maybe just in the notes.

===================================================================
`###` END INIT INFO

**# Create our needed syslinks to fan and temperature monitors,
  # as needed, as they will not exist after a reboot.
if [ ! -d  /tmp/asus-fan-shm ]; then
/usr/local/sbin/create_symlinks.sh
echo " * creating symlinks ..."
fi**

set -e

THERMALD="/usr/sbin/thermald"

====================================================================

daringer, 2015-08-21:
yes, well that's obviously a better way then executing it by hand like I did ;) (not rebooting my laptop often) will add this to the project thanks....

daringer, 2015-08-22:
mmmh, to actually do it "right" I would say we should maybe add a systemd and init.d script which make sure that the module is loaded and then creates all sym-links by executing create_symlinks.sh---before thermald is started. see issue #24 for this

frederickjh, 2015-08-25:
Yes, the init.d script would be a good way you just need to make sure it runs before thermald.

closed 2016-11-05 by daringer