daringer/asus-fan #19: UX303LNB

siddolo, 2015-07-12:
Hi,
can you help me to get this to work on asus UX303LNB?
Bios version: UX303LNB.206
Kernel: 4.0.5-040005-generic

Thanks

KastB, 2015-07-12:
Yes

KastB, 2015-07-12:
For a more specific answer more information on your problems would be required.
(=> what is/is not working, which problems arise, quote the build errors/dmesg errors...)

siddolo, 2015-07-13:
Thanks KastB.

    # lsmod | grep asus
    asus_nb_wmi            24576  0 
    asus_wmi               28672  1 asus_nb_wmi
    sparse_keymap          16384  1 asus_wmi
    wmi                    20480  2 mxm_wmi,asus_wmi
    video                  28672  2 i915,asus_wmi

    # ls -l /sys/class/hwmon/
    total 0
    lrwxrwxrwx 1 root root 0 lug 13 11:47 hwmon0 -> ../../devices/virtual/hwmon/hwmon0
    lrwxrwxrwx 1 root root 0 lug 13 11:48 hwmon1 -> ../../devices/platform/coretemp.0/hwmon/hwmon1
    lrwxrwxrwx 1 root root 0 lug 13 11:48 hwmon2 -> ../../devices/platform/asus-nb-wmi/hwmon/hwmon2

    # ls /sys/class/hwmon/hwmon0/
    name  power  subsystem  temp1_crit  temp1_input  temp2_crit  temp2_input  temp3_crit  temp3_input  uevent

    # ls /sys/class/hwmon/hwmon1/
    device  power      temp1_crit        temp1_input  temp1_max   temp2_crit_alarm  temp2_label  temp3_crit        temp3_input  temp3_max
    name    subsystem  temp1_crit_alarm  temp1_label  temp2_crit  temp2_input       temp2_max    temp3_crit_alarm  temp3_label  uevent

    # ls /sys/class/hwmon/hwmon2/
    device  name  power  pwm1  subsystem  temp1_input  uevent

    # tee /sys/class/hwmon/hwmon2/pwm1 <<< 10
    tee: /sys/class/hwmon/hwmon2/pwm1: Permission denied
    10

    # modprobe asus_fan

    in dmesg:
    [  484.001381] asus-fan (init) - finished init

    # ls -l /sys/class/hwmon
    total 0
    lrwxrwxrwx 1 root root 0 lug 13 11:53 hwmon0 -> ../../devices/virtual/hwmon/hwmon0
    lrwxrwxrwx 1 root root 0 lug 13 11:53 hwmon1 -> ../../devices/platform/coretemp.0/hwmon/hwmon1
    lrwxrwxrwx 1 root root 0 lug 13 11:53 hwmon2 -> ../../devices/platform/asus-nb-wmi/hwmon/hwmon2
    lrwxrwxrwx 1 root root 0 lug 13 11:56 hwmon3 -> ../../devices/platform/asus_fan/hwmon/hwmon3

    # tee /sys/class/hwmon/hwmon3/pwm1 <<< 10
    10
    # tee /sys/class/hwmon/hwmon3/pwm2 <<< 10
    10

fan speed doesn't change, and no more line was logged in dmesg

    # lsmod | grep asus
    asus_fan               16384  0 
    asus_nb_wmi            24576  0 
    asus_wmi               28672  1 asus_nb_wmi
    sparse_keymap          16384  1 asus_wmi
    wmi                    20480  2 mxm_wmi,asus_wmi
    video                  28672  2 i915,asus_wmi

    # modinfo asus_fan
    filename:       /lib/modules/4.0.5-040005-generic/extra/asus_fan.ko
    license:        GPL
    description:    ASUS fan driver (ACPI)
    author:         Bernd Kast <kastbernd@gmx.de>
    author:         Markus Meissner <coder@safemailbox.de>
    author:         Felipe Contreras <felipe.contreras@gmail.com>
    srcversion:     405E620C3E39CB395F37B77
    depends:        
    vermagic:       4.0.5-040005-generic SMP mod_unload modversions

    # cat /etc/default/grub | grep LINUX_DEFAULT
    GRUB_CMDLINE_LINUX_DEFAULT="quiet splash acpi_osi= acpi_backlight=vendor resume=/dev/mapper/ubuntu--vg-swap_1"

KastB, 2015-07-13:
Thank you for your detailed report.
The module seems to recognize both fans correctly. 
You tried to set a very low speed - probably they should stop. I assume, that you tried a high value (200 for example) as well without any change?
To further dig into the problem it would be great if you could provide your dsdt, to check if the called ACPI functions are similar to those on a machine where the code is working (there are a lot of tutorials out there, that describe how to extract the dsdt)

siddolo, 2015-07-13:
Hi, thank you for your support :)
    # dmesg | grep DSDT
    [    0.000000] ACPI: DSDT 0x00000000CB6BD218 01938E (v02 _ASUS_ Notebook 01072009 INTL 20120913)

    # cat /sys/firmware/acpi/tables/DSDT > dsdt.dat
    # iasl -d dsdt.dat

You can see the dsdt.dsl here: http://nopaste.linux-dev.org/?629392
If that issue is related to ACPI, do  you think that the kernel parameter "acpi_osi=" can impact on this?

KastB, 2015-07-14:
The relevant methods are pretty much the same on your and on my machine.
I don't think, that this parameter will affect the module's behaviour as the ACPI methods don't call any _OSI related stuff. Nevertheless, please let me know if there is any change without this parameter.

We could call the acpi functions directly, to check if the module has a bug and doesn't report errors that are returned by ACPI
You need the "acpi_call" module
as root type:
`c(){ echo "$1" | sudo tee /proc/acpi/call >/dev/null && sudo cat /proc/acpi/call;echo;}`
which is a helper function, that executes the call given as a variable and prints the return value.

As a test we can read the fan speeds:
`c "\\_SB.PCI0.LPCB.EC0.TACH 0"` for the 1st fan
`c "\\_SB.PCI0.LPCB.EC0.TACH 1"` for the 2nd fan
something like `0x701lled` should be returned, which is the hexadecimal speed (0x701) and the leftover of `not called`. The speed probably changes with every call.

To set the fans' speed try:
`c "\\_SB.PCI0.LPCB.EC0.SFNV 0x1 0xff"
c "\\_SB.PCI0.LPCB.EC0.SFNV 0x2 0xff"`
the fans should run on full speed then and `0x0called` should be returned.

If you try to read the speeds again using
`c "\\_SB.PCI0.LPCB.EC0.TACH 0"
c "\\_SB.PCI0.LPCB.EC0.TACH 1"`
check if the still change (then something is different to our notebooks)

To reset the speeds call
`c "\\_SB.PCI0.LPCB.EC0.SFNV 0x0 0x0"

If any error occured during these calls we could try to fix that. Otherwise Asus probably removed the feature of controlling the fans using this ACPI methods. It will be pretty hard to control the fans, then.

siddolo, 2015-07-14:
Hi KastB,
thanks for your detailed answer.
All steps completed without error, here's the output:

    # c "\\_SB.PCI0.LPCB.EC0.TACH 0"
    0x1231led

    # c "\\_SB.PCI0.LPCB.EC0.TACH 1"
    0x1bfbled

    # c "\\_SB.PCI0.LPCB.EC0.SFNV 0x1 0xff"; c "\\_SB.PCI0.LPCB.EC0.SFNV 0x2 0xff"
    0x0called
    0x0called

    # c "\\_SB.PCI0.LPCB.EC0.TACH 0"; c "\\_SB.PCI0.LPCB.EC0.TACH 1"
    0x1227led
    0x1be3led

    # c "\\_SB.PCI0.LPCB.EC0.SFNV 0x0 0x0"; c "\\_SB.PCI0.LPCB.EC0.SFNV 0x0 0x0"
    0x0called
    0x0called

    # c "\\_SB.PCI0.LPCB.EC0.TACH 0"; c "\\_SB.PCI0.LPCB.EC0.TACH 1"
    0x1227led
    0x1bfbled

I tryed to set fan speeds by passing some different hex value as argument (instead of 0xff), but i "heard" always the same speed.

KastB, 2015-07-14:
The reported fan speeds are quite high - especially for the second fan nevertheless I can't find any mistakes that we did. 
Probalby we won't get that module working unless a bios update restores the old behaviour. The Interface still exists, but probably reports wrong fan speeds and obviously doesn't change the speeds.
The post on which this module is based on (http://forum.notebookreview.com/threads/fan-control-on-asus-prime-ux31-ux31a-ux32a-ux32vd.705656/) mentions a second way of controlling the fan, by writing directly to registers. Nevertheless chances are low, that this is still working while the interface, that should be more stable, stopped working. It's curious that the ACPI methods kept the same - even those that can be accessed through the WMI (MF13->SFNS).
I'm afraid I currently can't help you. If the behaviour changes for some reason, for example due to a bios update, or changed kernel parameters (that should have no influence on this behaviour), I'll try to help you getting that working.

siddolo, 2015-07-15:
Hi KastB.
I'll let you know if i have some new stuff.
Thanks a lot for putting me on the right way, and especially for your time :)

siddolo, 2015-07-16:
Hi KastB,
I think I made a mistake in my previous test.
After a reboot your asus_fan module seems to work correctly!
The difference from the first try, is that the module exposes only pwm1 interface.

KastB, 2015-07-17:
Hi,

is it working all the time now? Or does it fail again for example after suspending.
Do you know how many fans your notebook has? I asumed that there are two because of the dedicated graphics card.
Can you control or at least read out the speed of the 2nd fan with the manual method described in my previous post. It would be ok of there is no pwm2 as long as you can only read the 1st fans speed

daringer, 2015-07-19:
Added 'misc/test_module.sh' to have an automated test. Clears dmesg, reload module if needed to get a fresh environment. Uses sudo for root commands, please read file header, if you have any doubts running this script with partly root-rights.

My output looks like this for UX32VD:

```
|-$ misc/test_module.sh 
[i] ------------ Starting 'asus-fan' kernel module quick test
[+] asus-fan loaded! REMOVING (rmmod) asus-fan (asus_fan) module
[sudo] password for dariball: 
[i] Clearing 'dmesg' (kernel) log
[i] Trying to modprobe the module: 'asus_fan'
[+++] Successfully inserted module: 'asus_fan'

[i] ------------ Inspecting dmesg a.k.a. kernel log
[+++] 'dmesg' reports finished module init

[i] ------------ Search /sys files
[i] Checking for fan1 /sys/ files:
[+] Found /sys path (/sys/devices/platform/asus_fan)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon84)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon84/fan1_input)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon84/fan1_label)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon84/fan1_min)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon84/fan1_speed_max)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon84/pwm1)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon84/pwm1_enable)
[i] Checking for fan2 /sys/ files:
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon84/fan2_input)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon84/fan2_label)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon84/fan2_min)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon84/pwm2)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon84/pwm2_enable)
[+++] All /sys files found for primary fan (no: 1)
[+++] All /sys files found for secondary fan (no: 2)
[i] Looks like this machine has two fans!

[i] ------------ Test functionality
[i] Checking fan1 functionality! (label: 'CPU Fan')
[+++] Reading fan speed: 3743
[+++] fan1 speed is updated (changed)
[+] Change found in try no: 6!
[+] Could read and verify (default) activated auto-mode: 0
[+] Successfully read the min speed for fan1: 10
[i] Set fan speed manually starting with fan1_min (10) using 15 as stepping
[i] You should hear/feel the changing fan1 speed!
[i] Set manual speed to: 15 30 45 60 75 90 105 120 135 150 
                         165 180 195 210 225 240 255 
[+++] Getting speed during manual mode for fan1 successfully done
[+++] Setting manual speeds for fan1 successful!
[+] Auto-mode (/sys/devices/platform/asus_fan/hwmon/hwmon84/pwm1_enable) flag behaves as intended for fan1!
[+] Auto-mode (/sys/devices/platform/asus_fan/hwmon/hwmon84/pwm1_enable) was successfully reseted!
[i] Checking fan2 functionality! (label: 'GFX Fan')
[+++] Reading fan speed: 3776
[+++] fan2 speed is updated (changed)
[+] Change found in try no: 2!
[+] Could read and verify (default) activated auto-mode: 0
[+] Successfully read the min speed for fan2: 10
[i] Set fan speed manually starting with fan2_min (10) using 15 as stepping
[i] You should hear/feel the changing fan2 speed!
[i] Set manual speed to: 15 30 45 60 75 90 105 120 135 150 
                         165 180 195 210 225 240 255 
[+++] Getting speed during manual mode for fan2 successfully done
[+++] Setting manual speeds for fan2 successful!
[+] Auto-mode (/sys/devices/platform/asus_fan/hwmon/hwmon84/pwm2_enable) flag behaves as intended for fan2!
[+] Auto-mode (/sys/devices/platform/asus_fan/hwmon/hwmon84/pwm2_enable) was successfully reseted!
[i] ------------
[i] ------------ Final result overview:
[RESULT]                Houskeeping  ->  GOOD
[RESULT]           DMESG inspection  ->  GOOD
[RESULT]       /sys files available  ->  GOOD
[RESULT]        Functionality Fan 1  ->  GOOD
[RESULT]        Functionality Fan 2  ->  GOOD
```

closed 2016-11-05 by daringer