daringer/asus-fan #20: would a asus-fan module be instantly denied?

KastB, 2015-07-12:
To answer the question of the readme:
I really think so. Felipe Contrera first tried a non WMI-using module to be pushed to the kernel which seemd similar to this one and failed as calling ACPI functions directly is very specific while the WMI interface is regarded stable for different models.
Llinux 4.2 will at least be able to control the first fan using the wmi interface out of the box, by doing "minor" changes to an existing module.
"asus-fan" would need additional files, to build an additional module, that is activated on a whitelisted set of machines. Seems very hard to get that to the kernel.

Tharre, 2015-07-12:
I agree, it would be instantly denied.

daringer, 2015-07-19:
So I guess it's time to focus on asus-wmi :D see #13 .... reopen ... and direct close!

closed 2015-07-19 by daringer