daringer/asus-fan #21: Module crashes upon loading in UX32VD

Tharre, 2015-07-19:
Here is the `dmesg` output:

```
[  134.787394] WARNING: CPU: 3 PID: 2457 at fs/sysfs/group.c:61 internal_create_group+0x22c/0x2c0()
[  134.787395] Attribute fan2_label: Invalid permissions 01
[  134.787396] Modules linked in: asus_fan(O+) fuse joydev mousedev bbswitch(O) nls_iso8859_1 nls_cp437 vfat fat rtsx_usb_ms memstick btusb btbcm btintel bluetooth asus_nb_wmi asus_wmi sparse_keymap mxm_wmi iTCO_wdt iTCO_vendor_support arc4 intel_rapl iosf_mbi x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm iwldvm snd_hda_codec_hdmi mac80211 psmouse pcspkr serio_raw snd_hda_codec_realtek iwlwifi snd_hda_codec_generic cfg80211 snd_hda_intel snd_hda_controller rfkill thermal snd_hda_codec fan evdev int3403_thermal mac_hid tpm_tis snd_hda_core tpm snd_hwdep wmi snd_pcm int3402_thermal ac battery int340x_thermal_zone snd_timer snd int3400_thermal mei_me acpi_thermal_rel mei i2c_i801 soundcore lpc_ich shpchp processor sch_fq_codel vboxnetflt(O) vboxdrv(O) ip_tables x_tables ext4 crc16 mbcache
[  134.787420]  jbd2 algif_skcipher af_alg rtsx_usb_sdmmc led_class mmc_core rtsx_usb dm_crypt dm_mod sd_mod atkbd libps2 crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel aesni_intel aes_x86_64 lrw gf128mul glue_helper ablk_helper cryptd xhci_pci xhci_hcd ahci libahci libata ehci_pci ehci_hcd scsi_mod usbcore usb_common i8042 serio i915 button intel_gtt i2c_algo_bit video drm_kms_helper drm i2c_core
[  134.787434] CPU: 3 PID: 2457 Comm: modprobe Tainted: G     U  W  O    4.1.2-2-ARCH #1
[  134.787435] Hardware name: ASUSTeK COMPUTER INC. UX32VD/UX32VD, BIOS UX32VD.214 01/29/2013
[  134.787435]  0000000000000000 00000000ec00c6fd ffff88003832f7d8 ffffffff81585c8e
[  134.787437]  0000000000000000 ffff88003832f830 ffff88003832f818 ffffffff81078c9a
[  134.787439]  ffffffff8162b360 000000000000000a ffff8800c8c94e88 0000000000000000
[  134.787440] Call Trace:
[  134.787442]  [<ffffffff81585c8e>] dump_stack+0x4c/0x6e
[  134.787443]  [<ffffffff81078c9a>] warn_slowpath_common+0x8a/0xc0
[  134.787445]  [<ffffffff81078d25>] warn_slowpath_fmt+0x55/0x70
[  134.787447]  [<ffffffff8125c5ec>] internal_create_group+0x22c/0x2c0
[  134.787449]  [<ffffffff8125c6e9>] sysfs_create_groups+0x49/0xa0
[  134.787451]  [<ffffffff813f115b>] device_add+0x67b/0x6c0
[  134.787453]  [<ffffffff813f11be>] device_register+0x1e/0x30
[  134.787454]  [<ffffffff81430c11>] hwmon_device_register_with_groups.part.0+0xc1/0x120
[  134.787456]  [<ffffffff81430ce1>] hwmon_device_register_with_groups+0x71/0x80
[  134.787458]  [<ffffffffa080d7a1>] asus_fan_probe+0x91/0x110 [asus_fan]
[  134.787460]  [<ffffffff813f67b4>] platform_drv_probe+0x34/0xa0
[  134.787462]  [<ffffffff813f4439>] driver_probe_device+0x209/0x4c0
[  134.787464]  [<ffffffff813f47cb>] __driver_attach+0x9b/0xa0
[  134.787466]  [<ffffffff813f4730>] ? __device_attach+0x40/0x40
[  134.787468]  [<ffffffff813f2193>] bus_for_each_dev+0x73/0xc0
[  134.787469]  [<ffffffff813f3cfe>] driver_attach+0x1e/0x20
[  134.787471]  [<ffffffff813f38c0>] bus_add_driver+0x180/0x250
[  134.787473]  [<ffffffff813f5074>] driver_register+0x64/0xf0
[  134.787475]  [<ffffffff813f66fa>] __platform_driver_register+0x4a/0x50
[  134.787477]  [<ffffffff813f6865>] __platform_driver_probe+0x25/0xc0
[  134.787479]  [<ffffffff813f6d95>] __platform_create_bundle+0xb5/0xe0
[  134.787481]  [<ffffffffa080d710>] ? asus_fan_remove+0x40/0x40 [asus_fan]
[  134.787483]  [<ffffffffa0013000>] ? 0xffffffffa0013000
[  134.787485]  [<ffffffffa080d9f5>] asus_fan_register_driver+0x55/0x70 [asus_fan]
[  134.787487]  [<ffffffffa00130a1>] fan_init+0xa1/0x1000 [asus_fan]
[  134.787489]  [<ffffffff81002128>] do_one_initcall+0xb8/0x200
[  134.787491]  [<ffffffff8158502e>] do_init_module+0x60/0x1cb
[  134.787492]  [<ffffffff81102a4a>] load_module+0x21aa/0x2800
[  134.787494]  [<ffffffff810fea20>] ? store_uevent+0x70/0x70
[  134.787496]  [<ffffffff81103205>] SyS_init_module+0x165/0x1b0
[  134.787499]  [<ffffffff8158b56e>] system_call_fastpath+0x12/0x71
[  134.787500] ---[ end trace c3a566a0fc6f7976 ]---
[  134.787534] asus-fan (init) - finished init
```

And of course `uname -a`

    Linux xultrabook 4.1.2-2-ARCH #1 SMP PREEMPT Wed Jul 15 08:30:32 UTC 2015 x86_64 GNU/Linux

daringer, 2015-07-19:
mmh, interessting! Had a 4.0.8-ck kernel running, which worked perfectly, now I switched to stock -ARCH and got exactly the same - reproduced! ;D

- the module is kind of loaded 
- `rmmod asus_fan` works -> `asus-fan (exit) - module unloaded - cleaning up...`
- and a loooot of upses.... see here: http://pastebin.com/2exisCQX
- the following is repeated every dump:
```
WARNING: CPU: 3 PID: 15972 at fs/sysfs/group.c:61 internal_create_group+0x22c/0x2c0()
Attribute fan2_label: Invalid permissions 01
```

Tharre, 2015-07-19:
It's not only stock -ARCH, but also linux-next from yesterday. The dumps are basically all the same, don't know why they happen so often.

Guess it's caused by https://github.com/torvalds/linux/commit/d8bf8c92e80fed9119eb222c7e5cc88acf57c12c

daringer, 2015-07-19:
inside `asus-fan.c` the first ups comes out here:
```
692   platform_device =                                                              
693       platform_create_bundle(platform_driver, asus_fan_probe, NULL, 0, NULL, 0);
```
And yes, the commit you've shown seems to be the reason, but the effect damn more than just a warning, but maybe because of the amount of added /sys-files ... playing around with modes now i guess

daringer, 2015-07-19:
Strange, but yes: https://github.com/torvalds/linux/commit/d8bf8c92e80fed9119eb222c7e5cc88acf57c12c is clearly the output shown. Each "warning" also has a changed (/sys)filename, but nothing I change seems to be reflected in it... its always "01" ... tried anything: just readable/writable, even simply 0664 (which already feels like trial&error ;D) .... same output every time---can't even recall or find another spot where the modes could be overridden ... evil Linus ;)

Tharre, 2015-07-19:
Cloning of the linux kernel is nearly finished ...

I really shouldn't cloned the hole damn thing.

daringer, 2015-07-19:
There are a lot of DEVICE_ATTR() lines in asus-fan, but they nearly don't differ from KastR/asus_wmi. https://github.com/daringer/asus-fan/blob/master/asus_fan.c#L693 is the line where it first happens, in fact there happens nearly nothing right before ;) really wondering, which mode value we are seeing in this error message `Invalid permissions 01` as I haven't managed to change this, yet. 

Yeeees, how about debugging ;D I should maybe learn some kernel debugging basics ... is this call-stack usable and delivers gdb-like info with the right translator? line-no? filename? etc...

Tharre, 2015-07-19:
Use `addr2line -e ~/linux/vmlinux ffffffff810737d6` to translate this stuff. Although I believe this only works if you compile with `CONFIG_DEBUG_INFO`.

That being said, I think I already know what's going on here, give me a few more minutes.

PR #22 (2015-07-19): Return correct value for is_visible attribute.
Fixes #21.
commit 24d64586 (2015-07-19): Return correct value for is_visible attribute.

Fixes #21.
closed 2015-07-19 by daringer