daringer/asus-fan #24: DKMS and build-scripts for various distributions

daringer, 2015-08-22:
As the title suggests, DKMS support should be pretty easy to add as I saw, further this should be incorporated with build scripts ... I can add a PKGBUILD for ArchLinux (maybe add this to AUR then?), Debian and any others would also be a nice-to-have if anyone can contribute I'll add it...

daringer, 2015-08-23:
ArchLinux: DKMS and PKGBUILD for "-git" based build looks good... systemd-support pending...

Tharre, 2015-08-23:
A few suggestions:
- I'm not sure if makedepending on linux-headers is a good idea, as it's specific to the standard arch linux kernel. All dkms packages I'm looking at just depend on dkms, which optionally depends on linux-headers and linux-lts-headers
- Just using the commit hash as pkgver is kind of bad, as neither pacman nor the user can tell which version is newer. I'd suggest using version.release.commit_id like shown [here](https://github.com/Tharre/AUR/blob/28ef1ecdd7e6c96249fdb488a3a017b4b46aa42d/st-custom/PKGBUILD#L5).
- Also, the package should only include files that are actually needed, no .git and all that stuff.

I hope that's helpful.

daringer, 2015-08-23:
shouldn't push all the time ;) but thanks a lot, some time passed since I wrote my last PKGBUILD ... looks way better now...

closed 2016-11-05 by daringer