daringer/asus-fan #25: GFX Fan not shown in "sensors" command after bootup

RKBK, 2015-10-17:
I've tried asus-fan on my UX32VD (Ubuntu 15.04). It builds ok, and loads up on boot after being added to /etc/modules. However, after restarting the computer, sometimes the GFX Fan numbers and name is not shown when running sensors. This is what it looks like currently, just after booting up and logging in:

```
acpitz-virtual-0
Adapter: Virtual device
temp1:        +53.0°C  (crit = +108.0°C)

asus_fan-isa-0000
Adapter: ISA adapter
CPU Fan:     3185 RPM  (min =   10 RPM)
gfx_temp:      +0.0°C  (crit =  +0.1°C)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +53.0°C  (high = +87.0°C, crit = +105.0°C)
Core 0:         +53.0°C  (high = +87.0°C, crit = +105.0°C)
Core 1:         +51.0°C  (high = +87.0°C, crit = +105.0°C)

asus-isa-0000
Adapter: ISA adapter
temp1:        +53.0°C
```
Unloading the module
sudo modprobe -r asus_fan
and then reloading the module with
sudo modprobe asus_fan
fixes the issue, and I then get:
````
acpitz-virtual-0
Adapter: Virtual device
temp1:        +53.0°C  (crit = +108.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +53.0°C  (high = +87.0°C, crit = +105.0°C)
Core 0:         +51.0°C  (high = +87.0°C, crit = +105.0°C)
Core 1:         +50.0°C  (high = +87.0°C, crit = +105.0°C)

asus-isa-0000
Adapter: ISA adapter
temp1:        +53.0°C  

asus_fan-isa-0000
Adapter: ISA adapter
CPU Fan:     3227 RPM  (min =   10 RPM)
GFX Fan:     3032 RPM  (min =   10 RPM)
gfx_temp:      +0.0°C  (crit =  +0.1°C)
````

There doesn't seem to be any errors in kern.log except (what I guess is standard):
````
Oct 18 00:44:37 UX32VD kernel: [    8.551159] asus_fan: module verification failed: signature and/or  required key missing - \
tainting kernel                                                                                                                      
Oct 18 00:44:37 UX32VD kernel: [    8.572756] asus-fan (init) - finished init 
````

Any ideas about what might be causing this?

Tharre, 2015-10-17:
The error message indicates that you're using a kernel compiled with the signed kernel module feature, but have not signed the asus-fan module. However, that should indeed not be the problem, as the module is loaded anyway.

But is the module actually loaded on boot every time, as in does it show up when you run `lsmod` directly after boot?

RKBK, 2015-10-18:
The module does seem to be loaded after booting, as even before logging in the module is shown with lsmod:

RKBK, 2015-10-18:
I wonder if the problem could be related to Bumblebee, which I am running to disable the discrete graphics card. It seems that asus-fan is loaded before bumblebee (/var/log/kern.log):
````
Oct 18 11:39:06 UX32VD kernel: [    8.575572] asus_fan: module verification failed: signature and/or  required key missing - tainting kernel                                                                  
Oct 18 11:39:06 UX32VD kernel: [    8.597079] asus-fan (init) - finished init 
```
and then later

````
Oct 18 11:39:06 UX32VD kernel: [   10.645411] bbswitch: version 0.7                                                                                                                                           
Oct 18 11:39:06 UX32VD kernel: [   10.645416] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0                                                                                             
Oct 18 11:39:06 UX32VD kernel: [   10.645421] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP                                                                                          
Oct 18 11:39:06 UX32VD kernel: [   10.645428] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)                          
Oct 18 11:39:06 UX32VD kernel: [   10.645503] bbswitch: detected an Optimus _DSM function                                                                                                                     
Oct 18 11:39:06 UX32VD kernel: [   10.645551] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on                                                                                                  
Oct 18 11:39:06 UX32VD kernel: [   10.647298] bbswitch: disabling discrete graphics                                                                                                                           
Oct 18 11:39:06 UX32VD kernel: [   10.647306] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20141107/nsarguments-95)
````

daringer, 2015-10-18:
hey RKBK, having similar problems, same model, I am guessing this is bbswitch related, currently recreating the symlinks on each boot (standby is my friend :)... but anyways, it most likely is related to the order in which the gfx-related modules are loaded, haven't found the right yet ... will first include the systemd snippet into the repository and the buildscript to have at least this out of the box, module loading on-boot is still on the agenda to be fixed... :/

saultdon, 2016-02-10:
I've noticed this happening randomly on an ASUS G75VX (NVIDIA GTX 670MX). Arch Linux 4.4.1.

I tried creating /etc/modprobe.d/asus-fan.conf with
`install nvidia /sbin/modprobe --ignore-install nvidia; /sbin/modprobe asus-fan;` trying to get the nvidia module to load before the asus-fan module to see if that was the issue.

But it still misses the gfx temp randomly and when that happens journalctl will output that the systemd-modules-load.service failed to start.

Sometimes a `modprobe -r asus-fan` and then a `modprobe asus-fan` to reload the module sometimes gets the gfx temp to show up.

daringer, 2016-11-05:
you should try the latest git sources, for the systemd.service was flawed and shall be fixed now...

daringer, 2017-04-22:
closing due to age, please feel free to re-open if needed

closed 2017-04-22 by daringer