daringer/asus-fan #26: Unable to preload create_symlinks.sh for thermald

RKBK, 2015-10-17:
Hi,
I've attempted to use the solution for thermald with asus-fan on my UX32VD. I tried to apply the solution of https://github.com/daringer/asus-fan/issues/18#issuecomment-132172869 (running create_symlinks.sh as a part of starting thermald), but it does not appear to have any effect on my computer (Ubuntu 15.04). Thermald does start, but there is no notice of creating symlinks, and the folder in /tmp/ is not created. 

Any suggestions?

Tharre, 2015-10-17:
I'm no expert on Ubuntu, but I think 15.04 is using systemd, so putting anything into `/etc/init.d/thermald` should've no effect on the thermald service.

You probably have to write a replacement unit file and place it under `/etc/systemd/system/thermald.d/`, that is executing the `create_symlinks.sh` script.

RKBK, 2015-10-18:
Thanks, Tharre!

That seems to be exactly my problem. I created a file called asus-fan.service in /etc/systemd/system , containing the following
````
[Unit]                                                                                        
Description=Create tmp files needed to operate asus-fan                                       
                                                                                              
[Service]                                                                                     
ExecStart=/usr/local/sbin/create_symlinks.sh                                                  
                                                                                              
[Install]                                                                                     
WantedBy=multi-user.target
````
and then ran
````
sudo systemctl enable asus-fan.service 
````

and now the asus-fan files are in /tmp/ upon bootup!

daringer, 2015-10-19:
seems to be fixed, at least for me ;) closing...

closed 2015-10-19 by daringer

RKBK, 2015-10-19:
Thanks, @daringer!