daringer/asus-fan #27: Different thermald configurations for running on battery power and on AC?

RKBK, 2015-10-19:
As far as I understand, the thermald configuration file included with asus-fan is optimized for performance, and kicks in the fans to reduce temperature rather than reducing the CPU performance. Is there any way to control thermald so that it switches configuration based on whether the laptop is running on battery or AC power? That way, the battery life might be maximized by not using the fans except if necessary. Right now, my UX32VD (using the default packaged configuration file for thermald, not the one from asus-fan) spins up the fans to high levels intermittently no matter if the laptop runs on battery or not, which I guess drains the battery.

There is some mention online of using dbus to control the thermald policy, but I don't really understand exactly how that is done. Anyway, I guess the way it is supposed to be done is to have one setup as "Quiet" in the conf file, and one as "performance" and then get DBUS to tell thermald to switch policy when plugging the cable in or out. 

Does anyone have any more ideas about how this might be done, exactly?

daringer, 2015-10-19:
You could use the a script provided by thermal_daemon (https://github.com/01org/thermal_daemon/blob/master/tools/thermald_set_pref.sh) so set the two modes, but any other dbus-connector will do it to ...

inside the thermald.conf you should configure QUIET and PERFORMANCE, simply reducing/increasing the appropriate values for the specific parts may then be done to tune your both presets...

further, to do this automatically you could use laptop-mode-tools to execute a script (i.e., to set the thermald preset) if the battery status changes...

ghost11111, 2015-11-07:
Hi,
could you explain how to use the "QUIET profile" in your thermald configuration file?
The thermald default one is a little bit too sensitive for me (fan is kicking on at 40°C).
Thanks!

daringer, 2015-11-21:
Interesting, for me the current ```thermal-conf.xml``` behaves pretty silent until CPUs get above 70°, feels even a little too late---but concerning your issue---usually either "quiet" or "performance" may be set inside the xml file through: ```<Preference>QUIET</Preference>```.
```man thermal-conf.xml``` may also be very helpful. Further, if you would like to stay quiet I would suggest to stick with the original ```thermal-conf.xml```.... maybe even disabling the manual (user-space) fan-control and stick with the automatic-mode, for lower-to-medium loads the automatic mode performs very good.

closed 2016-11-05 by daringer