daringer/asus-fan #29: lm_sensors cant read GFX temp as 'd' is appended, error 43000d: value too great for base

saultdon, 2016-02-10:
The /usr/sbin/fancontrol script from lm_sensors package has trouble reading the GFX temperature value in lines 537, 539, and 543.

The error for each of those lines is `((: 43000d: value too great for base (error token is "43000d")`.

It looks like the string is literally 43000d but maybe lm_sensors expects it to be parsed to 43000 (without the 'd').

I don't know if this is a problem with asus-fan or lm_sensors not formatting the string properly for the needed variable in the fancontrol script. If it's a bug report for lm_sensors, please close and I can file it there for upstream fix.

But it looks like the gfx_temp var is the only one with a 'd' appended to it:
```
saultdon@archasus % cat /tmp/asus-fan-shm/core1_temp
29000
saultdon@archasus % cat /tmp/asus-fan-shm/core2_temp
30000
saultdon@archasus % cat /tmp/asus-fan-shm/die_temp
36000
saultdon@archasus % cat /tmp/asus-fan-shm/gfx_temp
28000d <--
saultdon@archasus % cat /tmp/asus-fan-shm/tz1_temp
35000
saultdon@archasus % cat /tmp/asus-fan-shm/tz2_temp
```

Here it looks like it's the gfx_temp is the only value loaded from the asus-fan module:
```
saultdon@archasus % ls -lah /tmp/asus-fan-shm
total 0
drwxr-xr-x  2 root root 280 Feb 10 11:27 .
drwxrwxrwt 14 root root 360 Feb 10 11:40 ..
lrwxrwxrwx  1 root root  57 Feb 10 11:27 core1_temp -> /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp2_input
lrwxrwxrwx  1 root root  57 Feb 10 11:27 core2_temp -> /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp3_input
lrwxrwxrwx  1 root root  57 Feb 10 11:27 die_temp -> /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp1_input
lrwxrwxrwx  1 root root  55 Feb 10 11:27 fan_cpu_manual_mode -> /sys/devices/platform/asus_fan/hwmon/hwmon0/pwm1_enable
lrwxrwxrwx  1 root root  54 Feb 10 11:27 fan_cpu_rpm -> /sys/devices/platform/asus_fan/hwmon/hwmon0/fan1_input
lrwxrwxrwx  1 root root  48 Feb 10 11:27 fan_cpu_speed -> /sys/devices/platform/asus_fan/hwmon/hwmon0/pwm1
lrwxrwxrwx  1 root root  55 Feb 10 11:27 fan_gfx_manual_mode -> /sys/devices/platform/asus_fan/hwmon/hwmon0/pwm2_enable
lrwxrwxrwx  1 root root  54 Feb 10 11:27 fan_gfx_rpm -> /sys/devices/platform/asus_fan/hwmon/hwmon0/fan2_input
lrwxrwxrwx  1 root root  48 Feb 10 11:27 fan_gfx_speed -> /sys/devices/platform/asus_fan/hwmon/hwmon0/pwm2
lrwxrwxrwx  1 root root  55 Feb 10 11:27 gfx_temp -> /sys/devices/platform/asus_fan/hwmon/hwmon0/temp1_input
lrwxrwxrwx  1 root root  45 Feb 10 11:27 tz1_temp -> /sys/devices/virtual/hwmon/hwmon1/temp1_input
lrwxrwxrwx  1 root root  58 Feb 10 11:27 tz2_temp -> /sys/devices/platform/asus-nb-wmi/hwmon/hwmon3/temp1_input
```

I'm running lm_sensors 3.4.0 and asus-fan-dkms-git 0.9.1 (r6.g2472ad6).

gluwer, 2016-07-03:
Hi,

I had the same problem. It looks there is a typo in sprintf and removing d (as u already formats everything correctly), so I changed the line 583 from:

    return sprintf(buf, "%llud\n", value*1000);

to:

    return sprintf(buf, "%llu\n", value*1000);

After the change everything works just fine.

daringer, 2016-11-05:
fixed, thx

closed 2016-11-05 by daringer