daringer/asus-fan #30: UX32LN - module not working

tomek-sztuk, 2016-02-22:
Hi, I have a strange problem - the module reports the speeds but can't control the fans - the are spinning at the top speed no matter what... here are the results of a test: http://pastebin.com/VARPVB3q - although it says all is well, I couldn't hear the fans slowing down.

frederickjh, 2016-02-24:
Hi @tomek-sztuk !

I have a Asus UX31A and I have notice something similar. After a kernel upgrade I run my script to clone the repo and then install the module. It also restarts thermald. Sometimes after this the fan will run at top speed until I reboot. After that it is fine. 

I also find that restarting thermald can clear a fan stuck on high speed. Some times that seems to happen to me but not often enough. Normally I just have too much stuff running.
God Bless!
Frederick

daringer, 2016-11-05:
second that, as of today still no permanent fix available, only think that helps is restarting asus-fan.service and thermald.service ...

daringer, 2016-11-05:
@tomek-sztuk please check if the latest git sources fix it

tomek-sztuk, 2016-11-06:
@daringer unfortunately I can't check it, don't own the machine anymore.

closed 2016-11-06 by daringer