daringer/asus-fan #31: UX31A "modprobe: ERROR: could not insert 'asus_fan': No such device"

hystryfe, 2016-03-02:
Hello,

I have a Zenbook UX31A with Arch Linux x86-64. Recently the fan stopped working although I didn't notice until I almost burned myself! As luck and timing would have it, such a wonderful project is available. So I pulled your asus_fan module in from the AUR which I believe comes straight from git:

https://aur.archlinux.org/packages/asus-fan-dkms-git/

The module compiles fine. When I try to insert the module, I get the error seen in the title:

`modprobe: ERROR: could not insert 'asus_fan': No such device`

I can provide any information needed just ask.

frederickjh, 2016-03-02:
Hi!
I have a ux303 running Ubuntu 14.04.4
Here is the script that I use after kernel updates to recompile. Maybe the module does not match your kernel. Maybe try and build from source. Sometimes my fan will stick on high after the initial rebuild after a new kernel. A reboot solves that.

### asus-fan-install.sh
```
#!/bin/bash
# This is a script to setup the asus-fan modules from github. https://github.com/daringer/asus-fan - Frederick Henderson
PS4=':${LINENO} + '
#set -x

# Console colors
red='\033[0;31m'
green='\033[0;32m'
yellow='\033[1;33m'
NC='\033[0m'

# function to run commands as super user. This will keep give the user
# option to re-enter the password till they get it right or allow them
# to exit if something goes wrong instead of continuing on. 
# Usage:
#  run_sudo_command [COMMANDS...] -FJH 2010.03.17
run_sudo_command() {
# grab the commands passed to the function and put them in a variable for safe keeping
echo -en "${red}"
sudocommand=$*
sudo $sudocommand
# Check the exit status if it is not 0 (good) then assume that the password was not entered correctly and loop them till they get it right or cancel the running of this script.
echo -en "${NC}"
while [ ! $? = 0 ]; do
echo -e -n "${yellow}>>>${NC} Something is not right here. (Did you correctly enter your password? Is the Caps-Locks on?) Do you want to try to enter the password again${green}(Yes)${NC} or exit this script${red}(No)${NC}?"
  read -r response
response=${response,,}    # tolower
  if [[ $response !=  "y" && $response != "Y"  && $response != "yes" && $response != "Yes" ]]; then
    echo -e "${yellow}>>>${NC} ${red}Exiting! User aborted the script!${NC}"
		exit
		else
		echo -en ${red}
		sudo $sudocommand
		echo -en "${NC}"
	fi
	echo -en "${NC}"
done
}
finished(){
  echo ""
  echo -e "${yellow}>>>${NC} All done setting up the ${yellow}asus-fan module${NC}."
  echo -e "${yellow}>>>${NC} ${green}Hopefully you will now have a ${yellow}cool${green} runninh ${red}Asus laptop!${NC}"
}

cd /tmp/aaa-tmp/
git clone git@github.com:daringer/asus-fan.git
cd asus-fan/
make
run_sudo_command cp asus_fan.ko /lib/modules/$(uname -r)/kernel/drivers/thermal/
run_sudo_command depmod -a
run_sudo_command modprobe asus_fan
run_sudo_command /etc/init.d/thermald restart

```

frederickjh, 2016-03-02:
Are you running `depmod -a` as the super user before `modprobe asus_fan` as the super user?

hystryfe, 2016-03-03:
@frederickjh, The source comes from git and it is compiled against the kernel headers for the version I have. The version of the kernel is 4.4.1. I have not tried depmod -a, but I did that and it says the same thing. The fan is either not running or running at a very slow speed, but shows as running at max speed:

`acpitz-virtual-0
Adapter: Virtual device
temp1:        +57.0°C  (crit = +108.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +58.0°C  (high = +87.0°C, crit = +105.0°C)
Core 0:         +58.0°C  (high = +87.0°C, crit = +105.0°C)
Core 1:         +53.0°C  (high = +87.0°C, crit = +105.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:     25500 RPM
temp1:        +57.0°C  
`

Could there be something interfering? I do have lm_sensors running but disabling that doesn't seem to have an effect. (I was thinking that because it makes the cpu_fan show up above I think.)

frederickjh, 2016-03-03:
hi @hystryfe !

Have you rebooted since you booted into the new kernel after compiling asus-fan?  I find that some times until I reboot after a compile for a new kernel that the fan will stick at one setting until I reboot.

I am using the [psensor](http://wpitchoune.net/blog/psensor/) gui to monitor my temperatures and the fan.

Can you feel air coming out? On my Zenbook it exits in the slot between the keyboard and screen on the left hand side, blowing up across the screen. 

If you got it too hot there is the possibly that the fan or the RPM sensor is damaged. Even if  it is not damage it I would check that the heatsink compound has not melted like butter and flow out from between the CPU and heatsink lowering the efficiency of heat transfer. I have had this happen in the past.

I hope this helps and that you can get your machine up and cooling!
Frederick

hystryfe, 2016-03-03:
Yes, I have rebooted. The sensors don't show any crazy temperatures right now. I do not have the tools at the moment to check the heatsink/fan. It feels like there could be a wee little bit coming out or maybe that's just ambient heat dissipating. The fan is decidedly off now as I have heard clicking from it, like it's trying to start. The clicking is new, since today, but probably unrelated to the asus_fan module since it won't even load right now. I'm going to try to just get the fan running any way possible.

According to pwmconfig, there are no working fan sensors. I just heard the fan two days ago so I am not sure what happened. It might be damaged like you said @frederickjh.

I guess this issue is closed until I can get the sensor working properly.

closed 2016-03-03 by hystryfe