daringer/asus-fan #35: DKMS Error - modprobe: ERROR: could not insert 'asus_fan': Exec format error - kernel 3.13.0-88-generic

frederickjh, 2016-06-20:
Hi!
After upgrading to kernel 3.13.0-88-generic on Ubuntu 14.04 I saw that the asus-fan module was not loading. Trying to load it with modprobe returns this error:
`$ sudo modprobe asus_fan
modprobe: ERROR: could not insert 'asus_fan': Exec format error`

I have tried uninstalling  and then installing with DKMS but I am still getting this error. Any ideas on how to proceed / debug?

Thanks!
Frederick

daringer, 2016-11-05:
still encountering this? feels like the module is not correctly compiled against the running kernel...

frederickjh, 2016-11-05:
I forced grub to stay booting kernel version 3.13.0-79-generic which was the last version it worked for. I have not tried with newer kernels, but I need no revisit this and see if a solution can be found.

Any advice on how to go about debugging this would be appreciated.

daringer, 2016-11-05:
3.13. is pretty outdated, (at least from my humble arch-linux point of view :laughing: )
once you have time try it out and I'll support ... I suppose it's not something obvious like a hardened kernel needing signed modules ... any additional dmesg outputs could also be helpful

daringer, 2016-12-03:
you should try the debug-mode maybe during building, maybe this changes something...

frederickjh, 2016-12-12:
I just fixed this in my fork of the project. I created a pull request #41. The dmks.conf for Archlinux does not work for Ubuntu. I had to create a new dkms.conf file for Ubuntu. Then move the Archlinux version out of the way then move the Ubuntu version into place in my Ubuntu DKMS installation script.

You can close this issue upon merging the pull request.

frederickjh, 2016-12-12:
Forgot to say that the problem was the location of the build modules. DKMS could not find the built module to load it.

Regarding the kernel version I am running Ubuntu 14.04 LTS (want to upgraded to 16.04 LTS soon). They regularly release new kernel versions the latest now being 3.13.0-105-generic. I see that ArchLinux is on a 4.x version of the Linux kernel so it may look old. ;)

closed 2017-01-28 by daringer