daringer/asus-fan #37: The PKGBUILD on aur is often outdated

qlonik, 2016-11-06:
Hello,

The PKGBUILD in the aur for this package is often falling behind the actual PKGBUILD in this repo. Would you like to take over and update PKGBUILD in  the aur at the same time you update it in this repo?

daringer, 2016-11-14:
Technically this shouldn't be a problem, but have to admit that I've never have done anything in the AUR before... Can you give me a short walk through?

qlonik, 2016-11-15:
Basically, AUR has packages, whenever you install from it, it downloads PKGBUILD and other files, builds that package (tar.gz) and installs that with pacman. In previous AUR, they had tar balls to represent each package, but they switched to git repos now.

So, now every package there is a git repo, and releasing the package on AUR is just a `git push` now. There are couple requirements that they have. I think they require that every single commit has a `PKGBUILD` and `.SRCINFO` files. 

`PKGBUILD` is whatever you have here in your repo, and `.SRCINFO` is file generated from `PKGBUILD`. To get `.SRCINFO` the command is `mksrcinfo` from `pkgbuild-introspection` package. You also can have additional files if you need (like `file.install` or `file.desktop` or `file.patch` or `file.service`).

I guess another thing is that you need to register on AUR, add your ssh key (not sure if its required, but I did that) and whenever you push, you need to specify that key.

There is more detailed info on arch wiki, if you would like to read it:
* [AUR, submitting and maintaining packages](https://wiki.archlinux.org/index.php/Arch_User_Repository#Sharing_and_maintaining_packages)
* [Arch packaging standards](https://wiki.archlinux.org/index.php/Arch_packaging_standards#AUR_packages)
* [Description of PKGBUILD](https://wiki.archlinux.org/index.php/PKGBUILD)
* [Article about package creation](https://wiki.archlinux.org/index.php/Creating_packages)

There is some "Manage co-maintainers" link in the package actions box on the side. I've never tried it, but maybe this thing will allow you to push updates into this little repo.

If you have any more questions, hit me up. I will try to answer them.

daringer, 2016-11-27:
Once a PKGBUILD update shall be done I'll try to do it, will keep this issue open to track/discuss the next/first tries....

daringer, 2017-04-22:
looks like it should work, login and stuff, need to get to my Linux box soon, then I'll update the current one in the AUR fixing the permission issue, good minimal change to start with :smile:

erkexzcx, 2017-05-16:
Hi. AUR fails again, therefore not sure if it's PKGBUILD script issue or program itself, since I am unable to find actual module from lsmod command. See full raw output [here](https://paste.ee/p/TYH01).

daringer, 2017-06-27:
mmh, good that it works was updated end of May, either way I do not fully get how to get push rights for a package? @qlonik is this maybe something you have to grant me? "Manage co-maintainers" sounds like, you'll have to manage them, maybe you could just add me?

Either way, as currently AUR + PKGBUILD works fine, I will close this issue...

closed 2017-06-27 by daringer

qlonik, 2017-07-07:
Yes, I need to add you as a co-maintainer on that package. Is your handle on aur the same as here - daringer?

daringer, 2017-07-17:
nope there it's "dariball" ...

qlonik, 2017-07-17:
Added

daringer, 2017-07-21:
hehe nice, last time I wanted to be an AUR maintainer I was like 14 :smile: 
but alright, also pushed some minimal change---all seems to work, feel free to bug me whenever something is weird there...