daringer/asus-fan #38: ArchLinux LTS kernel fails to autoinstall

hexadecagram, 2016-11-07:
This kernel module isn't being rebuilt with every LTS kernel upgrade. I have to manually run dkms autoinstall every time. I've included a successful zfs-dkms in the pacman output, to demonstrate that it seems to be a problem specifically with asus-fan-dkms-git:

:: Running post-transaction hooks...
(1/1) Install DKMS modules
==> dkms install -m asus_fan -v git -k 4.4.30-1-lts
Error! Bad return status for module build on kernel: 4.4.30-1-lts (x86_64)
Consult /var/lib/dkms/asus_fan/git/build/make.log for more information.
==> dkms install -m spl -v 0.6.5.8 -k 4.4.30-1-lts
==> dkms install -m zfs -v 0.6.5.8 -k 4.4.30-1-lts

Here's some additional stuff:

% sudo dkms autoinstall -k 4.4.30-1-lts

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make KERNELRELEASE=4.4.30-1-lts KVERSION=4.4.30-1-lts...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.4.30-1-lts (x86_64)
Consult /var/lib/dkms/asus_fan/git/build/make.log for more information.

% tail /var/lib/dkms/asus_fan/git/build/make.log 
DKMS make.log for asus_fan-git for kernel 4.4.30-1-lts (x86_64)
Sun Nov  6 11:46:30 PST 2016
make -C /lib/modules/4.4.27-1-lts/build M=$PWD modules
make[1]: *** /lib/modules/4.4.27-1-lts/build: No such file or directory.  Stop.
make: *** [Makefile:6: all] Error 2

% ls /lib/modules/
4.4.30-1-lts  extramodules-4.4-lts

% uname -rs
Linux 4.4.27-1-lts

So it appears that make is trying to build with the old linux-headers-lts installation, which it should not be doing.

If I reboot and retry, I succeed:

% sudo dkms autoinstall

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make KERNELRELEASE=4.4.30-1-lts KVERSION=4.4.30-1-lts...
cleaning build area...
Kernel cleanup unnecessary for this kernel.  Skipping...

DKMS: build completed.

asus_fan.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/4.4.30-1-lts/kernel/drivers/acpi/

depmod...

DKMS: install completed.

daringer, 2016-11-14:
mmmh, ok looks like this could solve it.... could you confirm maybe?

hexadecagram, 2016-11-15:
Yep, that seems to have done the trick.

closed 2016-11-26 by daringer