daringer/asus-fan #39: modprobe: ERROR: could not insert 'asus_fan': No such device

Nightbr, 2016-12-02:
Hi,

I try to make my fan works on my ASUS ZenBook3U.

My configuration:
```
inxi -Fxz
System:    Host: NB-ZenBook Kernel: 4.4.0-51-generic x86_64 (64 bit gcc: 5.4.0)
           Desktop: Cinnamon 3.0.7 (Gtk 3.18.9-1ubuntu3.1)
           Distro: Linux Mint 18 Sarah
Machine:   System: ASUSTeK (portable) product: UX390UAK v: 1.0
           Mobo: ASUSTeK model: UX390UAK v: 1.0
           Bios: American Megatrends v: UX390UAK.303 date: 08/31/2016
CPU:       Dual core Intel Core i7-7500U (-HT-MCP-) cache: 4096 KB
           flags: (lm nx sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx) bmips: 11615
           clock speeds: max: 2701 MHz 1: 1600 MHz 2: 1100 MHz 3: 400 MHz
           4: 1100 MHz
Graphics:  Card: Intel Device 5916 bus-ID: 00:02.0
           Display Server: X.Org 1.18.3 drivers: intel (unloaded: fbdev,vesa)
           Resolution: 1920x1080@60.05hz
           GLX Renderer: Mesa DRI Intel Kabylake GT2
           GLX Version: 3.0 Mesa 11.2.0 Direct Rendering: Yes
Audio:     Card Intel Device 9d71 driver: snd_hda_intel bus-ID: 00:1f.3
           Sound: Advanced Linux Sound Architecture v: k4.4.0-51-generic
Network:   Card: Intel Wireless 8260 driver: iwlwifi bus-ID: 02:00.0
           IF: wlp2s0 state: up mac: <filter>
Drives:    HDD Total Size: 512.1GB (3.0% used)
           ID-1: /dev/sda model: Micron_1100_MTFD size: 512.1GB temp: 32C
Partition: ID-1: / size: 232G used: 6.8G (4%) fs: ext4 dev: /dev/sda5
           ID-2: swap-1 size: 8.47GB used: 0.00GB (0%) fs: swap dev: /dev/sda6
RAID:      No RAID devices: /proc/mdstat, md_mod kernel module present
Sensors:   System Temperatures: cpu: 40.0C mobo: N/A
           Fan Speeds (in rpm): cpu: N/A
Info:      Processes: 188 Uptime: 14 min Memory: 739.6/7867.0MB
           Init: systemd runlevel: 5 Gcc sys: 5.4.0
           Client: Shell (bash 4.3.421) inxi: 2.2.35 
```
I know my MB model is not supported by your code but there no driver for model UX390UAK...

So, I download the the code in /usr/src:
```
$ ls /usr/src/
asus_fan-master                 linux-headers-4.4.0-51-generic
linux-headers-4.4.0-21          master.tar.gz
linux-headers-4.4.0-21-generic  ndiswrapper-1.59
linux-headers-4.4.0-51          virtualbox-guest-5.0.24
```

And I installed with DKMS:
```
dkms add -m asus_fan -v master
dkms install -m asus_fan -v master
echo asus_fan >>/etc/modules
```

Everything works fine at this point.

Reboot.

And after reboot:

```
$ sudo modprobe asus_fan
modprobe: ERROR: could not insert 'asus_fan': No such device

 $ lsmod | grep asus
asus_nb_wmi            24576  0
asus_wmi               28672  1 asus_nb_wmi
sparse_keymap          16384  1 asus_wmi
wmi                    20480  1 asus_wmi
video                  40960  2 i915_bpo,asus_wmi

```

Any idea? At this moment, I have no fan control and it's quite dangerous... Or I can contribute in testing and/or developing the compatibility of this driver with the model UX390UAK. 

Thanks in advance!

daringer, 2016-12-03:
please try building the most recent git version by yourself using `make DEBUG=1`, this will activate the debugging output in `dmesg` once the module is loaded. What's written there will hopefully be a good hint what's going on. 

Further you may pass `force_load=1` to modprobe/insmod to avoid the device probing and checks.

Nightbr, 2016-12-03:
ok so I have done this:

```shell
sudo su
cd /usr/src
git clone https://github.com/daringer/asus-fan.git
cd asus-fan
```

```shell
# make DEBUG=1
make -C /lib/modules/4.4.0-51-generic/build M=$PWD modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.4.0-51-generic »
  CC [M]  /usr/src/asus-fan/asus_fan.o
/usr/src/asus-fan/asus_fan.c:44:0: warning: "DEBUG" redefined
 #define DEBUG (true)
 ^
<command-line>:0:0: note: this is the location of the previous definition
/usr/src/asus-fan/asus_fan.c: In function ‘__fan_get_cur_state’:
/usr/src/asus-fan/asus_fan.c:294:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
   int rpm = __fan_rpm(fan);
   ^
In file included from include/linux/printk.h:6:0,
                 from include/linux/kernel.h:13,
                 from include/linux/list.h:8,
                 from include/linux/module.h:9,
                 from /usr/src/asus-fan/asus_fan.c:8:
/usr/src/asus-fan/asus_fan.c: In function ‘__fan_set_cur_state’:
include/linux/kern_levels.h:4:18: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘long unsigned int’ [-Wformat=]
 #define KERN_SOH "\001"  /* ASCII Start Of Header */
                  ^
include/linux/kern_levels.h:13:19: note: in expansion of macro ‘KERN_SOH’
 #define KERN_INFO KERN_SOH "6" /* informational */
                   ^
/usr/src/asus-fan/asus_fan.c:49:23: note: in expansion of macro ‘KERN_INFO’
     if (DEBUG) printk(KERN_INFO "asus-fan (debug) - " fmt "\n", ##__VA_ARGS__);
                       ^
/usr/src/asus-fan/asus_fan.c:314:3: note: in expansion of macro ‘dbg_msg’
   dbg_msg("set fan-id: %d to state: %d", fan, state);
   ^
include/linux/kern_levels.h:4:18: warning: too many arguments for format [-Wformat-extra-args]
 #define KERN_SOH "\001"  /* ASCII Start Of Header */
                  ^
include/linux/kern_levels.h:11:22: note: in expansion of macro ‘KERN_SOH’
 #define KERN_WARNING KERN_SOH "4" /* warning conditions */
                      ^
/usr/src/asus-fan/asus_fan.c:64:12: note: in expansion of macro ‘KERN_WARNING’
     printk(KERN_WARNING "asus-fan (" title ") - " fmt "\n", ##__VA_ARGS__); \
            ^
/usr/src/asus-fan/asus_fan.c:317:5: note: in expansion of macro ‘warn_msg’
     warn_msg("set pwm", "illegal value provided: %d ", fan, (unsigned int) stat
     ^
/usr/src/asus-fan/asus_fan.c: In function ‘__fan_set_cur_control_state’:
include/linux/kern_levels.h:4:18: warning: too many arguments for format [-Wformat-extra-args]
 #define KERN_SOH "\001"  /* ASCII Start Of Header */
                  ^
include/linux/kern_levels.h:13:19: note: in expansion of macro ‘KERN_SOH’
 #define KERN_INFO KERN_SOH "6" /* informational */
                   ^
/usr/src/asus-fan/asus_fan.c:49:23: note: in expansion of macro ‘KERN_INFO’
     if (DEBUG) printk(KERN_INFO "asus-fan (debug) - " fmt "\n", ##__VA_ARGS__);
                       ^
/usr/src/asus-fan/asus_fan.c:333:3: note: in expansion of macro ‘dbg_msg’
   dbg_msg("set fan-id: %d control-state", fan, state);
   ^
/usr/src/asus-fan/asus_fan.c: In function ‘fan_set_speed’:
/usr/src/asus-fan/asus_fan.c:343:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
   union acpi_object args[2];
   ^
In file included from include/linux/printk.h:6:0,
[... 71 lines cut ...]
[ 1003.897997] asus-fan (init) - forced loading of module: USE WITH CARE
[ 1003.898003] asus-fan (debug) - register asus fan driver
[ 1003.898217] asus-fan (debug) - probe for device
[ 1003.898224] asus-fan (debug) - init hwmon device
[ 1003.898454] asus-fan (init) - finished init, found 2 fan(s) to control
[ 1007.031174] asus-fan (debug) - get fan-id: 0 rpm
[ 1007.031809] asus-fan (debug) - get fan-id: 1 rpm
[ 1007.032456] asus-fan (debug) - get temperature 1 from acpi
[ 1007.032968] asus-fan (debug) - get temperature 1 from acpi
[ 1012.088359] asus-fan (debug) - get fan-id: 0 rpm
[ 1012.088975] asus-fan (debug) - get fan-id: 1 rpm
[ 1012.089563] asus-fan (debug) - get temperature 1 from acpi
[ 1012.090008] asus-fan (debug) - get temperature 1 from acpi
[ 1017.151279] asus-fan (debug) - get fan-id: 0 rpm
[ 1017.152246] asus-fan (debug) - get fan-id: 1 rpm
[ 1017.152927] asus-fan (debug) - get temperature 1 from acpi
[ 1017.153449] asus-fan (debug) - get temperature 1 from acpi
[ 1022.211033] asus-fan (debug) - get fan-id: 0 rpm
[ 1022.211665] asus-fan (debug) - get fan-id: 1 rpm
[ 1022.212261] asus-fan (debug) - get temperature 1 from acpi
[ 1022.212739] asus-fan (debug) - get temperature 1 from acpi
```

I reboot and post other comment with debug.
Thanks for your help!

Nightbr, 2016-12-03:
After rebooting:

```
# sensors
acpitz-virtual-0
Adapter: Virtual device
temp1:        +37.0°C  (crit = +103.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +38.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:         +36.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:         +37.0°C  (high = +100.0°C, crit = +100.0°C)

```
No more asus_fan :(

```
# dmesg | grep asus
[    8.040222] asus_fan: module verification failed: signature and/or required key missing - tainting kernel
[    8.041112] asus-fan (debug) - starting init procedure
[    8.041125] asus-fan (debug) - dmi sys info: 'ASUSTeK COMPUTER INC.'
[    8.041134] asus-fan (debug) - get fan-id: 0 rpm
[   10.296251] asus_wmi: ASUS WMI generic driver loaded
[   10.315791] asus_wmi: Initialization: 0x1
[   10.316108] asus_wmi: BIOS WMI version: 9.0
[   10.316328] asus_wmi: SFUN value: 0x21
[   10.362651] input: Asus WMI hotkeys as /devices/platform/asus-nb-wmi/input/input7
[   10.368928] asus_wmi: Number of fans: 0
[   10.586678] asus-fan (debug) - starting init procedure
[   10.586696] asus-fan (debug) - dmi sys info: 'ASUSTeK COMPUTER INC.'
[   10.586705] asus-fan (debug) - get fan-id: 0 rpm

```

daringer, 2016-12-03:
mmmmh,
interessting, I'll start with what I see here:
- regular loading of the module fails (during init it tries to get the current RPM, looks like this fails, so the conclusion taken there is that no fan is available)

- forced loading of the module seems to work and is more or less behaving as expected once forced (but hard to say so far, please try executing the .sh create symlinks script:
```bash
cd misc
./asus-fan-create-symlinks.sh /tmp/asus-test-dir # as root 
```
Afterwards check inside `/tmp/asus-test-dir` if the symlinks are working, so do stuff as explained in the README on the frontpage, like:
```bash
$ echo 1 > fan_cpu_manual_mode
$ echo 20 > fan_cpu_speed
$ echo 255 > fan_cpu_speed
```
If you hear the fan changing its speed we might be lucky. (Also check with the "gpu" symlinks please)

- after reboot it's clear that the module will disappear, as long as you not specifically load it during boot with the `force_load=1` thus, as far as I see this is expected behavior

Two things to get more details on this:
(1) From what we see until now, it feels like the ACPI call itself fails. The module uses ACPI to communicate with the fan(s) and temperature sensor(s). There is just a small number of these ACPI calls:
```bash
\_SB.PCI0.LPCB.EC0.SFNV 
\_SB.PCI0.LPCB.EC0.TACH <---- this one seems to fail
\_SB.ATKD.QMOD
\_SB.PCI0.LPCB.EC0.ST98 
\_SB.PCI0.LPCB.EC0.SFNV 
\_SB.PCI0.LPCB.EC0.TH1R
```
On Linux you can directly interact with acpi through the proc filesystem, therefore you need to install acpi_call, if not already existing, just check `/proc/acpi/call`.

Once this is done, just do the following as root:
```bash
echo '\_SB.PCI0.LPCB.EC0.TACH' > /proc/acpi/call 
cat /proc/acpi/call; echo -e "\n"
```
If things are there as expected, you should see something like:
```bash
-$ cat /proc/acpi/call; echo -e "\n"
0xcb4lled
```
On the other side, if the path is not valid for some reason (laptop model, whatever...) you will see something like that:
```bash
-$ cat /proc/acpi/call; echo -e "\n"
Error: AE_BAD_PATHNAME
```
You could also check the other calls, mainly if any of those leads to a bad path name. Once we know this I can at least try to avoid these calls (sacrificing features most likely) but this will bring us closer to loading the module without forcing it.

(2) Further, please pull again from the asus-fan repository and repeat the process above (no need for rebooting). I've added some more debug output into the suspected parts of the code, maybe this could also help.

Nightbr, 2016-12-03:
Symlink seems to work:

```
NB-ZenBook nightbringer # cd /usr/src/
NB-ZenBook src # cd asus-fan/misc/
NB-ZenBook misc # ./asus-fan-create-symlinks.sh /tmp/asus-test-dir
[i] starting, target directory: /tmp/asus-test-dir
[+] finished....
NB-ZenBook misc # ls /tmp/asus-test-dir/
core1_temp  core2_temp  die_temp  tz1_temp  tz2_temp
```
And more info:

```
asus-test-dir # ls -al
total 8
drwxr-xr-x  2 root root 4096 déc.   3 14:12 .
drwxrwxrwt 15 root root 4096 déc.   3 14:27 ..
lrwxrwxrwx  1 root root   57 déc.   3 14:12 core1_temp -> /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp2_input
lrwxrwxrwx  1 root root   57 déc.   3 14:12 core2_temp -> /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp3_input
lrwxrwxrwx  1 root root   57 déc.   3 14:12 die_temp -> /sys/devices/platform/coretemp.0/hwmon/hwmon2/temp1_input
lrwxrwxrwx  1 root root   45 déc.   3 14:12 tz1_temp -> /sys/devices/virtual/hwmon/hwmon0/temp1_input
lrwxrwxrwx  1 root root   58 déc.   3 14:12 tz2_temp -> /sys/devices/platform/asus-nb-wmi/hwmon/hwmon*/temp1_input

NB-ZenBook asus-test-dir # cat core1_temp 
40000
NB-ZenBook asus-test-dir # cat core2_temp 
41000
NB-ZenBook asus-test-dir # cat die_temp 
42000
NB-ZenBook asus-test-dir # cat tz1_temp 
41000
```

Also, I found the control folder:
```
cd /sys/devices/platform/asus_fan/hwmon/hwmon3
NB-ZenBook hwmon3 # ls
device      fan1_speed_max  name         pwm2         temp1_input
fan1_input  fan2_input      power        pwm2_enable  temp1_label
fan1_label  fan2_label      pwm1         subsystem    uevent
fan1_min    fan2_min        pwm1_enable  temp1_crit
NB-ZenBook hwmon3 # cat pwm1
26NB-ZenBook hwmon3 # echo 255 > pwm
pwm1         pwm1_enable  pwm2         pwm2_enable  
NB-ZenBook hwmon3 # echo 255 > pwm1 # <---  fan not working
NB-ZenBook hwmon3 # echo 255 > pwm2 # <--- fan not working
NB-ZenBook hwmon3 # cat pwm1_enable 
1
NB-ZenBook hwmon3 # cat pwm2_enable 
1
```
No fan running/sound but the sensors tells us that it's at maximum speed:

```
 # sensors
acpitz-virtual-0
Adapter: Virtual device
temp1:        +40.0°C  (crit = +103.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +40.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:         +38.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:         +39.0°C  (high = +100.0°C, crit = +100.0°C)

asus_fan-isa-0000
Adapter: ISA adapter
CPU Fa:      3910 RPM  (min =   10 RPM)
GFX Fa:      3910 RPM  (min =   10 RPM)
[... 145 lines cut ...]
[  740.080170] asus-fan (debug) - starting init procedure
[  740.080179] asus-fan (init) - forced loading of module: USE WITH CARE
[  740.080185] asus-fan (debug) - register asus fan driver
[  740.080336] asus-fan (debug) - probe for device
[  740.080341] asus-fan (debug) - init hwmon device
[  740.080658] asus-fan (init) - finished init, found 2 fan(s) to control
[  740.202326] asus-fan (debug) - get fan-id: 0 rpm
[  740.202666] asus-fan (debug) - get fan-id: 1 rpm
[  740.202979] asus-fan (debug) - get temperature 1 from acpi
[  740.203327] asus-fan (debug) - get temperature 1 from acpi
[  745.229825] asus-fan (debug) - get fan-id: 0 rpm
[  745.230073] asus-fan (debug) - get fan-id: 1 rpm
[  745.230303] asus-fan (debug) - get temperature 1 from acpi
[  745.231174] asus-fan (debug) - get temperature 1 from acpi
[  750.250993] asus-fan (debug) - get fan-id: 0 rpm
[  750.251151] asus-fan (debug) - get fan-id: 1 rpm
[  750.251299] asus-fan (debug) - get temperature 1 from acpi
[  750.251551] asus-fan (debug) - get temperature 1 from acpi
[  755.264032] asus-fan (debug) - get fan-id: 0 rpm
[  755.264154] asus-fan (debug) - get fan-id: 1 rpm
```

Next now, I have not acpi_call installed, so I will install it: https://github.com/mkottman/acpi_call

Tell me if you need more debug, I come back with more info about acpi_call.

Nightbr, 2016-12-03:
For acpi_call, it seems I have the same problem to make install (SSL error):

```
NB-ZenBook acpi_call # make
make -C /lib/modules/4.4.0-51-generic/build M=/usr/src/acpi_call modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.4.0-51-generic »
  CC [M]  /usr/src/acpi_call/acpi_call.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /usr/src/acpi_call/acpi_call.mod.o
  LD [M]  /usr/src/acpi_call/acpi_call.ko
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.4.0-51-generic »
NB-ZenBook acpi_call # make install
make -C /lib/modules/4.4.0-51-generic/build M=/usr/src/acpi_call modules_install
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.4.0-51-generic »
  INSTALL /usr/src/acpi_call/acpi_call.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-51-generic
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.4.0-51-generic »
NB-ZenBook acpi_call # cat /proc/acpi/call
cat: /proc/acpi/call: Aucun fichier ou dossier de ce type
NB-ZenBook acpi_call # modprobe acpi_call
modprobe: FATAL: Module acpi_call not found in directory /lib/modules/4.4.0-51-generic
NB-ZenBook acpi_call # modprobe acpi_call force_load=1
modprobe: FATAL: Module acpi_call not found in directory /lib/modules/4.4.0-51-generic

```

Nightbr, 2016-12-03:
Also reboot, make, make install, modprobe asus_fan force_load=1

And I can hear the fan running when the temperature increase over 40°C (or 42°C).
```
NB-ZenBook asus-fan # sensors
acpitz-virtual-0
Adapter: Virtual device
temp1:        +40.0°C  (crit = +103.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +40.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:         +38.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:         +40.0°C  (high = +100.0°C, crit = +100.0°C)

asus_fan-isa-0000
Adapter: ISA adapter
CPU Fa:        -1 RPM  (min =   10 RPM)
GFX Fa:        -1 RPM  (min =   10 RPM)
gfx_tem:       +0.0°C  (crit =  +0.1°C)

```

Nightbr, 2016-12-03:
I have reboot and this time I haven't make, make install and modprobe asus_fan and I can hear the fan start running when the temperature increase over 40°C. So now it's just a matter of modprobe in order to control and monitor properly fan.

daringer, 2016-12-03:
I was already wondering about the ssl error in your first post, but apparently this has not much to do with our problem here. 

The reason for this ssl error is that ubuntu (thus also Mint) requires signed modules, this leads to this error. See [here about secure boot](https://wiki.ubuntu.com/SecurityTeam/SecureBoot) and [some more discussions specifically about module signing](http://askubuntu.com/questions/773734/how-to-install-module-ko-module-without-kernel-signature-or-kernel-rebuild-in-ub).

So your distribution simply forbids to copy non-signed modules into your kernel-module directory, so this is also the reason why `modprobe` is not working as intended. (To be clear: this is actually a feature^^) From now on please just load the module using `insmod asus-fan.ko`, so you can also be totally sure that you actually load the file into the kernel (instead of some `depmod` generated linkage) to avoid any confusion here.

The same will happen for `acpi_call`, but as stated before to actually load a kernel module you don't need either `modprobe` nor `depmod`, just `insmod` the file directly.

Now again back to the actual topic: I totally forgot that there is also a test bash script, which uses all the possible interfaces. So please make sure a recent (debug) version of `asus-fan` is build (updates in the repository again) and then start this script: `/misc/test_modules.sh`. As `modprobe` is not working with the module you should start this bash script from inside the repo, roughly like this:

```bash
cd /tmp
git clone https://github.com/daringer/asus-fan.git
cd asus-fan
make DEBUG=1
bash misc/test_module.sh force_load=1
```

The output shall be something like [this](http://pastebin.com/tfSBuXpp). Further you should listen, because the script will try to adjust the fan speed from nearly off up to the maximum, so if you hear a changing fan-speed during this test-module.sh run, there is a good chance we can make it work.

Nightbr, 2016-12-03:
Thanks a lot for your explaination!

I tried the test_modules script but unfortunately my fan not started...

```
B-ZenBook tmp # git clone https://github.com/daringer/asus-fan.git
Clonage dans 'asus-fan'...
remote: Counting objects: 480, done.
remote: Compressing objects: 100% (62/62), done.
remote: Total 480 (delta 33), reused 0 (delta 0), pack-reused 415
Réception d'objets: 100% (480/480), 127.35 KiB | 19.00 KiB/s, fait.
Résolution des deltas: 100% (238/238), fait.
Vérification de la connectivité... fait.
NB-ZenBook tmp # cd asus-fan/
NB-ZenBook asus-fan # make DEBUG=1
make -C /lib/modules/4.4.0-51-generic/build M=$PWD modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.4.0-51-generic »
  CC [M]  /tmp/asus-fan/asus_fan.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /tmp/asus-fan/asus_fan.mod.o
  LD [M]  /tmp/asus-fan/asus_fan.ko
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.4.0-51-generic »
NB-ZenBook asus-fan # bash misc/test_module.sh force_load=1
[i] ------------ Starting 'asus-fan' kernel module quick test
[i] Clearing 'dmesg' (kernel) log
[i] Trying to modprobe the module: 'asus_fan'
[+++] Successfully inserted module: 'asus_fan'

[i] ------------ Inspecting dmesg a.k.a. kernel log
[+++] 'dmesg' reports finished module init
[i] Remaining dmesg lines with related information:
[  322.785433] asus-fan (init) - forced loading of module: USE WITH CARE

[i] ------------ Search /sys files
[i] Checking for fan1 /sys/ files:
[+] Found /sys path (/sys/devices/platform/asus_fan)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon3)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon3/fan1_input)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon3/fan1_label)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon3/fan1_min)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon3/fan1_speed_max)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon3/pwm1)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon3/pwm1_enable)
[i] Checking for fan2 /sys/ files:
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon3/fan2_input)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon3/fan2_label)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon3/fan2_min)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon3/pwm2)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon3/pwm2_enable)
[+++] All /sys files found for primary fan (no: 1)
[+++] All /sys files found for secondary fan (no: 2)
[i] Looks like this machine has two fans!

[i] ------------ Test functionality
[i] Checking fan1 functionality! (label: 'CPU Fan')
[+++] Reading fan speed: -1
[---] Looks like the fan1 speed is not updated ... (/sys/devices/platform/asus_fan/hwmon/hwmon3/fan1_input)
```

daringer, 2016-12-04:
mmh really doesn't look good, 
so the only remaining option is to investigate how the ACPI fan stuff works on your laptop, then I could maybe realize this inside the module.
A good starting point would be the the `acer_ec.pl` perl script in the `misc/` directory, there you can start to investigate which registers change if your fan speed changes (depending on temperature, which you have to force by generating load). Anything beyond that gets quite complicated like dumping the acpi tables using `acpidump` and searching the web all this is quite reverse engineering driven. Not really fun to do...

Nightbr, 2016-12-04:
oh ok, thank you for your help anyway. 
Do you have any contact with Asus technical guy ? Because they can give us some documentation. With documentation I can create the driver or remap this one to the right registers and create a PR. 
I will try `acer_ec.pl` and investigate which registers change if my fan speed changes... But yeah it is really annoying that Asus don't give proper driver or documentation for the OpenSource communities...

Keep in touch :)

daringer, 2016-12-04:
Nope no ASUS contact here, but yes any contribution is highly appreciated, just ask if I can support...

HorstBaerbel, 2017-04-12:
First of all: Thansk for your work!
I have a similar issue here on the Asus ZenBook UX3410UA (integrated GPU, only one fan). I follow the steps to build and install the module using ./ubuntu_dkms_sudo_install.sh (which never asks me for my su password btw). The module seems to be built, but I get the same SSL error. When I use "./misc/test_module.sh force_load=1" the module loads:

> [i] ------------ Starting 'asus-fan' kernel module quick test
> [+] asus-fan loaded! REMOVING (rmmod) asus-fan (asus_fan) module
> [i] Clearing 'dmesg' (kernel) log
> [i] Trying to modprobe the module: 'asus_fan'
> [+++] Successfully inserted module: 'asus_fan'
> 
> [i] ------------ Inspecting dmesg a.k.a. kernel log
> [+++] 'dmesg' reports finished module init
> [i] Remaining dmesg lines with related information:
> [ 3196.173248] asus-fan (init) - forced loading of module: USE WITH CARE
> 
> [i] ------------ Search /sys files
> [i] Checking for fan1 /sys/ files:
> [+] Found /sys path (/sys/devices/platform/asus_fan)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon6)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon6/fan1_input)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon6/fan1_label)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon6/fan1_min)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon6/fan1_speed_max)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon6/pwm1)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon6/pwm1_enable)
> [i] Checking for fan2 /sys/ files:
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon6/fan2_input)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon6/fan2_label)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon6/fan2_min)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon6/pwm2)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon6/pwm2_enable)
> [+++] All /sys files found for primary fan (no: 1)
> [+++] All /sys files found for secondary fan (no: 2)
> [i] Looks like this machine has two fans!
> 
> [i] ------------ Test functionality
> [i] Checking fan1 functionality! (label: 'CPU Fan')
> [+++] Reading fan speed: 3003
> [+++] fan1 speed is updated (changed)
> [+] Change found in try no: 2!
> [+] Could read and verify (default) activated auto-mode: 0
> [+] Successfully read the min speed for fan1: 10
> [i] Set fan speed manually starting with fan1_min (10) using 15 as stepping
> [i] You should hear/feel the changing fan1 speed!
> [i] Set manual speed to: 15 30 45 60 75 90 105 120 135 150 
>                          165 180 195 210 225 240 255 
> [+++] Getting speed during manual mode for fan1 successfully done
> [+++] Setting manual speeds for fan1 successful!
> [+] Auto-mode (/sys/devices/platform/asus_fan/hwmon/hwmon6/pwm1_enable) flag behaves as intended for fan1!
> [+] Auto-mode (/sys/devices/platform/asus_fan/hwmon/hwmon6/pwm1_enable) was successfully reseted!
> [i] Checking fan2 functionality! (label: 'GFX Fan')
> [+++] Reading fan speed: -1
> [---] Looks like the fan2 speed is not updated ... (/sys/devices/platform/asus_fan/hwmon/hwmon6/fan2_input)
But the fan never changes its speed...
Manually insmodding the module work but trying to set the fan speed using "echo 123 > /sys/.../pwm1" fails with a "permisson denied", even with sudo?!

HorstBaerbel, 2017-04-12:
This is the NotebookFanControl xml file used to control the Fan:

> <?xml version="1.0"?>
> <FanControlConfigV2 xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance">
>   <NotebookModel>UX3410UA</NotebookModel>
>   <Author>bk</Author>
>   <EcPollInterval>150</EcPollInterval>
>   <ReadWriteWords>false</ReadWriteWords>
>   <CriticalTemperature>75</CriticalTemperature>
>   <FanConfigurations>
>     <FanConfiguration>
>       <ReadRegister>151</ReadRegister>
>       <WriteRegister>151</WriteRegister>
>       <MinSpeedValue>0</MinSpeedValue>
>       <MaxSpeedValue>8</MaxSpeedValue>
>       <IndependentReadMinMaxValues>false</IndependentReadMinMaxValues>
>       <MinSpeedValueRead>0</MinSpeedValueRead>
>       <MaxSpeedValueRead>0</MaxSpeedValueRead>
>       <ResetRequired>true</ResetRequired>
>       <FanSpeedResetValue>9</FanSpeedResetValue>
>       <TemperatureThresholds>
>         <TemperatureThreshold>
>           <UpThreshold>45</UpThreshold>
>           <DownThreshold>0</DownThreshold>
>           <FanSpeed>0</FanSpeed>
>         </TemperatureThreshold>
>         <TemperatureThreshold>
>           <UpThreshold>55</UpThreshold>
>           <DownThreshold>50</DownThreshold>
>           <FanSpeed>37.5</FanSpeed>
>         </TemperatureThreshold>
>         <TemperatureThreshold>
>           <UpThreshold>65</UpThreshold>
>           <DownThreshold>60</DownThreshold>
>           <FanSpeed>62.5</FanSpeed>
>         </TemperatureThreshold>
>         <TemperatureThreshold>
>           <UpThreshold>75</UpThreshold>
>           <DownThreshold>70</DownThreshold>
>           <FanSpeed>87.5</FanSpeed>
>         </TemperatureThreshold>
>         <TemperatureThreshold>
>           <UpThreshold>100</UpThreshold>
>           <DownThreshold>75</DownThreshold>
>           <FanSpeed>100</FanSpeed>
>         </TemperatureThreshold>
>         <TemperatureThreshold>
>           <UpThreshold>50</UpThreshold>
>           <DownThreshold>45</DownThreshold>
>           <FanSpeed>12.5</FanSpeed>
>         </TemperatureThreshold>
>         <TemperatureThreshold>
>           <UpThreshold>60</UpThreshold>
>           <DownThreshold>55</DownThreshold>
>           <FanSpeed>50</FanSpeed>
>         </TemperatureThreshold>
>         <TemperatureThreshold>
>           <UpThreshold>70</UpThreshold>
>           <DownThreshold>65</DownThreshold>
>           <FanSpeed>75</FanSpeed>
>         </TemperatureThreshold>
>       </TemperatureThresholds>
>       <FanSpeedPercentageOverrides />
>     </FanConfiguration>
>   </FanConfigurations>
>   <RegisterWriteConfigurations>
>     <RegisterWriteConfiguration>
>       <WriteMode>Set</WriteMode>
>       <WriteOccasion>OnWriteFanSpeed</WriteOccasion>
>       <Register>160</Register>
>       <Value>10</Value>
>       <ResetRequired>true</ResetRequired>
>       <ResetValue>10</ResetValue>
>       <ResetWriteMode>Set</ResetWriteMode>
>     </RegisterWriteConfiguration>
>     <RegisterWriteConfiguration>
>       <WriteMode>Set</WriteMode>
>       <WriteOccasion>OnWriteFanSpeed</WriteOccasion>
>       <Register>166</Register>
>       <Value>10</Value>
>       <ResetRequired>true</ResetRequired>
>       <ResetValue>10</ResetValue>
>       <ResetWriteMode>Set</ResetWriteMode>
>     </RegisterWriteConfiguration>
>   </RegisterWriteConfigurations>
> </FanControlConfigV2>

HorstBaerbel, 2017-04-21:
I toyed around a bit with acer_ec.pl:
"sudo perl ./acer_ec.pl := 97 0" turns off the fan until the CPU temperature rises gain. Then register 0x97 (151 dec) gets set to 1 again and the fan turns on again. Writing values 1-8 makes the fan run successively faster. It is fast to spin up, but only slowly becomes slower. When setting the value to 9 it will get reset to 1 again after some time.
The fan speed is related to 0x93/0x94. The value decreases (wtf) to ~400 (0x94 * 256 + 0x93) as the fan spins up and increases gain to ~700 when the fan spins down to its regular state. It is zero when the fan is off. Writing the registers does nothing though.
The CPU temperature can be found in 0xA0 and/or 0xA6. These do not increase with fan speed, but when the CPU activity goes up.
All this sounds like the behaviour described here: http://forum.notebookreview.com/threads/fan-control-on-asus-prime-ux31-ux31a-ux32a-ux32vd.705656/

daringer, 2017-04-22:
Hey @HorstBaerbel, as it looks like we are talking about another hardware / product here, can we please continue [here in another issue](https://github.com/daringer/asus-fan/issues/44) to keep them in separated issues #39 + #44 (even if the visible immediate error is the same (module load fails).

Looks like we might need some automation to debug hardware, which is not supported out of the box. Feature request here: #46

daringer, 2019-06-21:
seems to be covered by the other issues, please re-open if not

closed 2019-06-21 by daringer