daringer/asus-fan #40: fancontrol fails to read gfx_temp

saultdon, 2016-12-07:
Fancontrol is failing to start because it can't read the value for gfx_temp even though the asus_fan module is loaded and fresh links created with the asus-fan-create-symlinks.sh script.

```
● fancontrol.service - Start fan control, if configured
   Loaded: loaded (/usr/lib/systemd/system/fancontrol.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Wed 2016-12-07 10:53:56 MST; 4s ago
  Process: 1025 ExecStart=/usr/sbin/fancontrol (code=exited, status=1/FAILURE)
 Main PID: 1025 (code=exited, status=1/FAILURE)

Dec 07 10:53:56 archasus fancontrol[1025]:   MINPWM=90
Dec 07 10:53:56 archasus fancontrol[1025]:   MAXPWM=255
Dec 07 10:53:56 archasus fancontrol[1025]: Enabling PWM on fans...
Dec 07 10:53:56 archasus fancontrol[1025]: Starting automatic fan control...
Dec 07 10:53:56 archasus fancontrol[1025]: Error reading temperature from ///tmp/asus-fan-shm/gfx_temp
Dec 07 10:53:56 archasus fancontrol[1025]: Aborting, restoring fans...
Dec 07 10:53:56 archasus fancontrol[1025]: Verify fans have returned to full speed
Dec 07 10:53:56 archasus systemd[1]: fancontrol.service: Main process exited, code=exited, status=1/FAILURE
Dec 07 10:53:56 archasus systemd[1]: fancontrol.service: Unit entered failed state.
Dec 07 10:53:56 archasus systemd[1]: fancontrol.service: Failed with result 'exit-code'.
```

gfx_temp has a value, that looks like it should work...:

```
% cat /tmp/asus-fan-shm/gfx_temp
36000%
```

And sensors at first glance has a readable output showing the asus_fan module has loaded correctly, but after a minute or two and checking sensors again the GFX Fa value shows -1 RPM!:
```
% sensors                                                                                                                              =95%
asus_fan-isa-0000
Adapter: ISA adapter
CPU Fa:      2005 RPM  (min =   10 RPM)
GFX Fa:        -1 RPM  (min =   10 RPM)
gfx_tem:      +39.0°C  (crit =  +0.1°C)
```

This is on Arch Linux kernel 4.8.12

Tried reloading the asus_fan module with modprobe and then recreating the links with asus-fan-create-symlinks.sh but that results in the same behaviour as above.

saultdon, 2016-12-09:
Looks like maybe it has something to do with the % symbol seen in the output of cat.

The other temp links don't have it:

```
% cat /tmp/asus-fan-shm/gfx_temp
37000%                                                                                                                                                          % cat /tmp/asus-fan-shm/core1_temp
32000
% cat /tmp/asus-fan-shm/core2_temp
33000
% cat /tmp/asus-fan-shm/die_temp
39000
% cat /tmp/asus-fan-shm/tz1_temp
38000
% cat /tmp/asus-fan-shm/tz2_temp
38000
```

saultdon, 2016-12-09:
Fans can be manually controlled with:

```
$ echo 1 > fan_{cpu,gfx}_manual_mode
$ echo 20 > fan_{cpu,gfx}_speed
$ echo 255 > fan_{cpu,gfx}_speed
```

daringer, 2017-01-28:
mmmh for me there is no "%" at the end of gfx_temp, can you confirm that this problem still exists?

saultdon, 2017-01-30:
Just re-installed with kernel 4.9.6 and the % is still in the gfx_temp file.

daringer, 2017-02-19:
you might try a recent git build, applied a rather dirty hack to work around your problem ...

saultdon, 2017-02-20:
Looks like the same issue is still there. Running linux 4.9.9 on Arch Linux and built latest asus-fan (v0.9.1.r49.gb8a358f)

saultdon, 2017-04-10:
fancontrol is running successfully now after updating to latest kernel 4.10.8-1 on arch linux, and asus-fan is v0.9.1.r51.g5517472

closed 2017-04-10 by saultdon