daringer/asus-fan #44: [unsupported hardware: ZenBook UX3410UA] module load fails

daringer, 2017-04-22:
Hey, thanks and good tests, let's see if we can dig deeper,
this is the second time I see the `test_module.sh` script does not recognize a non working fan control. Somehow fails on the main task it should do :disappointed: 

So generally we should check the following points:
- what's the reason the module is not loaded
- is the acpi call similar (or missing, or somehow changed) to [your link](http://forum.notebookreview.com/threads/fan-control-on-asus-prime-ux31-ux31a-ux32a-ux32vd.705656/), think I used this page, too---initially. 
- disable specific functions, which maybe block the other functionalities...

From here try the following way:
1) compile the module manually in debug-mode: [see here how](https://github.com/daringer/asus-fan#debug-module)
2) please post what `dmesg` says, if you load the module using `insmod asus_fan.ko` from within the build dir---please post what happens with and without `force-load=1`. (I also added some more debug, just now, no linux box here, so it's untested, will check it later, but "looks good" :smile: ) (this will hopefully tell us what is happening under the hood)
3) next steps will then follow, once we know more, but under the line we either do some code changes to check, if this helps (based on the `dmesg` output we see) or should play with `acpi_call` using [your link](http://forum.notebookreview.com/threads/fan-control-on-asus-prime-ux31-ux31a-ux32a-ux32vd.705656/) this at least will hopefully show which ACPI paths are existing/working, please check this comment [here in the other issue](https://github.com/daringer/asus-fan/issues/39#issuecomment-264635950), generally we only have to test:
```
\_SB.PCI0.LPCB.EC0.SFNV 
\_SB.PCI0.LPCB.EC0.TACH
\_SB.ATKD.QMOD
\_SB.PCI0.LPCB.EC0.ST98 
\_SB.PCI0.LPCB.EC0.SFNV 
\_SB.PCI0.LPCB.EC0.TH1R
```
It is very likely that if any of those is missing or so, loading the module might fail. If you find out which one this is, I could work around this in the init phase (and/or exclude it completely). Depending on which one this is, we then also have to find the replacement to get full functionality for `asus_fan`.

HorstBaerbel, 2017-04-22:
Ok. I cloned the repo to "/usr/src", called "make DEBUG=1" and ran "./misc/test_module.sh".
W/O "force_load=1" I got:

> [i] ------------ Starting 'asus-fan' kernel module quick test
> [i] Clearing 'dmesg' (kernel) log
> [i] Trying to modprobe the module: 'asus_fan'
> modprobe: ERROR: could not insert 'asus_fan': No such device
> [i] Could not modprobe the module: 'asus_fan'
> [i] Will try to find the module (.ko file) (starting in '../')
> [+++] looks like I found: ../asus-fan/asus_fan.ko, trying insmod directly
> insmod: ERROR: could not insert module ../asus-fan/asus_fan.ko: No such device
> 
> [i] ------------ Inspecting dmesg a.k.a. kernel log
> [---] 'dmesg' reports no finished module init!

WITH "force_load=1" I got:

> [i] ------------ Starting 'asus-fan' kernel module quick test
> [i] Clearing 'dmesg' (kernel) log
> [i] Trying to modprobe the module: 'asus_fan'
> [+++] Successfully inserted module: 'asus_fan'
> 
> [i] ------------ Inspecting dmesg a.k.a. kernel log
> [+++] 'dmesg' reports finished module init
> [i] Remaining dmesg lines with related information:
> [ 3912.383694] asus-fan (init) - forced loading of module: USE WITH CARE
> 
> [i] ------------ Search /sys files
> [i] Checking for fan1 /sys/ files:
> [+] Found /sys path (/sys/devices/platform/asus_fan)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon5)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon5/fan1_input)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon5/fan1_label)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon5/fan1_min)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon5/fan1_speed_max)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon5/pwm1)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon5/pwm1_enable)
> [i] Checking for fan2 /sys/ files:
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon5/fan2_input)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon5/fan2_label)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon5/fan2_min)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon5/pwm2)
> [+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon5/pwm2_enable)
> [+++] All /sys files found for primary fan (no: 1)
> [+++] All /sys files found for secondary fan (no: 2)
> [i] Looks like this machine has two fans!
> 
> [i] ------------ Test functionality
> [i] Checking fan1 functionality! (label: 'CPU Fan')
> [+++] Reading fan speed: 3028
> [+++] fan1 speed is updated (changed)
> [+] Change found in try no: 2!
> [+] Could read and verify (default) activated auto-mode: 0
> [+] Successfully read the min speed for fan1: 10
> [i] Set fan speed manually starting with fan1_min (10) using 15 as stepping
> [i] You should hear/feel the changing fan1 speed!
> [i] Set manual speed to: 15 30 45 60 75 90 105 120 135 150 
>                          165 180 195 210 225 240 255 
> [+++] Getting speed during manual mode for fan1 successfully done
> [+++] Setting manual speeds for fan1 successful!
> [+] Auto-mode (/sys/devices/platform/asus_fan/hwmon/hwmon5/pwm1_enable) flag behaves as intended for fan1!
> [+] Auto-mode (/sys/devices/platform/asus_fan/hwmon/hwmon5/pwm1_enable) was successfully reseted!
> [i] Checking fan2 functionality! (label: 'GFX Fan')
> [+++] Reading fan speed: -1
> [---] Looks like the fan2 speed is not updated ... (/sys/devices/platform/asus_fan/hwmon/hwmon5/fan2_input)

dmesg returned:

> [ 4111.732246] asus-fan (set_auto) - failed reseting fan(s) to auto-mode! errcode: 5 - DANGER! OVERHEAT? DANGER!
> [ 4111.732249] asus-fan (init) - set auto-mode speed to active, failed! errcode: 5
> [ 4114.117718] asus-fan (debug) - starting initialization...
> [ 4114.117722] asus-fan (init) - dmi sys info: 'ASUSTeK COMPUTER INC.'
> [ 4114.117723] asus-fan (init) - dmi product name: 'UX410UAK'
> [ 4114.117724] asus-fan (debug) - dmi chassis type: '10'
> [ 4114.117727] asus-fan (debug) - fan-id: 0 | get RPM
> [ 4114.117728] asus-fan (debug) - |--> get RPM using acpi
> [ 4114.117729] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
> [ 4114.118084] asus-fan (debug) - |--> acpi request returned: 0
> [ 4114.118085] asus-fan (debug) - fan-id: 1 | get RPM
> [ 4114.118086] asus-fan (debug) - |--> get RPM using acpi
> [ 4114.118087] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
> [ 4114.118343] asus-fan (debug) - |--> acpi request returned: 0
> [ 4114.118344] asus-fan (debug) - __fan_rpm() calls succeeded, found 1 fan(s)
> [ 4114.118346] asus-fan (debug) - fan-id: (both) | set max speed: 255, force reset: 0
> [ 4114.119431] asus-fan (debug) - fan_set_max_speed() call succeeded, ret: 0
> [ 4114.119432] asus-fan (debug) - fan-id: (both) | set to automatic mode
> [ 4114.119445] asus-fan (set_auto) - failed reseting fan(s) to auto-mode! errcode: 5 - DANGER! OVERHEAT? DANGER!
> [ 4114.119451] asus-fan (init) - set auto-mode speed to active, failed! errcode: 5

Btw. How would I execute those API calls? Is there some way to do it via bash or is only C the way to go?

daringer, 2017-04-22:
Can you please also show me the dmesg output (of the debug-build-module) after you tried to `insmod asus_fan.ko` *without* `force-load=1`, there we maybe see what fails...

further acpi_call is under the line just a kernel module which allows you "echoing" into your acpi also from the bash for example. there will surely be a package, if you don't have it anyways, just go for a `modprobe acpi_call`, if this works you should find a `/proc/acpi/call` where you may directly echo into.

HorstBaerbel, 2017-04-23:
Oh. Sorry the dmesg output was using `./misc/test_module.sh` without `force-load=1`. Here ist is again, just to be sure:

> [27891.355150] asus-fan (set_auto) - failed reseting fan(s) to auto-mode! errcode: 5 - DANGER! OVERHEAT? DANGER!
> [27891.355153] asus-fan (init) - set auto-mode speed to active, failed! errcode: 5
> [27891.557172] asus-fan (debug) - starting initialization...
> [27891.557173] asus-fan (init) - dmi sys info: 'ASUSTeK COMPUTER INC.'
> [27891.557173] asus-fan (init) - dmi product name: 'UX410UAK'
> [27891.557174] asus-fan (debug) - dmi chassis type: '10'
> [27891.557174] asus-fan (debug) - fan-id: 0 | get RPM
> [27891.557174] asus-fan (debug) - |--> get RPM using acpi
> [27891.557175] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
> [27891.557314] asus-fan (debug) - |--> acpi request returned: 0
> [27891.557315] asus-fan (debug) - fan-id: 1 | get RPM
> [27891.557315] asus-fan (debug) - |--> get RPM using acpi
> [27891.557316] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
> [27891.557419] asus-fan (debug) - |--> acpi request returned: 0
> [27891.557420] asus-fan (debug) - __fan_rpm() calls succeeded, found 1 fan(s)
> [27891.557420] asus-fan (debug) - fan-id: (both) | set max speed: 255, force reset: 0
> [27891.558198] asus-fan (debug) - fan_set_max_speed() call succeeded, ret: 0
> [27891.558198] asus-fan (debug) - fan-id: (both) | set to automatic mode
> [27891.558202] asus-fan (set_auto) - failed reseting fan(s) to auto-mode! errcode: 5 - DANGER! OVERHEAT? DANGER!
> [27891.558204] asus-fan (init) - set auto-mode speed to active, failed! errcode: 5

HorstBaerbel, 2017-04-23:
I can toy around with the ACPI calls via `/proc/acpi/call` tomorrow if you tell me what I should be looking for...

daringer, 2017-04-23:
uh, ok so there I see a dmesg message without `force-load=1` ... ok so loading of the module seems fine, 

for acpi, see this [post](https://github.com/daringer/asus-fan/issues/39#issuecomment-264635950) there you see what to echo  into the acpi call file ... looks like disabling the auto-mode is the evil part here:
```
[27891.558202] asus-fan (set_auto) - failed reseting fan(s) to auto-mode! errcode: 5 - DANGER! OVERHEAT? DANGER!
[27891.558204] asus-fan (init) - set auto-mode speed to active, failed! errcode: 5
```

So there is this acpi entry handling this, currently with 2 params, 0 0 `\\_SB.PCI0.LPCB.EC0.SFNV` 
So you could try passing other parameters to to it using acpi call ... and we could also dig the internet for the appropriate acpi entry...

HorstBaerbel, 2017-04-23:
I installed `acpi-call-dkms` and started the module with `modprobe acpi_call`. It installs and starts without errors or messages in dmesg, but when I `sudo echo \_SB.PCI0.LPCB.EC0.SFNV > /proc/acpi/call` tells me "permission denied"... :/?

HorstBaerbel, 2017-04-23:
`ls -la /proc/acpi/`
```
dr-xr-xr-x   3 root root 0 Apr 23 22:31 .
dr-xr-xr-x 252 root root 0 Apr 23 20:49 ..
dr-xr-xr-x   3 root root 0 Apr 23 22:31 button
-rw-rw----   1 root root 0 Apr 23 22:31 call
-rw-r--r--   1 root root 0 Apr 23 22:31 wakeup
```
At least root should be able to execute it...

daringer, 2017-04-27:
you should try it directly being root, and with ""....
```
$ sudu su
$ echo "\_SB.PCI0.LPCB.EC0.SFNV 0" > call
$ cat call; echo
0x0called
```
This should be the roughly the output, try out some combinations as to be seen in the other post, especially the SFNV above. The argument "0" leads to low fan speeds on my side, using "1" and or "2" leads to very low (if not off) running fans... Try it out please and tell me if you hear any difference....

HorstBaerbel, 2017-04-27:
I get:

```
$ sudo su
$ echo "\_SB.PCI0.LPCB.EC0.SFNV 0" > call
$ cat call; echo

```

(empty result, just an empty line)
Fan speeds do not change regardless of the number (or choosing .TACH, .ST98 or whatnot). I might be doing something wrong. Does not matter if asus_fan.ko and acpi_call are loaded or not... :/

happel007, 2017-05-03:
I've got a UX310UAK here, which is internally the same as @HorstBaerbel's UX410UAK.

Also have to force load the module.

Using asus-fan I can manually set the fan speed. lm sensors even reports the fan speed according to the set speed. The fan doesn't actually change it's speed though. lm-sensors does show the actual speed of the fan when set to auto.

I've included the test_module.sh and acpi_call output below. Hope it helps, would be nice if we can get this to work. 

test_module.sh output:
```
[i] ------------ Starting 'asus-fan' kernel module quick test
[+] asus-fan loaded! REMOVING (rmmod) asus-fan (asus_fan) module
[i] Clearing 'dmesg' (kernel) log
[i] Trying to modprobe the module: 'asus_fan'
[+++] Successfully inserted module: 'asus_fan'

[i] ------------ Inspecting dmesg a.k.a. kernel log
[+++] 'dmesg' reports finished module init
[i] Remaining dmesg lines with related information:
[78303.292836] asus-fan (debug) - starting initialization...
[78303.292838] asus-fan (init) - dmi sys info: 'ASUSTeK COMPUTER INC.'
[78303.292838] asus-fan (init) - dmi product name: 'UX310UAK'
[78303.292839] asus-fan (debug) - dmi chassis type: '10'
[78303.292839] asus-fan (init) - forced loading of module: USE WITH CARE
[78303.292839] asus-fan (debug) - register asus fan driver
[78303.292891] asus-fan (debug) - probe for device
[78303.292892] asus-fan (debug) - init hwmon device

[i] ------------ Search /sys files
[i] Checking for fan1 /sys/ files:
[+] Found /sys path (/sys/devices/platform/asus_fan)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon7)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon7/fan1_input)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon7/fan1_label)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon7/fan1_min)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon7/fan1_speed_max)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon7/pwm1)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon7/pwm1_enable)
[i] Checking for fan2 /sys/ files:
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon7/fan2_input)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon7/fan2_label)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon7/fan2_min)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon7/pwm2)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon7/pwm2_enable)
[+++] All /sys files found for primary fan (no: 1)
[+++] All /sys files found for secondary fan (no: 2)
[i] Looks like this machine has two fans!

[i] ------------ Test functionality
[i] Checking fan1 functionality! (label: 'CPU Fan')
[+++] Reading fan speed: 3032
[+++] fan1 speed is updated (changed)
[+] Change found in try no: 4!
[+] Could read and verify (default) activated auto-mode: 0
[+] Successfully read the min speed for fan1: 10
[i] Set fan speed manually starting with fan1_min (10) using 15 as stepping
[i] You should hear/feel the changing fan1 speed!
[i] Set manual speed to: 15 30 45 60 75 90 105 120 135 150 
                         165 180 195 210 225 240 255 
[+++] Getting speed during manual mode for fan1 successfully done
[+++] Setting manual speeds for fan1 successful!
[+] Auto-mode (/sys/devices/platform/asus_fan/hwmon/hwmon7/pwm1_enable) flag behaves as intended for fan1!
[+] Auto-mode (/sys/devices/platform/asus_fan/hwmon/hwmon7/pwm1_enable) was successfully reseted!
[i] Checking fan2 functionality! (label: 'GFX Fan')
[+++] Reading fan speed: -1
[---] Looks like the fan2 speed is not updated ... (/sys/devices/platform/asus_fan/hwmon/hwmon7/fan2_input)
```
I got the following from acpi_call:
```
\_SB.PCI0.LPCB.EC0.SFNV    Error: AE_NOT_FOUND 
\_SB.PCI0.LPCB.EC0.TACH    0xbc3lled
\_SB.ATKD.QMOD             Error: AE_NOT_FOUND
\_SB.PCI0.LPCB.EC0.ST98    0x0called
\_SB.PCI0.LPCB.EC0.TH1R    0x0called
```

pasdVn, 2017-05-16:
Hi,
I have got an UX330UAK which (I guess) is quite similar to the mentioned devices above. So I just add my results here instead of opening a seperate issue:

build + install:
```
tobias@hugo:~/Entwicklung/asus-fan$ make DEBUG=1
make -C /lib/modules/4.8.0-51-generic/build M=$PWD modules
make[1]: Entering directory '/usr/src/linux-headers-4.8.0-51-generic'
  CC [M]  /home/tobias/Entwicklung/asus-fan/asus_fan.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/tobias/Entwicklung/asus-fan/asus_fan.mod.o
  LD [M]  /home/tobias/Entwicklung/asus-fan/asus_fan.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.8.0-51-generic'
tobias@hugo:~/Entwicklung/asus-fan$ sudo make install
make -C /lib/modules/4.8.0-51-generic/build M=$PWD modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.8.0-51-generic'
  INSTALL /home/tobias/Entwicklung/asus-fan/asus_fan.ko
At main.c:158:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.8.0-51-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.8.0-51-generic'
depmod -a
```

Running the tests with force_load=1 as otherwise loading fails when trying to set auto-mode :
```
tobias@hugo:~/Entwicklung/asus-fan$ misc/test_module.sh force_load=1
[i] ------------ Starting 'asus-fan' kernel module quick test
[i] Clearing 'dmesg' (kernel) log
[i] Trying to modprobe the module: 'asus_fan'
[+++] Successfully inserted module: 'asus_fan'

[i] ------------ Inspecting dmesg a.k.a. kernel log
[+++] 'dmesg' reports finished module init
[i] Remaining dmesg lines with related information:
[12091.548914] asus-fan (debug) - starting initialization...
[12091.548915] asus-fan (init) - dmi sys info: 'ASUSTeK COMPUTER INC.'
[12091.548915] asus-fan (init) - dmi product name: 'UX330UAK'
[12091.548916] asus-fan (debug) - dmi chassis type: '10'
[12091.548916] asus-fan (init) - forced loading of module: USE WITH CARE
[12091.548916] asus-fan (debug) - register asus fan driver
[12091.549050] asus-fan (debug) - probe for device
[12091.549051] asus-fan (debug) - init hwmon device

[i] ------------ Search /sys files
[i] Checking for fan1 /sys/ files:
[+] Found /sys path (/sys/devices/platform/asus_fan)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon10)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon10/fan1_input)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon10/fan1_label)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon10/fan1_min)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon10/fan1_speed_max)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon10/pwm1)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon10/pwm1_enable)
[i] Checking for fan2 /sys/ files:
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon10/fan2_input)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon10/fan2_label)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon10/fan2_min)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon10/pwm2)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon10/pwm2_enable)
[+++] All /sys files found for primary fan (no: 1)
[+++] All /sys files found for secondary fan (no: 2)
[i] Looks like this machine has two fans!

[i] ------------ Test functionality
[i] Checking fan1 functionality! (label: 'CPU Fan')
[+++] Reading fan speed: 3438
[... 143 lines cut ...]
[12114.107503] asus-fan (debug) - |--> acpi request returned: 0
[12115.116673] asus-fan (debug) - fan-id: 1 | get RPM
[12115.116676] asus-fan (debug) - |--> get RPM using acpi
[12115.116678] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
[12115.117224] asus-fan (debug) - |--> acpi request returned: 0
[12116.125127] asus-fan (debug) - fan-id: 1 | get RPM
[12116.125131] asus-fan (debug) - |--> get RPM using acpi
[12116.125133] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
[12116.126189] asus-fan (debug) - |--> acpi request returned: 0
[12117.133188] asus-fan (debug) - fan-id: 1 | get RPM
[12117.133190] asus-fan (debug) - |--> get RPM using acpi
[12117.133191] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
[12117.133407] asus-fan (debug) - |--> acpi request returned: 0
[12118.141026] asus-fan (debug) - fan-id: 1 | get RPM
[12118.141031] asus-fan (debug) - |--> get RPM using acpi
[12118.141033] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
[12118.141739] asus-fan (debug) - |--> acpi request returned: 0
[12119.150332] asus-fan (debug) - fan-id: 1 | get RPM
[12119.150334] asus-fan (debug) - |--> get RPM using acpi
[12119.150335] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
[12119.150697] asus-fan (debug) - |--> acpi request returned: 0
```
The fan speed is not changing during the test.

I also played a bit around with manually setting the fan speed

afilipovich, 2017-05-29:
Hi, I have Asus UX360UAK and experience the same issue: asus_fan module fails to load, if forced to load it shows current temp/fan status but does not change fan speed.

When I try ACPI calls I see the same result as @pasdVn, i.e. "\_SB.PCI0.LPCB.EC0.SFNV" and "\_SB.ATKD.QMOD" are missing.

After much tinkering with decompiled DSDT I found ACPI calls which can replace missing ones:
```
\_SB.PCI0.LPCB.EC0.WRAM 0x0521 0xc5   # full fan speed, locks register 0x97; .TACH method shows current fan speed
\_SB.PCI0.LPCB.EC0.WRAM 0x0521 0x85   # back to automatic fan speed control
\_SB.PCI0.LPCB.EC0.WRAM 0x0521 0x35   # manual fan control. Fan speed can be controlled by .ST84 call.
\_SB.PCI0.LPCB.EC0.ST84 Arg0 Arg1     # Arg0 - fan id (0x00 for CPU fan). Arg1 - speed 0x00 - 0xff
```
For GPU following calls should work but I cannot test them as my laptop does not have GPU:
```
\_SB.PCI0.LPCB.EC0.WRAM 0x0522 0xc5   # full CPU fan speed
\_SB.PCI0.LPCB.EC0.WRAM 0x0522 0x85   # back to auto fan speed control
\_SB.PCI0.LPCB.EC0.WRAM 0x0522 0x35   # manual fan control. Fan speed can be controlled by .ST84 call with Arg0 = 0x01
```
I am going to do quick and dirty update of current asus_fan.c for use on my laptop, however I think proper solution would auto-detect laptop model and select appropriate ACPI calls (or try original method first and fall-back to that more hacky one for newer laptops) which is beyond my C abilities.

Thank you for making this kernel module available!

HorstBaerbel, 2017-06-01:
Nice! It looks like this is going somewhere... Will try the ACPI calls on my laptop too when I find time.
Maybe we should make a list of models and if they have one or two fans:
One fan: UX310UAK?, UX330UAK?, UX360UAK, UX410UAK (integrated GPU)
Two fans: UX...UQ models (dedicated GPU)?

afilipovich, 2017-06-10:
I found thermal tipping point table in embedded controller RAM. It is between addresses 0x537. 0x53e.
Tipping point values can be changed via .WRAM ACPI call.

Default table:
35, 40, 45, 50, 55, 60, 65, 80
which means EC sets fan power to "1" if temperature exceeds 35C, increase to "2" if it exceeds 40, etc.

I have adjusted tipping points to following temp values and my laptop is silent most of the time:
45, 48, 51, 55, 58, 60, 65, 80

ACPI calls to achieve that:
```
\_SB.PCI0.LPCB.EC0.WRAM 0x537 0x2d
\_SB.PCI0.LPCB.EC0.WRAM 0x538 0x30
\_SB.PCI0.LPCB.EC0.WRAM 0x539 0x33
\_SB.PCI0.LPCB.EC0.WRAM 0x53a 0x37
\_SB.PCI0.LPCB.EC0.WRAM 0x53b 0x3a
```

Pros of this approach:
- Embedded Controller keeps controlling the fan and it works even without asus_fan kernel module.
- .TACH ACPI call shows fan RPM readings.

Cons:
- not sure if that works for GPU fan.
- other laptop models may contain that thermal table at different RAM region.
- it does not set exact fan speed, only defines tipping points.

afilipovich, 2017-06-11:
Proof-of-concept script to tweak cooling profile:
https://github.com/afilipovich/asus_ux360uak/blob/master/update_cooling_profile.py

HorstBaerbel, 2017-06-11:
It might sound stupid, but I'm stuck using acpi_call. I do

`echo '\_SB.PCI0.LPCB.EC0.WRAM <ADDRESS>' | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call`

Which souhld read read from <ADDRESS>. It does not matter what address I use, I always get the result 
```
\_SB.PCI0.LPCB.EC0.WRAM <ADDRESS>
0x1called
```
I tried addresses around 0x100, 0x200 and 0x537-53b, but always get the same result. What am I doing wrong with acpi_call?!

afilipovich, 2017-06-12:
@HorstBaerbel, WRAM call requires two arguments: `<memory address>` and `<value to write>`, e.g. `\_SB.PCI0.LPCB.EC0.WRAM 0x537 0x2d`. If you want to double-check if memory was updated, you can read that address with `\_SB.PCI0.LPCB.EC0.RRAM 0x537`

HorstBaerbel, 2017-06-12:
Doh. "RRAM" for reading makes sense...
My default fan table on the UX3410UA reads "0x23 0x34 0x3a 0x42 0x4d 0x54 0x5c 0x5c" -> "35° 52° 58° 66° 77° 84° 92° 92°". Looks a bit strange. Value before and after are 0x0, so this looks quite plausible.
I tried writing 0x2d to 0x537, but my fan does not immediately stop. It seems to work tough. When the temperature reaches 45°, the fan spins up and after the CPU reaches a temperature below 45° it will stay on a short while, then spin down.
The sad thing is that its enough to browse to a different webpage / click a link and the CPU temperature shoots up to 50°, the fan spins up an immediately down again. I guess this is why ASUS chose to pick the 35° lower limit. The fan not spin up / down so often...
The good thing is you can scroll a webpage of watch a youtube video now without the fan spinning...

afilipovich, 2017-06-12:
It takes 10-20 seconds for changes in these values to become effective on my laptop and it also switches the fan on/off when there are load spikes. In the end I set the first threshold to 48C to make this annoyance less frequent.

Also it seems fan reacts not only to CPU but also WiFi module temperature which is 5C hotter than CPU when idling in non-power-save mode.

HorstBaerbel, 2017-06-13:
So what I did for now was adding `acpi_call` to `/etc/modules` and adding a systemd script `fan_table.service` to `/etc/systemd/system/` that sets the lowest fan tipping point temperature:

> [Unit]
> Description=Adjusts fan temperature tipping points using acpi_call
> 
> [Install]
> WantedBy=default.target
> 
> [Service]
> ExecStart=/bin/bash -c '/bin/echo \\"\\\\_SB.PCI0.LPCB.EC0.WRAM 0x537 0x2d\\" | /usr/bin/tee /proc/acpi/call'
> ExecStop=/bin/bash -c '/bin/echo \\"\\\\_SB.PCI0.LPCB.EC0.WRAM 0x537 0x23\\" | /usr/bin/tee /proc/acpi/call'
> Type=oneshot
> RemainAfterExit=yes

(mind the escaping)

Then do a `systemclt daemon-reload`, enable it with `systemctl enable fan_table.service` and test it with `systemctl start fan_table.service`. Then you can reboot and your fan table should be adjusted...

daringer, 2017-06-25:
Sounds like this is a clear candidate to be exposed using asus_fan ...

daringer, 2017-07-22:
ok, let's say it's on the way, #52 was the first step, now #53 then this one...

epiphone, 2017-07-29:
@afilipovich  your replacement for the `_SB.PCI0.LPCB.EC0.SFNV` call works on my UX310UQ too, thank you very much! Changing the thermal tipping table values however seems to have no effect - perhaps the GPU uses another table as you proposed. I'm not sure how to find another fan table on the register, would you have any tips on that?

afilipovich, 2017-07-31:
@epiphone: to find a thermal table I scanned embedded controller memory with RRAM ACPI call in the range from 0x100 to 0xFFF looking for all occurrences of 0x23 which is hex(35) as it seemed that fan started running when sensor reading was reaching 35 degrees.

There was a number of suspect addresses, for each of them I checked adjacent values, in one case it was a vector of 40, 45, 50, ... etc. Then I tried to change these values with WRAM ACPI call and it silenced the fan.

However effects of changes to thermal table are not immediate, for my laptop it takes up to a minute for the fan to completely shut down. Also on a hot day, with power-saving disabled, WiFi module gets as hot as 48C which triggers the fan even if CPU is 43C (if I set first tipping point to 45C).

epiphone, 2017-08-05:
Following the instructions above I found another table at `0x597...0x59e`. Tinkering with the values didn't have any effect on the fan though (even after waiting for a while). I guess there's another mechanism overriding the thermal table. Modifying `asus_fan.c` with @afilipovich's ACPI calls and running `fancontrol` works OK.

daringer, 2017-08-26:
status: first work done for #53, a little boring **(nothing should change, tell me if it breaks something please!)**, but important proof-of-concept for my idea how to do a more granulated (just expose features/functionalities which are probed successfully) module-init in the future...

please test if possible

erkexzcx, 2017-09-16:
I've got a new toy - Asus Zenbook UX430UQ. Unfortunatelly, asus-fan does not seem to work with this model. I will wait until you close this bug and report a new one if it still doesn't work.

daringer, 2017-09-17:
My impression is that the newer ones mainly provide another kind of interface, by mainly allowing to set upper and lower limits, instead of a full-blown manual setting of the fan speed---means chances are high that for you the above stated and explained Ideas could also help. Could you please try to go through some of the ideas earlier in this issue, this will give you an indication if things will be working at some time.

HorstBaerbel, 2018-04-14:
When in doubt how to sign your acpi-call-dkms kernel module for secure boot, see [here](https://lektiondestages.blogspot.de/2018/04/signing-your-kernel-modules-on-ubuntu.html).

HorstBaerbel, 2018-05-24:
After a BIOS update my fan has started spinning up again :(. The fan tipping point table seems to have changed position, but I haven't found out yet where it went. Any hints welcome. I created a little Python script to dump the EC memory values, because other scripts I found weren't working for me. You can find it [here](https://gist.github.com/HorstBaerbel/427a814c263f2d88a9c5092cc7363241).

afilipovich, 2018-05-25:
@HorstBaerbel, whats is your laptop model?
Mine is UX360UAK, I am running the latest firmware v.311 dated 2017-11-29 and tables are at the same location as before.

HorstBaerbel, 2018-05-25:
@afilipovich, UX3410UA (basically a UX410UA). The BIOS version is 308 from here: https://www.asus.com/Laptops/ASUS-ZenBook-UX410UA/HelpDesk_BIOS/
What did you use to scan for the tables. I wrote a Python script, but polling acpi_call sucks because it returns "0x2called" so the script can not say if it returned 0x2 or 0x2c... :/

afilipovich, 2018-05-25:
There is invisible null character between return value and "alled" garbage, so ACPI response looks like `'0x24\x00alled\x00'`. I parsed it with following Python function:
```def to_int(v):
    return int(v.split('\x00')[0], 16)
```
Unfortunately I do not have the whole script I used to scan ACPI table, just a few functions, the rest being IPython session.

You can try to boot cold laptop and register temperature at which fan starts spinning to get the lowest threshold, and then look for this value in ACPI table.

HorstBaerbel, 2018-05-25:
Thanks for the split function. The script works now. You can find it [here](https://gist.github.com/HorstBaerbel/427a814c263f2d88a9c5092cc7363241).
Usage "monitor_ec.py <START_ADRESS> <END_ADRESS> [INTERVAL]", e.g. "monitor_ec.py 0x500 0x5ff".
My EC-RAM now looks like this:
```
 Base   00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 
--------------------------------------------------------
0x0400: 41 53 55 53 E1 00 00 00 00 00 00 00 00 00 00 00 
0x0410: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0420: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0430: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0440: 3C 3C 50 04 A4 01 00 00 00 00 00 00 00 00 00 00 
0x0450: 00 00 00 00 00 07 E0 00 00 00 00 80 00 03 00 00 
0x0460: DA 00 5A 5A 00 5A 00 00 00 DA 00 00 00 00 00 00 
0x0470: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0480: 00 A3 A3 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0490: 00 00 00 00 00 00 00 96 24 90 00 00 00 00 00 00 
0x04A0: 01 C0 00 72 0E 05 00 FF 00 00 00 00 00 00 00 55 
0x04B0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x04C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x04D0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x04E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x04F0: 13 00 00 02 00 00 B1 00 00 00 00 00 00 00 3A 00 
0x0500: 00 50 57 27 02 3D 00 27 1C 00 00 00 FF 57 57 2D 
0x0510: 64 20 20 00 00 27 2D 22 00 00 0F 00 00 00 01 22 
0x0520: 89 85 23 C7 C7 81 00 00 00 00 90 00 00 00 00 00 
0x0530: 6C 67 69 63 62 3A 00 23 30 36 3C 44 4B 54 5C 00 
0x0540: 14 04 05 05 06 06 07 07 57 67 78 8B 9F B6 CB E2 
0x0550: 0A 23 32 3C 01 01 02 05 0A 23 32 3C 01 01 01 01 
0x0560: 01 67 01 64 5A 55 4B 46 C7 50 4B E0 00 00 00 00 
0x0570: 54 27 1C 0A 0A 02 02 00 00 00 00 00 00 57 57 00 
0x0580: 8F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0590: 00 00 00 00 00 00 00 23 30 36 3C 44 4B 54 5C 00 
0x05A0: 14 04 05 05 06 06 07 07 54 64 76 8B 9F B4 CB E2 
0x05B0: 0A 23 32 3C 01 01 02 05 0A 23 32 3C 01 01 01 01 
0x05C0: 03 41 02 CE 02 79 02 39 02 01 01 D4 01 B9 01 9E 
0x05D0: 03 41 02 CE 02 79 02 39 02 01 01 D4 01 B9 01 9E 
0x05E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x05F0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0600: 01 03 03 00 10 00 00 01 3C 01 2C D9 00 00 00 00 
0x0610: 00 02 78 00 00 02 3C 91 01 8A 00 00 00 00 12 00 
0x0620: 89 34 01 00 00 00 00 00 FD 30 F4 FF 00 00 00 00 
0x0630: 71 28 1E 41 53 31 39 4D 4D 45 33 69 42 00 00 00 
0x0640: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0650: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0660: 00 00 00 00 00 00 00 00 00 00 EE 6B F2 53 2B 28 
0x0670: 26 04 00 00 0D 34 0D 66 00 00 00 00 EA 83 00 00 
0x0680: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0690: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x06A0: 0A 01 00 16 44 21 66 2C 88 05 05 05 00 00 00 00 
0x06B0: 77 00 40 80 00 00 06 00 00 00 00 00 00 00 00 00 
0x06C0: 0A 31 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x06D0: 09 01 00 01 FA 32 10 CE 4E 32 3B 3D 14 96 00 00 
0x06E0: 00 24 80 08 00 00 00 00 00 00 00 00 00 00 00 00 
0x06F0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
```
There seems to be a table at 0x537 and 0x597, but changing them does not seem to do anything...
Guess I'll have to wait for a cold evening to test the low temperature limit... ;)

afilipovich, 2018-05-25:
There could have been new mode of operation added... Next thing I would do is look into decompiled DSDT as described here: http://forum.notebookreview.com/threads/fan-control-on-asus-prime-ux31-ux31a-ux32a-ux32vd.705656/
Just in case there is an obvious trigger for fan mode.

HorstBaerbel, 2018-05-25:
Could you take a look at the [decompiled DSDT](https://ufile.io/sgbph)? I got no idea what to look for...

afilipovich, 2018-05-25:
To be honest I do not know what exactly to look for either. Alternative option is to try if following codes work. They allow manual speed control on my laptop:
```
\_SB.PCI0.LPCB.EC0.WRAM 0x0521 0xc5   # full fan speed, locks register 0x97; .TACH method shows current fan speed
\_SB.PCI0.LPCB.EC0.WRAM 0x0521 0x85   # back to automatic fan speed control
\_SB.PCI0.LPCB.EC0.WRAM 0x0521 0x35   # manual fan control. Fan speed can be controlled by .ST84 call.
\_SB.PCI0.LPCB.EC0.ST84 Arg0 Arg1     # Arg0 - fan id (0x00 for CPU fan). Arg1 - speed 0x00 - 0xff
```

bastimeyer, 2018-07-06:
@HorstBaerbel Did you manage to solve this? I was running the python script of @afilipovich for a few months and it was working quite well (increased the threshold even more), but unfortunately I had to send in my device because of a broken display and their repair service updated the BIOS even though I explicitly told them to not do this, because I read your comment here a few weeks ago that it won't work after upgrading. The ASUS page lists this version for UX410UA devices, so I would like to know if you got it to work with version 308 before I attempt a downgrade to 307. Thanks!

HorstBaerbel, 2018-07-11:
No. Sorry. If you find something out, please post here.

HorstBaerbel, 2019-04-28:
I revisited this. The main thing was that **I had to disable secure boot**. Though I had my acpi_call kernel module signed, and used sudo there currently seems to be no way to call it, except as real root (`sudo su`) or with secure boot disabled. This is unfortunate, but [there currently seems to be no other solution](https://stackoverflow.com/questions/55213166/executing-rdmsr-wrmsr-with-secure-boot-on).
I the rescanned the EC RAM. Additionally to the fan dipping tables stating at 0x537 / 0x597 (8 entries) there seems to be a value related to fan speed at 0x397 / 0x398 with values that go from 1-5. These seem to be the current fan speed mode values. Writing 0 to 0x397 stops my fan (`echo \\_SB.PCI0.LPCB.EC0.WRAM 0x397 0x00 | sudo /usr/bin/tee /proc/acpi/call`). It spins up when CPU temperature rises.
I then tried the ACPI calls @afilipovich suggested a while ago:

> \_SB.PCI0.LPCB.EC0.WRAM 0x0521 0xc5   # full fan speed, locks register 0x97; .TACH method shows current fan speed
> \_SB.PCI0.LPCB.EC0.WRAM 0x0521 0x85   # back to automatic fan speed control [...]

and guess what: They all work as described! I then tried the fan dipping table and they also work. Killing the fan by setting 0x397 to 0x00 and then setting 0x537 to e.g. 0x2d makes the fan spin up much later.  

This was good. The bad news was that once the fan has spun up, it never spins down, regardless of the CPU temperature, so I searched on.
And found the fan speed tables corresponding to the temperature values at 0x548 / 0x5A8 (8 entries). Setting 0x548 to 0 will make your fan slowly spin down. When CPU temperature rises it will spin up again.  
This is the EC RAM dump:
```
0x0530: 6C 67 69 63 62 3A 00 23 30 36 3C 44 4B 54 5C 00 
0x0540: 14 04 05 05 06 06 07 07 58 67 76 8B 9F B4 CB E2 
0x0550: 0A 23 32 3C 01 01 02 05 0A 23 32 3C 01 01 01 01 
0x0560: 01 67 01 64 5A 55 4B 46 C7 50 4B E0 00 00 00 00 
0x0570: 54 2E 0F 0A 0A 02 02 00 00 00 00 00 00 58 58 00 
0x0580: 8F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0590: 00 00 00 00 00 00 00 23 30 36 3C 44 4B 54 5C 00 
0x05A0: 14 04 05 05 06 06 07 07 54 64 76 8B 9F B4 CB E2
```
The remaining problem: The seems to be an automatic failsafe that will increase the value of 0x548 to its original value of 0x58 after some time. You set 0x548 and the fan spins down slowly. After reaching the value passed, it will increase again to 0x58 +1\-1 - go figure. I have no idea where that value is read from...

HorstBaerbel, 2019-04-28:
I checked the ACPI calls you mentioned again:
\_SB.PCI0.LPCB.EC0.SFNV 0 -> Error: AE_NOT_FOUND
\_SB.PCI0.LPCB.EC0.TACH 0 -> 0xa16lled
\_SB.PCI0.LPCB.EC0.ST98 0 -> 0x0called
\_SB.PCI0.LPCB.EC0.TH1R 0 -> 0x0called
\_SB.ATKD.QMOD 0 -> Error: AE_NOT_FOUND

Could we teach asus-fan to use the calls @afilipovich mentioned to control the fan(s)? Fan mode, fan speed, CPU temperature should be there...

Btw. The ACPI calls can be tested from the command line by: `echo \\_SB.PCI0.LPCB.EC0.SFNV 0 | sudo /usr/bin/tee /proc/acpi/call && sudo cat /proc/acpi/call; echo -e "\n"`

daringer, 2019-05-05:
On the first glance the answer is: **yes**

My main issue is a (currently) missing Zenbook to test properly, but if you would like I can guide you through. Your force-load output looks promising, means likely not too much to do.

Product model based variations have been on my list for quite some time. Just added a small commit to introduce an enum into the global data struct. Just compile-checked and pushed (evil me). What is roughly missing is:

* determine the non-default hw-variant in `fan_init()` using `DMI_PRODUCT` and set `asus_data->variant` accordingly
* then simply use this information to change the acpi calls like this:

  * `fan_set_auto()` add `if` on `asus_data.variant` to use `WRAM` instead of `SFNV`
  * `temp1_input()` add `if` on `...variant` to use `TH0R` instead of `TH1R`
  * `fan_set_speed()` add `if` on`...variant` to use `ST84` instead of `SFNV`

* then worst case the bad probing inside `fan_init()` might hide the `hwmonX/*` files, forcing should work, also easily fixed then

best...

HorstBaerbel, 2019-05-06:
Thanks. I'll try it out this week when I find time.

HorstBaerbel, 2019-05-11:
The module is now working on my UX3410UA. See my (preliminary) commit [here](https://github.com/HorstBaerbel/asus-fan/commit/92e706f6d82e4a1c1e185ef06ee380bc2b5f216e).  
I would suggest I clean the commit up and add a system where "models" can be defined. These would be seperate .h/.c files aka (atm) "model_default" and "model_ux410uak" that contain the all the functions like _fan_set_auto() and the like for that specific machine. They are all registered in "asus_fan.c" and used according to the machine detected. This will be a rather big change, but imo it makes sense to keep "asus_fan.c" clean and enable support of different laptop models.

daringer, 2019-05-24:
nice to see it working, great stuff...

the models system sounds like a pretty good idea to have a clean separation for the models.
Concerning `thermald` I also haven't used it with auto mode, only more or less manual, might be possible, but admittedly I haven't been using `thermald` for quite some time now.

HorstBaerbel, 2019-05-26:
Just a conceptional question: What is fan auto mode supposed to do? Reset the fan to its default behaviour / EC control?

daringer, 2019-05-26:
Yes,
for my last Zenbook (UX32VD) one had to first disable "auto" fan management so that setting an actual fan speed had an impact onto the fan's RPM. Once auto was switched on again, the manual settings got overwritten and the internal fan controller is in charge again.

HorstBaerbel, 2019-05-27:
OK. This is somehing in between manual and auto mode... What should I call it, should I add another sysfs entry?

daringer, 2019-05-31:
I would say this is "auto", but adjustable. 
"non-auto" in my understanding translates to: "controlled exclusively in/from user-space"
"auto" translates to: "controlled in kernel-space (within given parameters)"

dominiksalvet, 2019-06-04:
@afilipovich So far I didn't find any better way how to directly say thank you as you have made possible creating my [asus-fan-control](https://github.com/dominiksalvet/asus-fan-control) repository, mainly focused on my ASUS ZenBook UX430UA and later being extended to support ASUS ROG Strix GL553VD as well. The very core of the mentioned project stands on [this comment](https://github.com/daringer/asus-fan/issues/44#issuecomment-307589414) in this issue. So **thank you**! :smiley:

@daringer I am sorry for the "spam" in this issue, but @afilipovich has no public email in his GitHub profile and I want to let him know.

afilipovich, 2019-06-04:
Thank you @dominiksalvet! I still remember the day when that annoying fan went quiet :)

klausenbusk, 2019-08-03:
> I found thermal tipping point table in embedded controller RAM. It is between addresses 0x537. 0x53e.

I can confirm it works on UX430UNR (54 58 62 66 72 78 88 96).

> So what I did for now was adding `acpi_call` to `/etc/modules` and adding a systemd script `fan_table.service` to `/etc/systemd/system/` that sets the lowest fan tipping point temperature:

FWIW: Using systemd-tmpfiles works perfectly:
```
$ cat /etc/tmpfiles.d/fan.conf
w /proc/acpi/call - - - - \\_SB.PCI0.LPCB.EC0.WRAM 0x537 0x3a # 58 degree
``` 

> This is good. The bad news is that once the fan has spun up again, it never spins down completely, regardless of the CPU temperature, so I searched on.

On my laptop the fan spins up at 54° and spins down at ~40°, but if i change the first tipping point to 58° the fan spins down at ~45°. Does the controller use some kind of sliding offset for spin down? maybe?

---

A few test scripts, maybe they are useful for someone.

Read all tipping points (0x537-0x53e)
```
#!/bin/bash

# 0x537-0x53e
for i in {1335..1342}; do
	HEX="$(printf '%x\n' $i)"
	echo "\_SB.PCI0.LPCB.EC0.RRAM 0x${HEX}" > /proc/acpi/call
	VALUE="$(cat /proc/acpi/call | sed 's/\x0.*//g')"
	echo $((16#${VALUE#0x}))
done
```

Read all (?) EC memory
```
#!/bin/bash

# 0x0-0xfff
for i in {0..4095}; do
	HEX="$(printf '%x\n' $i)"
	echo "\_SB.PCI0.LPCB.EC0.RRAM 0x${HEX}" > /proc/acpi/call
	VALUE="$(cat /proc/acpi/call | sed 's/\x0.*//g')"
	echo -e "0x${HEX}\t$((16#${VALUE#0x}))"
done

```

klausenbusk, 2019-08-03:
Not related to fan control at all, but I think some of you could find the information useful.

Most newer Zenbooks supports battery charge control (https://www.asus.com/us/support/FAQ/1032726/).

So I did some tinkering and found the relevant address (Zenbook UX430UNR, bios 308):
```
0x3af	100 # start charge threshold
0x3b0	95 # current battery percent
```
I use the following, to limit the battery to max 80%:
```
$ cat /etc/tmpfiles.d/battery.conf 
w /proc/acpi/call - - - - \\_SB.PCI0.LPCB.EC0.WRAM 0x3af 0x50 # 80%
```

klausenbusk, 2019-08-05:
> After much tinkering with decompiled DSDT I found ACPI calls which can replace missing ones:

Support for newer Zenbooks should be supported out-of-the-box when [this patch](http://git.infradead.org/linux-platform-drivers-x86.git/commitdiff/e3168b874321d04c160c9eb937919eb926ae232f?hp=f1fc032192d2872caef69fb899e939bf7cc24fc7) gets merged (5.4 or maybe 5.3).

dominiksalvet, 2020-02-24:
@daringer Oops, here we go again... but you already know the email address thing, don't you? Sorry for that... still could not help myself. :cry:

@klausenbusk Hey man! Do you know about [asus-fan-control](https://github.com/dominiksalvet/asus-fan-control)? It does basically the same as your script for setting temperature does but in a more sophisticated and consistent way - it sets all temperatures, performs various checks, it supports a decent amount of ASUS laptop models and much more. Thanks to @agura-lex we also have systemd integration and an [AUR package](https://aur.archlinux.org/packages/asus-fan-control) for our dear Arch users. Your model name seems to be very similar to mine (UX430UA), so there is a decent chance it will work out of the box (but explicitly adding it as tested would be really great). Please, consider giving asus-fan-control a try. :smile: 

@klausenbusk BTW thank you so much for the code snippets above. :heart: Some people have reported me that asus-fan-control did not work on their devices and I have been thinking a bit what to do with that. Once I had seen your snippets, I got an idea and I have created [afc-scout](https://github.com/dominiksalvet/afc-scout), which you can basically use to find your base ACPI fan address by reading a given ACPI address range. Yep, you need to process it manually but it is certainly better than nothing! :rocket:

daringer, 2020-03-08:
HAHA, this is pretty next level: direct-marketing at the *competitor*, at least you come over to visit [asus-fan](https://github.com/daringer/asus-fan) - but seriously: no worries, it's not like this is a competition for the highest number of hardware-specific adaptations within some piece of code :nerd_face: 

Personally I believe hardware should not be controlled by bash scripts (too often) :D so it will be tough for you to get me to [asus-fan-control](https://github.com/dominiksalvet/asus-fan-control) and if your project gets better because of the discussions/investigations here, it's already win-win. Despite the fact that [afc-scout](https://github.com/dominiksalvet/afc-scout) will be helpful to support more models, so keep up the good work. At least until I have a zenbook again :+1: :weary:

dominiksalvet, 2020-03-09:
@daringer Glad to hear that! I have no interest to compete with [asus-fan](https://github.com/daringer/asus-fan). We both develop software that does basically the same (except for two fans support) but **we use different approaches**.

I understand your skepticism about "bash scripts" but keep in mind that my scripts are almost without an exception **POSIX shell scripts** and so they run practically everywhere without any special dependencies (even on macOS and even these days). I also realize the fact that shell scripts may be very fragile and hence I really keen on mechanisms preventing those issues - AND/OR lists, no command substitution in another command's parameter, no pipes (except with echo), no non-POSIX command options, functions, which modify calling shell's environment, are strictly put into their own subshells, etc.

BTW, I really wouldn't post here if @klausenbusk had an email address on his GitHub profile. For the sake of completeness, @afilipovich really did not have an email address on his profile that time (in case you miss it)... He just reacted quickly and added it. :smile: 

Hope you will be happy with a new ZenBook! :rocket:

klausenbusk, 2020-03-09:
> BTW, I really wouldn't post here if @klausenbusk had an email address on his GitHub profile. For the sake of completeness, @afilipovich really did not have an email address on his profile that time (in case you miss it)... He just reacted quickly and added it. smile

Finding a email address for a Github user isn't hard if they have some public activity :)
[image]

> @klausenbusk Hey man! Do you know about [asus-fan-control](https://github.com/dominiksalvet/asus-fan-control)? It does basically the same as your script for setting temperature does but in a more [...]

It is hard to compete with `w /proc/acpi/call - - - - \\_SB.PCI0.LPCB.EC0.WRAM 0x537 0x3a # 58 degree`, I prefer to keep it simple and `tmpfiles.d` is just simpler. It is a cool tool though!

> Your model name seems to be very similar to mine (UX430UA), so there is a decent chance it will work out of the box (but explicitly adding it as tested would be really great). Please, consider giving asus-fan-control a try. smile

The `base fan control address` is located at the same location (`1335`) so I assume it will work. BTW: I think you should move the fallback addr behind a `--yes-i-know-what-im-doing` flag, writing to "random" EC memory locations can be dangerous!

Out of curiosity: Are you experiencing throttling on your model? My CPU was capped at ~8w (IIRC) until I did: https://github.com/erpalma/throttled/tree/88216b119ad512d6d345e66dd304ba7b6b3e93c0#static-fix

dominiksalvet, 2020-03-10:
@klausenbusk Thanks man! Didn't know about the email address. :+1: I would gladly add your model as tested but since you haven't tried it with asus-fan-control, I can't do so. :cry: But I completely understand your argument of simplicity and I respect it! :smile: Still, should you change your mind, make sure you will add your model, please.

Here is the thing with writing to "random" EC addresses... The fact that there is no such flag from the very beginning of asus-fan-control is probably the main reason why asus-fan-control is not called ux430ua-fan-control as was initially... BTW, I think that afc-scout brings at least something to the table in this matter... :smile:

I do not notice any symptoms of CPU throttling on my device. BTW, I have i5-7200U and default UX430UA asus-fan-control configuration... :rocket: 

---

Ok, I am truly sorry @daringer for the spam and since now I know how to reach people even without public GitHub email, this is most likely my last post here. Also, feel free to remove my posts here (e.g., in a week) as their messages have been delivered and I must admit that they do not contribute to this issue very much. Nevertheless, thank you so much for the given space! :heart:

daringer, 2020-03-10:
@dominiksalvet, absolutly ok, it drives things/projects forward, you have my full support :D

[1 more reply]