daringer/asus-fan #48: remove exec bit from asus-fan.service

saultdon, 2017-06-01:
Looks like systemd is complaining about the asus-fan.service file and requests that the executable bit be removed.

```
Jun 01 15:52:34 archasus systemd[1]: Configuration file /usr/lib/systemd/system/asus-fan.service is marked executable. Please remove executable permission bits.
```

daringer, 2017-06-27:
assuming Arch Linux now, or are you on Ubuntu? Thought this is clear and easy, but cannot reproduce actually:

mmh, strange in the `PKGBUILD` the install shall happen with mode 664 (no exec bit):
```bash
install -m 664 misc/systemd/asus-fan.service ${pkgdir}/usr/lib/systemd/system/asus-fan.service
```
also the file inside the repository is not +x
can you please try re-installing? and make sure your old asus-fan.service is removed before you kick-on the dkms install or whatever you use....

daringer, 2017-06-27:
looks like the PKGBUILD in AUR was updated on May 25th with this issue, cross checked locally, thus closing, ...

closed 2017-06-27 by daringer