daringer/asus-fan #49: [UNSUPPORTED HARDWARE:  UX360UAK] - Can't run make

jonnyteronni, 2017-06-03:
Trying to get this working on a UX360UAK:

$ make
make -C /lib/modules/4.11.3-1-ARCH/build M=$PWD modules
make[1]: Entering directory '/usr/lib/modules/4.11.3-1-ARCH/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/4.11.3-1-ARCH/build'
make: *** [Makefile:14: all] Error 2

afilipovich, 2017-06-10:
I guess "make" does not work for you because of non-default Makfile name.

PKGBUILD runs make as "make -f Makefile.dkms"

daringer, 2017-06-27:
Weeeird, especially on Arch this should work without issues, just `make` shall use the kernel headers from the `/lib/modules/<kernelversion>/build` and build the module. As far as I know I suppose there are the following possibilities that this fails:
- you just ran `pacman -Syu` or updated your kernel, thus `uname -a` points to the still running kernel, but its sources are already cleaned up---means not existing anymore
- the "linux-headers" package is not installed? (is this even possible)

For me running `make` looks like this:
```bash
|-$ make
make -C /lib/modules/4.11.2-1-ARCH/build M=$PWD modules
make[1]: Entering directory '/usr/lib/modules/4.11.2-1-ARCH/build'
  CC [M]  /home/<user>/dev/asus-fan/asus_fan.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/<user>/dev/asus-fan/asus_fan.mod.o
  LD [M]  /home/<user>/dev/asus-fan/asus_fan.ko
make[1]: Leaving directory '/usr/lib/modules/4.11.2-1-ARCH/build'
```

daringer, 2017-08-25:
seems to be stalled or resolved, please feel free to re-open if issue still exists...

closed 2017-08-25 by daringer