daringer/asus-fan #57: UX330UAR Help RPM FAN

mauro25987, 2017-12-11:
Hi daringer,
How are you?
I have an asus UX330UAR and I want to see the rpm of the CPU fan.
I can not see it with lm-sensors and I found your project.
I'm not sure if with asus_fan I can see the RPM.
Could you help me? or guide, what should I do?
I have done some tests but they have thrown me error.
How do I delete the asus_fan to start back?

root@mgonzalez-UX330UAR:~# sudo modprobe asus_fan force_load=1
root@mgonzalez-UX330UAR:~# dmesg | grep asus
[18418.792180] asus-fan (init) - dmi sys info: 'ASUSTeK COMPUTER INC.'
[18418.792181] asus-fan (init) - dmi product: 'UX330UAR'
[18418.792182] asus-fan (init) - forced loading of module: USE WITH CARE
[18418.792429] asus-fan (init) - created hwmon device: hwmon6
[18418.792430] asus-fan (init) - finished init, found 2 fan(s) to control
root@mgonzalez-UX330UAR:~# sudo sensors
iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +32.0°C  

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +34.5°C  

acpitz-virtual-0
Adapter: Virtual device
temp1:        +38.0°C  (crit = +103.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +38.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +38.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +37.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +38.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +36.0°C  (high = +100.0°C, crit = +100.0°C)

does not output the RPM fan CPU

daringer, 2018-01-28:
urgs late, sry...
you should check where your hwmon directory is and what's created there (readable? writeable?)
to remove asus_fan, if not dkms or package-manger-driven install 
```find /lib/modules/`uname -r` -name "*asus_fan*"``` might help to find remainers...

mauro25987, 2018-01-28:
already erase it, how are the steps to install it?
this model has a fan on the cpu, this package serves to see the information of the fan's rpm?
thanks for you help

daringer, 2018-03-29:
mh, I believe I do not fully understand the point, apparently you had installed it before? or how would you've been generating these dmesg outputs above, redoing the steps above will then likely be the installation procedure. 

Nevertheless, it also might be that we see an incompatibility here, the UX3300UAR as reported in issue #57 does also not work as intended, the solution is also already proposed within #57 and #56. I currently suffer from the fact that I have no zenbook available as easy as before, thus this story will at least take until I have my next one (which is planned for sure)...

I'll close this, feel free to re-open,

closed 2018-03-29 by daringer