daringer/asus-fan #59: fake RPM?! (4 small issues)

RayOfLight1, 2018-03-05:
Thank you for this project, I've gained control of both fans thanks to it, but when I check its fan speed, it is completely made up (I've read in your source code that you are kind of aware of this situation), I can set pwm2 to 100 and it reports some speed (about 2277 rpm), but in truth, my fan is not even spinning (not perfect bearings, eh?), such a let down (I'll replace it when I can).

Then I discovered "calc_fan_relation.py": [ISSUE 1]

```
sudo ./calc_fan_relation.py 2 /sys/class/hwmon/hwmon5
-----------------------------------collecting fan info:
Traceback (most recent call last):
  File "./calc_fan_relation.py", line 88, in <module>
    rpms = get_rpm(vals[0][1], vals[1][1])
  File "./calc_fan_relation.py", line 71, in get_rpm
    ( (0x0041CDB4 / (fan2*2)) if fan2 is not None else 0) )
ZeroDivisionError: integer division or modulo by zero
```

It appears that you have not considered cases in where fan speed can be really zero.

Also, check hwmon5 !!, every time I reload the module (enabling and disabling debug, for example, it creates a new hwmon instance, making its settings unusable for fancontrol, forcing me to reboot the laptop.  [ISSUE 2]

Also, I get 7 warnings when compiling it, one:
`warning: ignoring return value of ‘sysfs_create_group’, declared with attribute warn_unused_result [-Wunused-result]`

and 6 on:
`warning: ignoring return value of ‘kstrtouint’, declared with attribute warn_unused_result [-Wunused-result]`

just saying that you are 7 lines away from a warning-less perfect code :)  [ISSUE 3]

My main motivation for typing this out, though, is if there is any possibility of finding the real fan speed (my laptop is an **UX32VD** btw)  [ISSUE 4]

Thank you again.

daringer, 2018-03-29:
Hmm, ok I have (had) perfectly the same Zenbook model, so yes as what I could find out there is no proper way to calculate the fan speed so that it will be _really_ true. OK but let's go through one after another:

(1) jup, totally, haven't thought that anyone will ever use this, had still no really useful values based on it (WONTFIX) :smile:

(2) yes this is known, at least on my side, looks like the module is not unloading correctly, or did you force unload? here also, likely not painful enough to invest time solving it :disappointed: 

(3) uh, but this is a good point, this very much looks like a bug during the *detection routines*, likely a zero is not understood as "no rpm" but instead as an error. Like the opposite of what happens at (2)---still I have created #60 should be an easy bug to fix.
In the meantime you might want to simple force load the module during boot-up, the probing or detection will then be skipped and all should work as expected also on cold days.

(4) uuuh, this is something you can lure me easily with, also easy bugfixing/codecleaning.
Feel free to provide a PR, would be really happy to see contributions on those things, especially as I am right now without a zenbook to test with, but this will also change sooner or later...

RayOfLight1, 2018-08-13:
(2) I have solved [ISSUE 2] thanks to "close2" from [archlinux forums](https://bbs.archlinux.org/viewtopic.php?pid=1415552#p1415552) and [linuxquestions.org](https://www.linuxquestions.org/questions/slackware-14/hwmon-keeps-changing-on-reboots-and-knocks-out-fancontrol-can't-figure-out-the-wildcard-4175583395/)

I've changed my fancontrol script from static hwmon numbering to a wildcard one, my code is as follows:
```
INTERVAL=5

DEVPATH=hwmon1=devices/platform/coretemp.0      hwmon3=devices/platform/asus_fan
DEVNAME=hwmon1=coretemp                                         hwmon3=asus_fan

FCTEMPS=hwmon3/pwm2=hwmon1/temp1_input          hwmon3/pwm1=hwmon1/temp1_input
FCFANS=hwmon3/pwm2=hwmon3/fan2_input            hwmon3/pwm1=hwmon3/fan2_input

MINTEMP=hwmon3/pwm2=55                                          hwmon3/pwm1=55
MAXTEMP=hwmon3/pwm2=80                                          hwmon3/pwm1=80

MINSTART=hwmon3/pwm2=60                                         hwmon3/pwm1=60
MINSTOP=hwmon3/pwm2=50                                          hwmon3/pwm1=50
```
to the following: (remember to drop devpath and devname)
```
INTERVAL=5

FCTEMPS=/sys/devices/platform/asus_fan/hwmon/hwmon[[:print:]]*/pwm2=/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_input           /sys/devices/platform/asus_fan/hwmon/hwmon[[:print:]]*/pwm1=/sys/devices/platform/coretemp.0/hwmon/hwmon1/temp1_input
FCFANS=/sys/devices/platform/asus_fan/hwmon/hwmon[[:print:]]*/pwm2=/sys/devices/platform/asus_fan/hwmon/hwmon[[:print:]]*/fan2_input            /sys/devices/platform/asus_fan/hwmon/hwmon[[:print:]]*/pwm1=/sys/devices/platform/asus_fan/hwmon/hwmon[[:print:]]*/fan2_input

MINTEMP=/sys/devices/platform/asus_fan/hwmon/hwmon[[:print:]]*/pwm2=55                                          /sys/devices/platform/asus_fan/hwmon/hwmon[[:print:]]*/pwm1=55
MAXTEMP=/sys/devices/platform/asus_fan/hwmon/hwmon[[:print:]]*/pwm2=80                                          /sys/devices/platform/asus_fan/hwmon/hwmon[[:print:]]*/pwm1=80

MINSTART=/sys/devices/platform/asus_fan/hwmon/hwmon[[:print:]]*/pwm2=60                                         /sys/devices/platform/asus_fan/hwmon/hwmon[[:print:]]*/pwm1=60
MINSTOP=/sys/devices/platform/asus_fan/hwmon/hwmon[[:print:]]*/pwm2=50                                          /sys/devices/platform/asus_fan/hwmon/hwmon[[:print:]]*/pwm1=50
```

Now I can load and unload the module and restore proper fancontrol without having to reboot (I'm at hwmon10 right now...) :)

daringer, 2019-06-21:
mmmh, 8months, at least the repository has changed a little, means especially the init-stuff and robustness has been improved, despite I can now see an own Zenbook again on the horizon, means new things will happen (tm) ... 
anyways, closing this issue, feel free to re-open or re-new depending on the issues

closed 2019-06-21 by daringer