daringer/asus-fan #62: Support for Asus UX490UAR

zauwn, 2018-05-03:
Hello,

I've compiled the module but it is not working:

may 03 21:52:45 kernel: asus-fan (debug) - starting initialization...
may 03 21:52:45 kernel: asus-fan (init) - dmi sys info: 'ASUSTeK COMPUTER INC.'
may 03 21:52:45 kernel: asus-fan (init) - dmi product: 'UX490UAR'
may 03 21:52:45 kernel: asus-fan (debug) - dmi chassis type: '10'
may 03 21:52:45 kernel: asus-fan (debug) - fan-id: 0 | get RPM
may 03 21:52:45 kernel: asus-fan (debug) - |--> get RPM using acpi
may 03 21:52:45 kernel: asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
may 03 21:52:45 kernel: asus-fan (debug) - |--> acpi request returned: 0
may 03 21:52:45 kernel: asus-fan (init) - fan-id: 1 | success getting rpm
may 03 21:52:45 kernel: asus-fan (debug) - fan-id: 1 | get RPM
may 03 21:52:45 kernel: asus-fan (debug) - |--> get RPM using acpi
may 03 21:52:45 kernel: asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
may 03 21:52:45 kernel: asus-fan (debug) - |--> acpi request returned: 0
may 03 21:52:45 kernel: asus-fan (init) - fan-id: 2 | failed to get rpm
may 03 21:52:45 kernel: asus-fan (init) - temp-id: 1 | success getting temp
may 03 21:52:45 kernel: asus-fan (debug) - fan-id: (both) | set max speed: 255, force reset: 0
may 03 21:52:45 kernel: asus-fan (debug) - fan_set_max_speed() call succeeded, ret: 0
may 03 21:52:45 kernel: asus-fan (debug) - fan-id: (both) | set to automatic mode
may 03 21:52:45 kernel: asus-fan (set_auto) - failed reseting fan(s) to auto-mode! errcode: 5 - DANGER! OVERHEAT? DANGER!
may 03 21:52:45 kernel: asus-fan (init) - set auto-mode speed to active, failed! errcode: 5

The module is not loaded properly. If you could point me in the right direction I could perform some tests and try to add support to this Asus Model.

If you need more information let me know.

Regards,

jarossi, 2018-10-29:
+1 on this
Also there are juicy details on debugging at:

https://github.com/daringer/asus-fan/issues/39

klarose, 2018-12-04:
My UX490UAR has been overheating on me when compiling, so I've been digging into this a bit.
```
$ for path in $(cat /tmp/acpi_paths.txt); do echo -n "$path: "; do_acpi_call $path; echo; done
\_SB.PCI0.LPCB.EC0.SFNV: Error: AE_NOT_FOUND
\_SB.PCI0.LPCB.EC0.TACH: 0xf18lled
\_SB.ATKD.QMOD: Error: AE_NOT_FOUND
\_SB.PCI0.LPCB.EC0.ST98: 0x0called
\_SB.PCI0.LPCB.EC0.SFNV: Error: AE_NOT_FOUND
\_SB.PCI0.LPCB.EC0.TH1R: 0x0called
```

I'm running Ubuntu 18.04.

It looks like the tachometer is working. Just need to figure out how to disable auto mode...

klarose, 2018-12-05:
The speed setting is here:

sudo ~/bin/do_acpi_call '\_SB_.PCI0.LPCB.EC0_.TSTP'

I figured this out one out by using  https://github.com/hirschmann/nbfc/ with the UX430UA configuration. That configuration points to 0x97 as the EC0 register for the speed configuration, and it seems to work.

The configuration *also* writes two other registers when writing the fan speed, which I suspect is how they disable auto mode. 0xA0 and 0xA6. 

my DSDT claims that 0xA6 is TH0R, while 0xA0 seems to be unnamed (unless I'm misinterpreting the Offset lines).

```
                Offset (0x93), 
                TAH0,   16, 
                TAH1,   16, 
                TSTP,   8, 
                Offset (0x9C), 
                CDT4,   8, 
                CDT5,   8, 
                Offset (0xA0), 
                Offset (0xA1), 
                Offset (0xA2), 
                Offset (0xA3), 
                EACT,   8, 
                TH1R,   8, 
                TH1L,   8, 
                TH0R,   8, 
                TH0L,   8, 
```
nbfc seems to be controlling the fan well enough for me, so I'm going to move on. Maybe somebody can figure out how to make this work better. FWIW, it looks to me like the DSDT on the UX490UAR is just incomplete.

dominiksalvet, 2019-06-12:
@zauwn @jarossi @klarose Hello guys. Do you still have problems with UX490UAR fan control? I maintain [asus-fan-control](https://github.com/dominiksalvet/asus-fan-control) repository and it supports multiple ASUS notebook models and unfortunately your model is not among them. However, if you are brave enough, you could try to install it and see whether it will work. In case of unrecognised model (which is your case), it will use the default temperature values of my UX430UA model. Nevertheless, if you succeed, I would add your model as tested and asus-fan-control will recognise it automatically.

It uses ACPI interface to write 8 temperature values (representing boundaries dividing fan speed levels) from base address 1335 decimal, 537 hexadecimal. Please, be careful and consider all possible consequences before installing asus-fan-control.

One wouldn't believe that such a thin notebook has a fan. :smiley:

daringer, 2019-06-21:
Yes, I know development is not happening currently, mainly still due to a non existing Zenbook on my side. Nevertheless the control concepts seem to be very similar with most if not all the recent Zenbooks. 

Means, you should clearly take a look at #44 which despite its huge history also brought up @HorstBaerbel, who is currently realizing some of the ideas how to have core parts of asus-fan more flexible in order to support a wider variety of ACPI incarnations. There is already something to test and what I've seen it's running, adapting this towards UX490UAR will be quite straight forward once there is some more progress on the transition.