daringer/asus-fan #64: Asus VivoBook Pro X580VD support

pezcurrel, 2018-09-12:
Hello,
it would be great if the module supported this notebook too.
Here is the dmesg output, I'd like to help more...

`
[ 2961.170710] asus-fan (debug) - starting initialization...
[ 2961.170713] asus-fan (init) - dmi sys info: 'ASUSTeK COMPUTER INC.'
[ 2961.170715] asus-fan (init) - dmi product: 'X580VD'
[ 2961.170716] asus-fan (debug) - dmi chassis type: '10'
[ 2961.170719] asus-fan (debug) - fan-id: 0 | get RPM
[ 2961.170720] asus-fan (debug) - |--> get RPM using acpi
[ 2961.170721] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
[ 2961.171136] asus-fan (debug) - |--> acpi request returned: 0
[ 2961.171138] asus-fan (init) - fan-id: 1 | success getting rpm
[ 2961.171139] asus-fan (debug) - fan-id: 1 | get RPM
[ 2961.171140] asus-fan (debug) - |--> get RPM using acpi
[ 2961.171142] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
[ 2961.171479] asus-fan (debug) - |--> acpi request returned: 0
[ 2961.171481] asus-fan (init) - fan-id: 2 | success getting rpm
[ 2961.171482] asus-fan (init) - temp-id: 1 | success getting temp
[ 2961.171484] asus-fan (debug) - fan-id: (both) | set max speed: 255, force reset: 0
[ 2961.173466] asus-fan (debug) - fan_set_max_speed() call succeeded, ret: 0
[ 2961.173467] asus-fan (debug) - fan-id: (both) | set to automatic mode
[ 2961.173472] asus-fan (set_auto) - failed reseting fan(s) to auto-mode! errcode: 5 - DANGER! OVERHEAT? DANGER!
[ 2961.173475] asus-fan (init) - set auto-mode speed to active, failed! errcode: 5
`

cdutsov, 2019-01-20:
I can confirm this behaviour.

`[11504.082332] asus-fan (init) - dmi sys info: 'ASUSTeK COMPUTER INC.'`
`[11504.082332] asus-fan (init) - dmi product: 'X580VD'`
`[11504.082524] asus-fan (init) - fan-id: 1 | success getting rpm`
`[11504.082652] asus-fan (init) - fan-id: 2 | success getting rpm`
`[11504.082652] asus-fan (init) - temp-id: 1 | success getting temp`
`[11504.084356] asus-fan (set_auto) - failed reseting fan(s) to auto-mode! errcode: 5 - DANGER! OVERHEAT? DANGER!`
`[11504.084358] asus-fan (init) - set auto-mode speed to active, failed! errcode: 5`

daringer, 2019-06-21:
You might try the git version, lately there have been some contributions addressing these issues.

cdutsov, 2019-06-22:
I have tried the latest version but the problem persist. I have compiled with DEBUG=1. Here is the log:

`[654965.614752] asus-fan (debug) - starting initialization...
[654965.614755] asus-fan (init) - dmi sys info: 'ASUSTeK COMPUTER INC.'
[654965.614756] asus-fan (init) - dmi product: 'X580VD'
[654965.614757] asus-fan (debug) - dmi chassis type: '10'
[654965.614758] asus-fan (debug) - fan-id: 0 | get RPM
[654965.614759] asus-fan (debug) - |--> get RPM using acpi
[654965.614759] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
[654965.615072] asus-fan (debug) - |--> acpi request returned: AE_OK
[654965.615073] asus-fan (init) - fan-id: 1 | success getting rpm
[654965.615074] asus-fan (debug) - fan-id: 1 | get RPM
[654965.615074] asus-fan (debug) - |--> get RPM using acpi
[654965.615075] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
[654965.615276] asus-fan (debug) - |--> acpi request returned: AE_OK
[654965.615277] asus-fan (init) - fan-id: 2 | success getting rpm
[654965.615277] asus-fan (init) - temp-id: 1 | success getting temp
[654965.615278] asus-fan (debug) - fan-id: (both) | set max speed: 255, force reset: 0
[654965.617953] asus-fan (debug) - fan_set_max_speed() call succeeded, ret: AE_OK
[654965.617954] asus-fan (debug) - fan-id: (both) | set to automatic mode
[654965.617963] asus-fan (set_auto) - failed reseting fan(s) to auto-mode! errcode: AE_NOT_FOUND - DANGER! OVERHEAT? DANGER!
[654965.617967] asus-fan (init) - set auto-mode speed to active, failed! errcode: AE_NOT_FOUND`
`

I will be glad to help with something if you need more info on the issue.