daringer/asus-fan #65: No fan under sensors (UX410UA, elementary OS 5 Juno, based on Ubuntu 18.04 LTS)

xtelcz, 2018-11-08:
Hi, I am an absolute Linux beginner :-)
Trying to get this work on my Asus UX410, but I cannot see any fan in `sensors` listing.
Please, see my config below and get me any clue, how to make asus-fan works. Thank you!

>Note: I have tried to do a standard install `ubuntu_dkms_sudo_install.sh`, but this not worked for me ` # modprobe asus_fan / modprobe: ERROR: could not insert 'asus_fan': No such device`, so I have  made a manual install,  `make, install, ...`. No secure boot on my notebook.

```
petr@UX410:~$ sudo modprobe asus_fan force_load=1
petr@UX410:~$ lsmod | grep asus
asus_fan               16384  0
asus_nb_wmi            28672  0
asus_wmi               28672  1 asus_nb_wmi
sparse_keymap          16384  1 asus_wmi
asus_wireless          16384  0
wmi                    24576  2 asus_wmi,wmi_bmof
video                  45056  2 asus_wmi,i915

petr@UX410:~$ sensors
iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +33.0°C  

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +32.5°C  

acpitz-virtual-0
Adapter: Virtual device
temp1:        +36.0°C  (crit = +103.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +35.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +34.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +34.0°C  (high = +100.0°C, crit = +100.0°C)

petr@UX410:~$ sudo /usr/src/asus_fan-master/misc/test_module.sh force_load=1
[i] ------------ Starting 'asus-fan' kernel module quick test
[+] asus-fan loaded! REMOVING (rmmod) asus-fan (asus_fan) module
[i] Clearing 'dmesg' (kernel) log
[i] Trying to modprobe the module: 'asus_fan'
[+++] Successfully inserted module: 'asus_fan'

[i] ------------ Inspecting dmesg a.k.a. kernel log
[+++] 'dmesg' reports finished module init
[i] Remaining dmesg lines with related information:
[  283.141155] asus-fan (debug) - starting initialization...
[  283.141157] asus-fan (init) - dmi sys info: 'ASUSTeK COMPUTER INC.'
[  283.141157] asus-fan (init) - dmi product: 'UX410UAK'
[  283.141158] asus-fan (debug) - dmi chassis type: '10'
[  283.141158] asus-fan (init) - forced loading of module: USE WITH CARE
[  283.141158] asus-fan (debug) - register asus fan driver
[  283.141209] asus-fan (debug) - probe for device
[  283.141210] asus-fan (debug) - init hwmon device
[  283.141244] asus-fan (debug) - fan-id: 0 | get RPM
[  283.141245] asus-fan (debug) - |--> get RPM using acpi
[  283.141245] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
[  283.141386] asus-fan (debug) - |--> acpi request returned: 0
[  283.141386] asus-fan (debug) - fan-id: 0 | get RPM
[  283.141387] asus-fan (init) - created hwmon device: hwmon6

[i] ------------ Search /sys files
[i] Checking for fan1 /sys/ files:
[+] Found /sys path (/sys/devices/platform/asus_fan)
[+] Found /sys path (/sys/devices/platform/asus_fan/hwmon/hwmon6)
[---] Could not find the correct /sys path! (/sys/devices/platform/asus_fan/hwmon/hwmon6/fan1_input)

petr@UX410:~$ sudo insmod asus_fan.ko; dmesg | tail
insmod: ERROR: could not load module asus_fan.ko: No such file or directory
[  283.141158] asus-fan (debug) - register asus fan driver
[  283.141209] asus-fan (debug) - probe for device
[  283.141210] asus-fan (debug) - init hwmon device
[  283.141244] asus-fan (debug) - fan-id: 0 | get RPM
[  283.141245] asus-fan (debug) - |--> get RPM using acpi
[  283.141245] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
[  283.141386] asus-fan (debug) - |--> acpi request returned: 0
[  283.141386] asus-fan (debug) - fan-id: 0 | get RPM
[  283.141387] asus-fan (init) - created hwmon device: hwmon6
[  283.141387] asus-fan (init) - finished init, found 2 fan(s) to control

petr@UX410:~$ ls /sys/class/hwmon/hwmon6
device  name  power  subsystem  uevent

petr@UX410:/usr/src/asus_fan-master$ dmesg | grep asus
[  283.141155] asus-fan (debug) - starting initialization...
[  283.141157] asus-fan (init) - dmi sys info: 'ASUSTeK COMPUTER INC.'
[  283.141157] asus-fan (init) - dmi product: 'UX410UAK'
[  283.141158] asus-fan (debug) - dmi chassis type: '10'
[  283.141158] asus-fan (init) - forced loading of module: USE WITH CARE
[  283.141158] asus-fan (debug) - register asus fan driver
[  283.141209] asus-fan (debug) - probe for device
[  283.141210] asus-fan (debug) - init hwmon device
[  283.141244] asus-fan (debug) - fan-id: 0 | get RPM
[  283.141245] asus-fan (debug) - |--> get RPM using acpi
[  283.141245] asus-fan (debug) - |--> evaluate acpi request: \_SB.PCI0.LPCB.EC0.TACH
[  283.141386] asus-fan (debug) - |--> acpi request returned: 0
[  283.141386] asus-fan (debug) - fan-id: 0 | get RPM
[  283.141387] asus-fan (init) - created hwmon device: hwmon6
[  283.141387] asus-fan (init) - finished init, found 2 fan(s) to control
```

```
petr@UX410:/usr/src/asus_fan-master$ inxi -Fxz
System:    Host: UX410 Kernel: 4.15.0-38-generic x86_64 bits: 64 gcc: 7.3.0
           Desktop: Gnome  (Gtk 3.22.30-1ubuntu1) Distro: elementary OS 5.0 Juno
Machine:   Device: laptop System: ASUSTeK product: UX410UAK v: 1.0 serial: N/A
           Mobo: ASUSTeK model: UX410UAK v: 1.0 serial: N/A
           UEFI: American Megatrends v: UX410UAK.308 date: 11/08/2017
Battery    BAT0: charge: 38.1 Wh 98.1% condition: 38.8/48.0 Wh (81%) model: ASUSTeK ASUS status: Full
CPU:       Dual core Intel Core i5-7200U (-MT-MCP-) arch: Kaby Lake rev.9 cache: 3072 KB
           flags: (lm nx sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx) bmips: 10848
           clock speeds: max: 3100 MHz 1: 600 MHz 2: 600 MHz 3: 600 MHz 4: 600 MHz
Graphics:  Card: Intel HD Graphics 620 bus-ID: 00:02.0
           Display Server: x11 (X.Org 1.19.6 ) drivers: modesetting (unloaded: fbdev,vesa)
           Resolution: 1920x1080@60.01hz
           OpenGL: renderer: Mesa DRI Intel HD Graphics 620 (Kaby Lake GT2)
           version: 4.5 Mesa 18.0.5 Direct Render: Yes
Audio:     Card Intel Sunrise Point-LP HD Audio driver: snd_hda_intel bus-ID: 00:1f.3
           Sound: Advanced Linux Sound Architecture v: k4.15.0-38-generic
Network:   Card: Intel Wireless 8260 driver: iwlwifi bus-ID: 02:00.0
           IF: wlp2s0 state: up mac: <filter>
Drives:    HDD Total Size: 512.1GB (8.2% used)
           ID-1: /dev/sda model: SanDisk_SD8SN8U5 size: 512.1GB
Partition: ID-1: / size: 467G used: 38G (9%) fs: ext4 dev: /dev/dm-1
           ID-2: /boot size: 705M used: 155M (24%) fs: ext4 dev: /dev/sda2
           ID-3: swap-1 size: 1.03GB used: 0.00GB (0%) fs: swap dev: /dev/dm-2
RAID:      No RAID devices: /proc/mdstat, md_mod kernel module present
Sensors:   System Temperatures: cpu: 38.0C mobo: N/A
           Fan Speeds (in rpm): cpu: N/A
Info:      Processes: 266 Uptime: 29 min Memory: 2880.7/23992.2MB Init: systemd runlevel: 5 Gcc sys: 7.3.0
           Client: Shell (bash 4.4.191) inxi: 2.3.56 

```

daringer, 2018-11-24:
Well, this does not look very good, my guess would be that none of the implemented ACPI targets are actually in use within the zenbooks with the 8th generation, without such a laptop and currently I do not even have my UX32VD anymore, have plans getting one from this generation, but likely not before beginning of 2019... even then it might be a pretty hard job to go the loop again to reverse engineer what ASUS has done there... there is a perl tool in the repo which gives you alot of ACPI infos, you could use this before you do anything with asus-fan to see which acpi endpoints are actually changed during changing fan-speeds .... some more ppl have done this investigation, just check the other issues

xtelcz, 2019-06-12:
Issue solved by https://github.com/dominiksalvet/asus-fan-control
Thank you @dominiksalvet

closed 2019-06-12 by xtelcz

dominiksalvet, 2019-06-12:
@xtelcz You are more than welcome! :smiley: