daringer/asus-fan #70: hwmonX not available

pedro2555, 2019-10-31:
## Description

I've installed via the provided `ubuntu_dkms_sudo_install.sh` script. _One weird thing is it didn't request for elevated permissions._ Aparently all went ok.

Running `sensors` does give some new outputs.
```bash
$ sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:     25500 RPM
temp1:        +38.0°C  

asus_fan-isa-0000
Adapter: ISA adapter
CPU Fan:     4813 RPM  (min =   10 RPM, max =  255 RPM)
GFX Fan:     6978 RPM  (min =   10 RPM, max =  255 RPM)
gfx_temp:         N/A  (crit =  +0.1°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +39.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +37.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +37.0°C  (high = +100.0°C, crit = +100.0°C)
```
_Where previously only the `coretemp-isa-0000``section was available._

But the `hwmonX` thing is not available.
```
/sys/class/hwmon$ ls
hwmon0  hwmon1  hwmon2  hwmon3  hwmon4  hwmon5  hwmon6  hwmon7  hwmon8
```
## The Machine
It is a `UX303UB`, and is currently facing an issue where the fan is always at full speed. From post.

daringer, 2019-11-02:
can you please check with `lsmod` whether the `asus-fan` module is loaded, if yes please also post your dmesg output that is related to `asus-fan` loading, you might `rmmod asus-fan` and then `modprobe asus-fan` in order to see the initialization of `asus-fan` at the end of the dmesg log...

pedro2555, 2019-11-02:
The module is loaded.
```bash
$ lsmod | grep asus
asus_nb_wmi            28672  0
asus_wmi               32768  1 asus_nb_wmi
sparse_keymap          16384  1 asus_wmi
asus_wireless          20480  0
asus_fan               16384  0
wmi                    32768  2 asus_wmi,mxm_wmi
video                  49152  2 asus_wmi,i915
```

Here is the output you asked for. Sounds like something is wrong 
indeed. Again, remind you this unit currently has an issue where the 
fans are always at 100% (highly supect an hardware issue), would be 
good to fix it via software.
```bash
dmesg | grep asus-fan
[   15.782739] asus-fan (init) - dmi sys info: 'ASUSTeK COMPUTER INC.'
[   15.782740] asus-fan (init) - dmi product: 'UX303UB'
[   15.782994] asus-fan (init) - fan-id: 1 | failed to get rpm
[   15.783219] asus-fan (init) - fan-id: 2 | failed to get rpm
[   15.783221] asus-fan (init) - temp-id: 1 | success getting temp
[   15.787325] asus-fan (init) - created hwmon device: hwmon6
[   15.787326] asus-fan (init) - finished init, found 1 fan(s) to 
control
```

daringer, 2019-11-02:
but this looks like the hwmon device was created:
`[   15.787325] asus-fan (init) - created hwmon device: hwmon6`

please check `/sys/class/hwmon6` for its contents, best case post the output of the following line while inside the hwmon directory:
`for a in *; do echo $a $(cat $a); done`

another option you have is to force_load the module to see if this changes anything (as described within the [README](https://github.com/daringer/asus-fan)

and digging even deeper your can try using the `test_module.sh` and `acer_ec.pl` scripts to be found in the [misc](https://github.com/daringer/asus-fan/tree/master/misc) directory...

pedro2555, 2019-11-03:
Oh. Ok, I see. That was a miss understanding from my part.

Thanks for the help. Everything looks fine indeed, I was just unsure about the exact path after all.

I don't seem to be able to control the fans, but that is probably another issue.

closed 2019-11-03 by pedro2555