daringer/asus-fan #71: Support for Asus FX505DU (dual fan)

otaj, 2020-01-18:
Hi, would it be possible to get support for Asus FX505DU (not a zenbook, I know). It is a dual fan notebook, however I have not been able to detect second fan.

`sudo modprobe asus_fan` fails with `modprobe: ERROR: could not insert 'asus_fan': No such device`

I am running Arch Linux, if it's any help. I am willing to get my hands dirty and help with debugging, however I do not know much about kernel (and kernel modules) development.

daringer, 2020-03-08:
I have not too much hope that non-Zenbooks might work. Have you tried `force=1` loading the module?
Furthermore my might play around with [this perl script](https://github.com/daringer/asus-fan/blob/master/misc/acer_ec.pl) as most of my early and other reverse engineered information originate from the observation of the ACPI values in realtime, while trying to come up with relations between fan speed(s) and the changing numbers.

I intentionally leave the huge #44 thread open because it is packed to the max with details on what and how to do to investigate ACPI, especially fan stuff on the Zenbooks, so feel free to get inspired :smile:

daringer, 2020-03-08:
btw. you could try [afc-scout](https://github.com/dominiksalvet/afc-scout) in order to get some ACPI hints...