daringer/asus-fan #72: modprobe: ERROR: could not insert 'asus_fan': No such device

ranjithshegde, 2020-03-07:
On Arch linux

followed your instructions, tried both methods of manual install and via AUR, installed without any errors

I get the following error when I modprobe

modprobe: ERROR: could not insert 'asus_fan': No such device

more-info:
Model: Asus Rog Strix G731
Chipset HM370
Super I/O IT8528E

daringer, 2020-03-08:
As the ASUS ACPI stack seems to only change marginally I am pretty sure I will continue, once I again have a zenbook at my disposal. Waiting since beginning of the year to finally get the latest ryzens 4xxx within a zenbook. 
So under the line not too much to hope from my side for short-term improvements/adaptations. The best thing you could do is to reverse engineer as many details as possible and play around with the (acpi) tools as discussed in #44. This would be anyways the information which I would need upfront, together with a device, which can at least test one (config)path of the module the support for your model wouldn't be that far I believe...

daringer, 2020-03-08:
there is also [afc-scout](https://github.com/dominiksalvet/afc-scout), which might also help you to ivestigate ACPI details.