daringer/asus-fan #74: .INSTALL: line 11: systemctl: command not found

zorbanaut, 2020-04-16:
When installing from AUR:

```
Enabling asus-fan as service in systemd
/tmp/alpm_oXXUVV/.INSTALL: line 11: systemctl: command not found
error: command failed to execute correctly
```

Commenting out that line in asus-fan-dkms-git.install helped.

daringer, 2020-04-17:
this weird, as `systemctl` should indeed exist, maybe the full path would help, trying a non-tested fix in a minute

daringer, 2020-04-17:
could you maybe re-run and report if it's working now?

zorbanaut, 2020-04-17:
Sorry, I was not clear about my environment. I don't have systemctl installed.
Is it possible to add a condition to skip this step when systemctl is not available?

daringer, 2020-04-18:
no problem, please try again now, should just warn and continue...

zorbanaut, 2020-04-19:
Thanks for the fix, it works for me now.

"xsystemctl" is a typo?

daringer, 2020-04-20:
kind off, intended debug-typo, thanks for pointing it out, hopefully now correct, 
closing here, don't hesitate to reopen if needed

closed 2020-04-20 by daringer